meerk40t/meerk40t #489: Report: Too Fast Laser Jog

tatarize, 2021-08-13:
https://www.facebook.com/groups/716000085655097/posts/901702760418161/

> Is there any way at all to slow down the rapid movements between parts? Here you can see my last job. All circles in left cut well from top to bottom, then it began the raster from bottom to top. Now it realised it needed to cut the rest of the circles. It decided to do this from bottom to top so hurtled it self to the bottom of the sheet in an attempt to break the speed record for a laser head, naturally breaking some law of friction/traction/who knows what and losing position.
If I don’t merge operations the same thing happens but for the whole cut. I just need some way to say “take it easy when moving to the next bit!”

It likely should have moved to the bottom at rapid speed but may have somehow used raster speed, and 50mm/s for the cuts would be fine but 140 mm/s for the raster would certainly have caused some issues. Investigation is required in case this is a critical bug.

tatarize, 2021-08-13:
#446 May be related

Sophist-UK, 2021-08-13:
[image]

1. User says there was "text off canvas". If this causes gantry to hit stops to try to get to it, that would cause a shift.
2. It is possible that a 50mm/s travel in Y-direction would cause a skip.
3. Note in the photo the over burned circle arc where circles overlap after the shift, and the corresponding underburn in the opposite arc. A shift by itself does not explain this. Unclear whether this is a separate issue (only simply coincidence that it happens at the same time as a shift) OR a cause or a consequence.

Note: This is an example why we should be able to optimise to group an outermost cut with all inner operations when we have Merge Operations and Cut Outer Last (or as currently named Cut Inner First - which suggests inner only applies to Cuts).