meerk40t/meerk40t #490: [Enhancement] Add element mirror action

Sophist-UK, 2021-08-13:
Add ability for user to reflect selected elements about a vertical or horizontal axis in the centre of the bounding box.

See [this facebook post](https://www.facebook.com/groups/2601804599938057/posts/4117569995028169/?comment_id=4117869264998242&reply_comment_id=4118019658316536&notif_id=1628868118679210&notif_t=group_comment_mention) for details.

tatarize, 2021-08-14:
This actually seems very similar to step & repeat which is a feature in 0.6.x but not working in 0.7.x. It basically does a grid but inside the planner.

Sophist-UK, 2021-08-14:
No - nothing like Step & Repeat. This is a Transform adjustment whereby you mirror in X or Y axis around the centre of the bounding box.

tatarize, 2021-08-14:
Sounds a little bit like flip. Which flipped a design around some material. So like if you had a certain material width it would flip it to the other end if you gave the location. I don't see any access to the functionality still existing but it does exist code-wise. 

Should be accessible under `dialog_flip` in the `tatarize/console-fixes` branch.

Sophist-UK, 2021-08-15:
As the code to do it actually exists, I can probably do a PR which adds the right-click menus.

I will see if there are any other menu stuff that is missing or could do with tweaking at the same time.

tatarize, 2021-08-16:
I did a hidden command for it in my last PR. You should be able to do `dialog_flip` as a console command. This should allow for a flip across a particular bit of material. All other affine transformations are equally allowed.

tatarize, 2021-08-21:
dialog_flip is live. Unsure if this settles the issue.

Sophist-UK, 2021-08-21:
No - user wants a GUI action for it.

Let's fix all the mouse issues, and then I will do a trawl through the UI and make a PR to make it all more consistent.

Sophist-UK, 2022-02-15:
Done.

closed 2022-02-15 by Sophist-UK