meerk40t/meerk40t #491: [UI] Group Pulses / Pulse Shifting

Sophist-UK, 2021-08-13:
In Advanced Operation Properties we have Group Pulses. In Device Config we have Pulse Shifting.

Are these alternative names for the same thing? If so can we change the Terminology so that it is consistent?

"Pulse Grouping"?

Sophist-UK, 2021-08-13:
P.S. If this is agreed, I can code the change.

P.P.S. Do these options work on Vector or Raster Operations or both?

P.P.P.S. Any chance of a simple explanation for mere mortals such as myself? It seems to work on groups of 4 bits - a simple explanation of what each of the 16x 4-bit combinations translates to would be great.

tatarize, 2021-08-13:
Yes, they are inconsistent names for the same thing.
Yes, they should have a name change.

They shift a pulse around.

Basically _ is no laser and X is laser.

```
____ -> ____
___X -> ___X
__X_ -> __X_
_X__ -> _X__
X___ -> X___
X__X -> X__X
X_X_ -> XX__
XX__ -> XX__
_X_X -> _XX_
_XX_ -> _XX_
__XX -> __XX
XX_X -> XXX_
XXX_ -> XXX_
X_XX -> _XXX
XXXX -> XXXX
```

Basically pulses get shifted by 1 where needed. So X_X_X_X_X_X_X as a long pulse train gets made _XXX___XXX___ as the X_ gets moved towards the X and the same on the other side. So pulses can be moved a bit in either direction.

Also, this happens real time in the pulse plotter so if you turn it on while it's running it *should* instantly apply. So you should go from stuttering to not stuttering if you engage the feature.

Sophist-UK, 2021-08-13:
What CutCode types does it apply to? Raster / Image, Vector, both?

tatarize, 2021-08-13:
All. The cutcode decides the positioning of the pulses, the PPI and shift pulses is in the pulse planner and comes afterwards. Much like things like dot length etc, it's actually done later and manipulates the pulses. So it applies to raw pulses so all cut types.

Sophist-UK, 2021-08-13:
Can we have an Emergency Stutter Fix pane/button (like the Emergency Stop one) that turns this on until the laser is idle again?

Sophist-UK, 2021-08-13:
I assume that the pulse planner has access to the operation settings at the time that it was sent to the spooler so it knows which operations had Pulse Group set?

tatarize, 2021-08-13:
A little bit yes, it's a bit hamfisted though. The Pulse Planner does get access to the settings. But, also communicates with some rather kludgy methodology as to whether there's new settings and or if certainly things should be checked. I did some research trying to see if there was a more organic and natural way to do that but left it as is.

tatarize, 2021-08-13:
The plotplanner has settings and the generated values are cutcode and all the cutcode has settings. If the setting from the previous cut is different than the current set it yields a `None None, PLOT_SETTING` This is read during the LhyStudiosDriver.plotplanner_process() which checks of the on value is equal to plot setting and if it is updates the settings., and does this on from the settings on the PlotPlanner, which would have been updated since it yielded that PLOT_SETTING value for the control code.

This code section may be the sloppiest bit of code in the whole set of routines. Since it's communicating basically with semiphores to notify the driver that it needs to update different sections. It has codes for updating settings, plot axis, direction, whether it should jog or rapid jog. 

Basically plot planner is supposed to give x, y, and on-off information so go to this point with a laser and laser there, go to this point. But, it also gives on values that mean things like update your settings since there was a speed change or something and the driver might have the older values.

tatarize, 2021-08-13:
There's also some general sloppiness in that cutcode basically generates COMMAND_PLOT and COMMAND_PLOT_START commands in lasercode. Which are generally interpreted by the drivers but these cutobjects are more or less always cutcode and really use the plotplanner to read them. It feels like some of these parts could be foregone or made more coherent and simpler but I couldn't see a very easy method of doing that. The drivers consequently have a plotplanner that reads the COMMAND_PLOT and turns it into the various movement and laser on/off code, that also uses those weird semiphores to communicate things like it should update the local settings since the plotplanner settings have been updated by a new bit of cutcode.

There's reasons for all of this but it seems like there's like 3 different highly general ideas tied into a knot that makes them highly linked together.

The spooler accepts lasercode. Cutcode produces laser code of COMMAND_PLOT commands. The PLOT commands are interpreted by the PlotPlanner with extra semiphores, to do things like tell it to update speed. 

The need for the plotplanner is that I wanted to be able to merge similar and identical plotting code. So the different cutobjects linked together would work correctly as a group. They can be flushed out of extra data so if some processes in the plotplanner require a buffer these can be corrected at the end. But, other than that it could do things like move pulses around, give data in grouped batches, apply ppi, etc. It would be fed a bunch of plot points and give a very nice pulse train as decided by general algorithms.

The fact that it needed to know speeds etc, changed was an issue that I just sort of kludged around. But, it also includes a bunch of weird things like telling the driver which direction will be happening next. The M2 Nano has a thing where to twitch the right direction, it needs to know which way it will go after entering program mode. So the plotplanner relays that information to the driver. But, this is likely only a thing with the M2 Nano. So it came together somewhat kludgier than I had hoped.

But, the pulse shifting is part of the plotplanning which is done independent of the backend being used. So pulse shifting could be done on a Moshi-board it just wouldn't mean much to the Moshi since stuttering isn't a problem there. And it would ignore the other signals that relay future but needed information for the M2 Nano. Like, after entering program mode, which directions should be set active because I will use those directions first, or whatnot.

tatarize, 2021-08-13:
The ZinglPlotter just produces position that perfectly trace a particular shape with integer values. So it's like the bresenham line draw algorithm but for bezier curves, etc. This gives me a series of pulses for the generator. But, the M2 Nano actually wants diagonal and orthogonal directions. So 3 right, 1 right-up diagonal, 3 right, 4 right-up diagonal. And to convert the plots into that took some planning which got put in plot-planner.

There's also a long standing bug in plot planner where it crashes here and there since it has non-orthogonal values at some point when it should absolutely have them planned out to be correctly orthogonal. But, the PlotPlanner is a series of plots passed through a series of yield commands.  Specifically: Group, Shift, PPI, and Single all wrap the plot in order. So the input values are turned into single position plots, then ppi is applied to that, then shifts are made in that PPI if it reduces the amount of plotting a bit, then it's put into orthogonal/diagonal groups. 

It crashes on rare occasions in Group() which might somehow be moving or failing to have flushed or something. And the group_dx and group_dy ends up greater than single step values. Though I've been over the code repeatedly without figuring out anything. 

And the whole thing feels kind of complex when it's supposed to be general but very powerful. In theory you could replace the plot-planner in the driver with a different one since it's really reading the lasercode and the cutcode would create COMMAND_PLOT commands and COMMAND_PLOT_START commands so it might be somewhat general but I don't really see any other methods of doing this or to make the various parts less intermingled.

tatarize, 2021-08-14:
> Can we have an Emergency Stutter Fix pane/button (like the Emergency Stop one) that turns this on until the laser is idle again?

If this is turned on while it's running, it should be able to do that in Config, and prevent the stuttering. I think it overrides the default setting in the Operation. Which is a distinction between the two settings. The one in operations is default, the one in Config overrides that regardless what value they have they will be forced to be turned on, which should quickly change the behavior. So both of them mean the same thing but the config one overrides whereas the one in operations just sets that value within that operation. Like if you wanted it for the raster, but didn't care for the vector stuff it would change it's setting accordingly.

tatarize, 2021-08-21:
Completed.

closed 2021-08-21 by tatarize