meerk40t/meerk40t #500: Raster operation optimisations

Sophist-UK, 2021-08-18:
Today I rastered an advertising sign which had a frame edging it. About 10" x 8".

The frame consists of top and bottom which rastered full width and two side bars which rastered only 1/4" each side with a 9" gap in between. These took 2s per sweep at 120mm/s and at step 3 = c. 300 steps per inch, each inch took c. 10mins. So the c. 5" sides-only took just under an hour, most of which was sweeping over a big non-burn area.

I could have manually split the frame path into two (top, bottom and one side in piece one, the other side in piece 2 in a separate Raster op), and then it could have burned each side separately. I estimate that this could have saved c. 40mins of the 50mins.

To put this in context, the travel optimisations are going to save perhaps at most 1-5 mins on a burn!!! So you can see just how big an impact you could have by automatically optimising rasters.

This issue proposes two possible approachs:

1. Dynamic variable speed - when you analyse a raster row, if there are wide areas of no burn then accelerate and decelerate to cross the white space in less time. Not sure whether the Lhymicro protocol will support this.
2. Area analysis - automatically break up images into smaller chunks that can then be optimised by combinations.

I can provide video and diagrams if needed to explain this more clearly.

tatarize, 2021-08-19:
I've considered some of these elements before and the best conclusion I came to is you could optimize these with a proper optimization routine.

You take an image and you turn it into a series of lines. You need to complete these lines in the minimum amount of time. So the question is, if we know that a sweep costs something like 115 ms or so. We could ask @joerlane how long the sweeps actually take. We could minimize the time, if we figured out what the fastest travel path, taking into account that breaking a sweep into multiple sweeps increases the time for that particular path but if the path length is reduced by an acceptable amount we could figure that path.

So the letter W for example might go down one side and back up the next in 3 different rasters one stepping down, then up, then down, then up. 

A somewhat easier solution might be to cut the image into sub-images. So the raster op could well turn into 4 images rather than 1 if the math called for it. We'd manage to capture the entire relevant area in 4 edges and never need to travel along white space. And the optimization could likely be done quite quickly on the image itself.

tatarize, 2021-08-19:
We might also do something akin to kmeans though the metrics might be difference since if it's in a correct line we're fine, but if you could slice an image excluding white you'd have the right methodologies there. And we could pre-process this as an image. It could well work like an image split routine and create a series of images out of one that could be rastered faster than the image as a whole.

You need a method to make K rectangles capture all black pixels in a field.

[image]

tatarize, 2021-08-19:
I posted a question on SO.

https://stackoverflow.com/questions/68842542/algorithm-for-cluster-splitting-a-black-and-white-image-multiple-images-without

There's a solid enough chance somebody could solve this. It's a fairly well defined problem at least conceptually. 

If we omit about 2.1cm per scanline it's more efficient to run the process as 2 rasters rather than one.
You can view this in terms of multiple overlapping images. And you are assigning a pixel to one image rather than another. And you only need to solve for the best cluster of images that reduces white space between pixels by enough to save the cost of the additional raster. 

In theory if we assigned the right most pixel to one image and the leftmost image to another image in those cases when the distance of whitespace between the pixels is above a certain threshold we might be able to solve the problem iteratively. Then when solved if the same algorithm applied to divided algorithm produces a non-null result then we are better off creating 3 images etc.

I could certainly see there being a fast and reasonable solution to this. Something like if we have a scanline with a gap of white pixels beyond a threshold we call that a cut. Then assign all black pixels to the right of the cut to a new image. It might be fairly close. We could then go back and assign pixels to the other image if they are closer but were not on the correct side of the cut.

Sophist-UK, 2021-08-19:
I think it's actually simpler than that - because of the distance needed to stop and start or to accelerate further and the decelerate, we only need to look for empty areas - which means zero bytes.

Accelerating further / decelerating can be calculated on a row by row basis on the fly (and you can calculate the impact when estimating time), but finding significant white areas is probably a little more complex.

Sophist-UK, 2021-08-19:
Here's the relevant calculation to determine whether a split into 2 (or more) rasters is appropriate:

For a raster speed `v`, acceleration `a` and gap `d` it is worth splitting into 2 if d> 2v^2/a.

An algorithm that I would propose is as follows:

Count the number of gaps where this is worth doing. Create that number of additional bitmap objects of exactly the same size. (In practice you would do this row by row adding additional bitmaps as needed.)

Then you split pixels for that row appropriately between bitmaps.

When you have done all rows, you crop the bitmaps to the minimum size containing all non-blank pixels. This splits the image into the several areas you need.

Then you can use optimisation to decide which order to raster in and if we do this optimisation, whether to allow rasterisation in the opposite direction i.e. if B2T is selected, allowed to do T2B.

tatarize, 2021-08-19:
https://github.com/meerk40t/meerk40t/commit/ec5fd5a60cb23da5ad9192ab1e6306296ce50aeb

I added in manual commands for these things. There is now `image slice <x>` and `image pop <left> <top> <right> <bottom>`

This will slice the image. So if there's a frame you could do `image slice 50%` and it would slice the image frame in half, doing the right and left sides independently which would gain you all the speed advantage, but you'd need to know this before hand and it wouldn't automatically be able to slice this in the best way.

I also added `image pop` which is very similar to crop except that the cropped image is put over the current image and the cropped bit is set to be white in the background.  So for example: `image pop -r 10% 10% 50% 50%` would pop off the section from 10% 10% to the center of the image, with the -r flag it would not crop the background to white. This could be helpful if you wanted to raster an image bit twice to give it a 3d feel. Which is a thing some skilled folks have managed to do. In theory popped sections of image should overlap nicely. But, if you took an image and popped all places where there were pixels, you'd have solved for a raster optimization.

closed 2022-07-20 by tatarize