meerk40t/meerk40t #501: [Proposal] wxPython wrapper for popup windows

Sophist-UK, 2021-08-18:
Glade code is pretty unreadable, yet we attempt to maintain it. I propose we create a wrapper which allows us to define pop-up window layouts simply and effectively as a simple hierarchy of calls, one call per window element, providing **all** the information needed to set up the window and calculate e.g. min, max and default sizes (and make it scale gracefully and with poise).  

The following conceptual example is based on the settings window:
[image]

Ideas are not fully researched or formed, and are intended to be conceptual only.

1. Window is defined by a hierarchy of calls, one call per graphical wx element:
```
wxmk.window([
    wxmk.sizerhorizontal([
        wxmk.sizervertical([
            wxmk.boxsizer([
                wxmk.radio(["Units",["mm", "cm", "inch", "steps"], other stuff),
                ], other stuff),
            wxmk.boxsizer("Language", [
                wxmk.dropdown(
```

Partial save to be edited.

tatarize, 2021-08-19:
I don't see this being that much easier. There is clearly some truth to the maintainability of the Glade code. There's not a very good builder for that stuff in wxPython. It's somewhat reasonable but I don't see this being easier to maintain. Since the main issue is still you want a dialog built in a gui-builder like Glade. And just generalizing that stuff doesn't seem to make that easier.

It's simply the case that the dialogs are not built in anything that is somewhat modular, and so copying out the relevant parts to make the window is still likely the best method forward. The power of building the gui in glade is that you get a wysiwyg frame.

If you have a glade pane or something I could certainly go through and integrate it into a meerk40t pane as needed. Or document the needed changes. But, hand building them with hand-building commands is nearly never going to be superior.

Sophist-UK, 2021-08-19:
Really?

1. When you know what you want to display, it is easy to code it IMO. Tweaking it is as simple as moving lines elsewhere.
2. One call per GUI item is way better than 3-5 calls as at present.
3. Because Glade designs are done by eye, they are **never** going to look consistent. The extra functionality behind the scenes in this proposal will make the GUI much more consistent and less ... um ... glitchy looking. This is important for what you might call brand-image. If you want to "compete" with Lightburn, this is important.

tatarize, 2021-08-21:
Well if you want, you could try a large bit of gui. There's a need for a grbl controller that has values to set all the $ commands. 

[image]

The idea being to allow the user to configure their GRBL devices and send that data to the backend through the character counting protocol. While it's not currently there, there is a notable need for a GRBL controller gui and a Ruida controller gui, as well as preferences for them. Not sure where the $ settings belong. But, the point remains that eventually supporting these would require having useful guis that can modify those settings. 

The GRBL settings would look like a bit like the LaserGRBL settings and the Ruida settings would look a lot like RDWorks settings. Both of which would need rather extensive windows.

Sophist-UK, 2021-08-21:
Sorry - but I think I need to start with one piece of existing code to write the wxPython wrappers. 

Also, I have tried creating a GRBL device, but cannot open LaserGRBL settings.

If I code it for Settings and you like the concept, I can then code all other Windows. We should also be able to have common code for Panes and Windows - say files in /meerk40/gui/widgets which we import into Panes or Windows as needed. 

Once I have coded existing widgets, we can then extend it to new settings.

tatarize, 2021-10-30:
I think there's actually pretty considerable overlap with this idea and the idea behind `choices`.

https://github.com/meerk40t/meerk40t/wiki/Tech:-Choices

The rightmost value there in Settings (which actually will be renamed to Preferences as confusing as that is, but it's a mac standard thing) is created by a dynamic list which isn't registered in the kernel yet, but easily could be as part of a choices set of values. Just as those choices are dynamically, we can dynamically make other choices with properties panes.

This provides, not a standard wrapper for designing the window, but a standard wrapper for defining what options we have and if we provide those options we can create dynamic interfaces that actually work rather robustly. So if we choices a value one place it will get altered in other places.

There's a couple needed updates:

* The `action` type for choices needs to get added to PropertiesPanel
* The dynamic Preferences here in Settings need to be converted to a `choices` set of value, and likely placed within `wxmeerk40t.py` for now (some of these are rarely used settings for things like the planner and just got dumped here).
* The meerk40t/core/elements `tree_operations_for_node` currently returns a function with a bunch of attributes that are checked. This function should actually generate choices for a node instead. Then we could display this information in a panel, if we wanted to for some unknown reason.

There's likely a few other things that could help define how we should render this kind of alterable properties.

---

Now this wouldn't create necessarily create our windows as you propose but it does fit the central idea behind this Issue.  It might be possible to render other controls within our window so we get options for a ComboBox or a `kernel.Channel` (large text box that watches that given channel). And these might be able to render something relatively well. While it wouldn't have the fine-grain control it would slap together a display and let you click the things you need to change.

Sophist-UK, 2021-10-30:
Choices is a complementary approach which I fully support. It effective decouples GUI from underlying functionality i.e. it places the definition of e.g. optimization options in the same MK plugin that provides the optimization. I might question whether it would be better as a dict of dicts rather than a list of dicts, but the underlying principle is IMO a good one.

The suggestion in this issue complements the above by abstracting wx window / panel / etc definitions to their simplest form and adding the glue functionality in common methods. This would simplify the creation / modification of MK gui windows / panes etc. and ensure that all aspects of the UI act consistently. It would also abstract away any O/S specific changes, and allow us to refine the UI further so as to e.g. prevent user typing alphabetic characters into numeric fields etc.

If this works and works well, it might be spun out into a separate wx toolkit, and possibly we could then get Glade to generate code for this directly in order to have a GUI designer.

Sophist-UK, 2021-11-01:
If we do this we can probably also integrate support for unit-testing using this: https://wxpython.org/Phoenix/docs/html/wx.UIActionSimulator.html

tatarize, 2022-07-20:
I think choices has advanced to being pretty solid for this. 

I would have no trouble creating either example in this Issue with a choices routine. The original note with the checkbox actually *is* choices now.