meerk40t/meerk40t #525: v0.7.0 RC9: Please report any issues found here

Sophist-UK, 2021-08-28:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b34 RC-9
The latest beta release is v0.7.0beta34 Release Candidate 9. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta34

# CAUTION
**CAUTION:** Please note that (depending on your design) if you use Merge Operations after you have used Duplicate Operation, or you use Merge Passes when you have set Operation Passes > 1, these can result in the laser beam burning the same path several times in quick succession without allowing time for the material to cool down in between. This can have unintended consequences. **Please do not use this feature of the beta whilst your K40 is unattended.**

Sophist-UK, 2021-08-28:
```
MeerK40t crash log. Version: 0.7.0 RC-9 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1478, in run
  File "meerk40t\core\drivers.py", line 124, in _driver_threaded
  File "meerk40t\core\drivers.py", line 135, in _process_spooled_item
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 846, in plotplanner_process
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1027, in goto_octent_abs
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1044, in goto_octent
ValueError: Not a valid diagonal or orthogonal movement. (dx=2, dy=-4)
```
Did engrave of one stitch hole in a small strap (6 holes as engrave, rectangular cut) and gave this error on what seemed to be the first travel move.

dpticar, 2021-08-28:
Hi,

I have tried cutting file in attachment and RC9 it's not fully cutting 2 paths (path881 and path1608).  On RC8 everything is ok but it's slow on engraving dashes.
[rc9_test1.zip](https://github.com/meerk40t/meerk40t/files/7071656/rc9_test1.zip)

tatarize, 2021-08-29:
I fixed a repeating error. Where it would fail if you ran the same object twice in a row that started and ended at the same point and had a origin command between the two commands. 

Another minor crash.

And the selected objects in the scene, fixed.

Hoping to push out RC10 rather quickly after fixing some of the worse errors.

tatarize, 2021-08-29:
@Sophist-UK your test file gave both errors. -2, 4 crash and the bottom three ran after outline bit. But, all are clearly marked as cuts.

tatarize, 2021-08-29:
Fixed the inner first bug, and the sophist crash there. I'm down to dpticar's report for RC10 (RC-9 is too buggy)

tatarize, 2021-08-29:
I was flummoxed for a bit how the hell you made these dashes. Then I saw they are legit paths. 

https://www.youtube.com/watch?v=sHMY-FEYfsY

Is the alternative there. Since inkscape dots like that end up having a width it'll draw around every single dot. Which while functional might be better off with a dot length.

tatarize, 2021-08-29:
https://www.youtube.com/watch?v=eHeuntGJaAs Seemed like it worked fine. Might have been some of the other things. Your dashed line is better than I've seen in inkscape it's actually pretty awesome. Not sure how you did that.

dpticar, 2021-08-29:
> 
> 
> https://www.youtube.com/watch?v=eHeuntGJaAs Seemed like it worked fine. Might have been some of the other things. Your dashed line is better than I've seen in inkscape it's actually pretty awesome. Not sure how you did that.

I made dashes in inkscape 0.92 (I have not tried in 1.x 
1)  Select path
2) Fill and Stroke  -> Stroke style -> Dashes (set your options)
3) Main Menu -> Extensions -> Modify Path -> Convert to Dashes

Sophist-UK, 2021-08-29:
RC10 is now available - please report all issue in #528.

closed 2021-08-29 by Sophist-UK

Sophist-UK, 2021-08-29:
@tatarize Thanks for such a quick response to the diagonal crash.