meerk40t/meerk40t #528: v0.7.0 RC10: Please report any issues found here

Sophist-UK, 2021-08-29:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b35 RC-10
The latest beta release is v0.7.0beta35 Release Candidate 10. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta35

# CAUTION
**CAUTION:** Please note that (depending on your design) if you use Merge Operations after you have used Duplicate Operation, or you use Merge Passes when you have set Operation Passes > 1, these can result in the laser beam burning the same path several times in quick succession without allowing time for the material to cool down in between. This can have unintended consequences. **Please do not use this feature of the beta whilst your K40 is unattended.**

Sophist-UK, 2021-08-30:
# Bugs
1. Execute two operations from right-click with Merge Operations NOT checked, and operations are merged nonetheless.
2. Path with v0 segment (and presumably h0) is held by svgelements as l0,0 and causes closed paths with even number of Merge Passes to be burned piecemeal i.e. back-and-forth rather than round-and-round.

Sophist-UK, 2021-09-02:
# Bugs
1. Do not allow user to cancel the current op in Spooler - it seems to delete the following op or two.

tatarize, 2021-09-02:
[image]

No critical lines found in sven's report but he clearly has them.

[image]

Further investigating...

tatarize, 2021-09-02:
[image]

tatarize, 2021-09-02:
* After restore, double click on cutcode in operations does not simulate it correctly. Cutcode is not flagged as optimized and will try to optimize anyway.
* The execute should detect automatically done stages and not require a click for them.

tatarize, 2021-09-02:
```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "mk40t.py", line 13, in <module>
...
File "meerk40t\gui\wxmeerk40t.py", line 1089, in __set_ribbonbar
  File "meerk40t\gui\icons.py", line 16, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```
Locale bug. Needs to use other version. 

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\mwindow.py", line 72, in on_close
  File "meerk40t\kernel.py", line 545, in close
  File "meerk40t\gui\mwindow.py", line 92, in finalize
  File "meerk40t\gui\wxmeerk40t.py", line 2173, in window_close
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\kernel.py", line 2456, in shutdown
  File "meerk40t\kernel.py", line 917, in shutdown
  File "meerk40t\kernel.py", line 1684, in process_queue
  File "meerk40t\gui\navigation.py", line 574, in on_emphasized_elements_changed
  File "meerk40t\gui\navigation.py", line 620, in select_ready
RuntimeError: wrapped C/C++ object of type BitmapButton has been deleted
```

X2 RC7 hopefully old.

* Fixed RC7 bug
* Fixed RC8 Bug

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\console.py", line 126, in on_entry
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\device\basedevice.py", line 99, in virtual_dev
  File "meerk40t\kernel.py", line 407, in match
  File "meerk40t\kernel.py", line 1036, in match
  File "re.py", line 234, in compile
  File "re.py", line 286, in _compile
  File "sre_compile.py", line 764, in compile
  File "sre_parse.py", line 924, in parse
  File "sre_parse.py", line 420, in _parse_sub
  File "sre_parse.py", line 813, in _parse
re.error: missing ), unterminated subpattern at position 24
```
Base device for the virtual dev got changed maybe used a () within the command itself. Could have a ( in the input_driver.type or command. Likely it would have to be command. Console command: `hell(o` with the ( appearing in the grouping causes the crash here.

* 3X Fixed RC7 bounding box error.

```
MeerK40t crash log. Version: 0.7.0 RC-10 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3733, in specific
  File "meerk40t\core\elements.py", line 1570, in inner
  File "meerk40t\core\elements.py", line 4516, in merge_elements
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
[... 140 lines cut ...]
  File "meerk40t\gui\panes\camerapanel.py", line 296, in <lambda>
TypeError: %d format: a number is required, not builtin_function_or_method
```

This is `self.cam.Bind(wx.EVT_MENU, lambda e: self.cam.setting("camera%d perspective reset\n" % self.index), id=item.GetId())` which means camera reset perspective failed for the self.index which somehow was a method rather than a value. This is written to one time during the init of the camera panel.
```python
def __init__(self, *args, context=None, gui=None, index=0, **kwds):
```

This is initialized one time with `for index in range(5):` which is by definition an int. I don't see how this crashes. Maybe it's fixed.

----

hell(o in console crashes things.
Sophist classification bug in RC10.
```
  File "meerk40t\core\elements.py", line 5216, in classify
AttributeError: 'NoneType' object has no attribute 'alpha'
```

Which relates to line:
```python
if is_straight_line or element.fill.alpha == 0: # Straight lines and transparent fill cannot be rastered
```
Where element.fill can be None but this isn't checked.

tatarize, 2021-09-02:
[image]

@Sophist-UK --- Sometimes short names serve purposes.

Sophist-UK, 2021-09-03:
Fixed in #538:
```
MeerK40t crash log. Version: 0.7.0 RC-10 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3733, in specific
  File "meerk40t\core\elements.py", line 1570, in inner
  File "meerk40t\core\elements.py", line 4221, in refresh_clasifications
  File "meerk40t\core\elements.py", line 5216, in classify
AttributeError: 'NoneType' object has no attribute 'alpha'
```

Sophist-UK, 2021-09-03:
> Sometimes short names serve purposes.

Yes - sometimes, not globally. Element labels can now be excessively long - I have seen longer ones that that. We should truncate element labels at first ":" or "(" or store a short form of the element label.

Sophist-UK, 2021-09-03:
I have fixed the excessively long menu items - looking good. Currently working on language strings and usability improvements in tree context menu.

Sophist-UK, 2021-09-03:
# Bug
1. Rotate context menu is missing a whole load of entries.

Sophist-UK, 2021-09-04:
# Bug
1. Special Operations not saved (or at least not restored) with rest of operations.

Sophist-UK, 2021-09-05:
# Suggestion
Rather than make tree text the color of the element, rely only on the color of the icon / mini-me to show the color - this avoids the issue of white text on a white background.

Sophist-UK, 2021-09-05:
# Bugs
1. Merge Passes introduces twitches
2. Merge Passes still doing paths in reverse (and so burning back and forth rather than round and round)

tatarize, 2021-09-07:
1x hell(o in console crash (correct)
1x locale crash #242
7x lexan crash Special Command in code failed (corrected)
.alpha crash (corrected).

```
MeerK40t crash log. Version: 0.7.0 RC-10 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 1186, in open_simulator
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
  File "meerk40t\core\planner.py", line 717, in plan_validate
  File "meerk40t\core\planner.py", line 71, in execute
  File "meerk40t\core\planner.py", line 209, in make_image
  File "meerk40t\core\planner.py", line 198, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 478, in make_raster
wx._core.wxAssertionError: C++ assertion "bmp.IsOk()" failed at ..\..\src\msw\graphics.cpp(2575) in wxGDIPlusRenderer::CreateContext(): Should select a bitmap before creating wxGCDC
```
Likely high memory error caused the bigmap to fail.

```
MeerK40t crash log. Version: 0.7.0 RC-10 on win32
Traceback (most recent call last):
  File "meerk40t\gui\lhystudios\lhystudioscontrollergui.py", line 567, in on_button_start_usb
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 370, in usb_connect
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1586, in open
  File "meerk40t\device\ch341\ch341.py", line 120, in connect
  File "meerk40t\device\ch341\ch341.py", line 181, in _connect_mock
  File "meerk40t\device\ch341\ch341.py", line 163, in _connect_attempt
AttributeError: 'int' object has no attribute 'get_chip_version'
```
Mock adapter with a chip version criteria set. Confirmed crashes. Fixed.

```
MeerK40t crash log. Version: 0.7.0 RC-10 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 1186, in open_simulator
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
  File "meerk40t\core\planner.py", line 717, in plan_validate
  File "meerk40t\core\planner.py", line 71, in execute
  File "meerk40t\core\planner.py", line 209, in make_image
  File "meerk40t\core\planner.py", line 198, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 463, in make_raster
wx._core.wxAssertionError: C++ assertion ""w > 0 && h > 0"" failed at ..\..\src\msw\bitmap.cpp(752) in wxBitmap::DoCreate(): invalid bitmap size
```
0 sized raster? This can apparently happen when a 1x1 size raster is made smaller with the raster step.

```
MeerK40t crash log. Version: 0.7.0 RC-10 on win32
Traceback (most recent call last):
  File "meerk40t\gui\pathproperty.py", line 208, in on_button_color
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
  File "meerk40t\core\elements.py", line 3536, in element_classify
  File "meerk40t\core\elements.py", line 5326, in classify
  File "meerk40t\core\elements.py", line 5109, in add_classify_op
ValueError: 'Unknown' is not in list
```

```
MeerK40t crash log. Version: 0.7.0 RC-9 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1478, in run
  File "meerk40t\gui\rasterwizard.py", line 189, in wiz_img
  File "meerk40t\image\imagetools.py", line 1564, in wizard_image
  File "PIL\ImageOps.py", line 153, in autocontrast
  File "PIL\ImageOps.py", line 58, in _lut
OSError: not supported for this image mode
```
x2 Raster Wizard apparently failed for likely `p` type images.
`empty_mask = image.point(lambda e: 0 if e == 0 else 255)` So point invert for grayscale. 
`if self.mode == "F":` throws this. -- Fixed maybe. Not sure what loads F mode images

Sophist-UK, 2021-09-07:
RC10 is now available - please report all issue in #548.

closed 2021-09-07 by Sophist-UK

Sophist-UK, 2021-09-07:
# Bugs
I have a B&W SVG logo (Portuguese Scouts) which consists of various paths, some with fills and some without, which are handled very badly by RC10. 
1. Classifies incorrectly - any path with non-transparent fill should be classified as raster and only vector if an exact color match to a vector op. These were B&W and so should be handled better in RC11, though this enhancement will still be required.
2.  Raster image size does not properly take into account wide stroke widths at edges. Bounding box for element needs include the stroke width - this may be off-spec, so we may need a raster-bbox that includes this whilst normal bbox doesn't.
3. Adjusting pane sizes - had a funny where bar moved up when I moved mouse down. Suspect a +/- reversal somewhere in this code.