meerk40t/meerk40t #548: v0.7.0 RC11: Please report any issues found here

Sophist-UK, 2021-09-07:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b36 RC-11
The latest beta release is v0.7.0beta36 Release Candidate 11. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta36

# CAUTION
**CAUTION:** Please note that (depending on your design) if you use Merge Operations after you have used Duplicate Operation, or you use Merge Passes when you have set Operation Passes > 1, these can result in the laser beam burning the same path several times in quick succession without allowing time for the material to cool down in between. This can have unintended consequences. **Please do not use this feature of the beta whilst your K40 is unattended.**

tatarize, 2021-09-08:
Remove no work... likely need to push a new version to fix that.

Sophist-UK, 2021-09-08:
Removes work well.

# Bugs
1. Merge passes with all optimisations selected and 3 big elliptical Paths and two small circular paths in each had most cuts done continuously round and round except one elipse which went back and forth on each quarter segment instead. Only difference was that this elipse was first after an image raster.
2. Select several elements in File. Right click shows "Remove x selected" **and** "Remove x elements".

dpticar, 2021-09-08:
Hi,

When opening svg file in attachment (one rectangle 272.192 mm x 151.384 mm), meerK40t is showing size of 272.0mm x 151.2mm.

[rect1_rc12.zip](https://github.com/meerk40t/meerk40t/files/7132185/rect1_rc12.zip)

Sophist-UK, 2021-09-09:
Not sure where it went, but @dpticar reported: "When opening svg file in attachment (one rectangle 272.192 mm x 151.384 mm), meerK40t is showing size of 272.0mm x 151.2mm."

I have also seen these slight changes. I suspect it may be about how Inkscape measures object sizes (differently in different places, sometimes including 1/2 the stroke width and sometimes going on the node coordinates) compared to how MK calculates this.

However I did report an issue with stroke widths not being included in the dimensions and being cut off at the edges when creating raster bitmaps - and this is perhaps a similar issue.

Sophist-UK, 2021-09-09:
# Bug
1. I think this is a weird bug. With optimise ops and not optimise paths, it engraved a series of small holes, then  moved to raster an image extremely slowly, then started the image and the head literally jumped and banged into the end-stops. I closed MK and reopened and turn off / on the K40 and repeated and it did the same thing. Turned off Merge ops and it burned fine.

tatarize, 2021-09-09:
I checked through that and javascript didn't include their stroke widths in the size of objects.  So I'm fine saying things are that size. 

Though there is a minimum grab size for a 0 height object of zero making it impossible to move with the mouse. I did  `line 0 0 0 5in` to make a line then I couldn't move it with the mouse since the size to trigger those grabs didn't exist. I needed it for a ramp test but I had to move it with my numkeys.

Sophist-UK, 2021-09-09:
I agree that this is the correct definition, but MK nevertheless needs to take the stroke width into account when determining e.g. raster box sizes.

Grab boxes - I think that was always the case, but I can take another look at the code. It is supposed to prioritise move over resize handles anyway.

dpticar, 2021-09-09:
Hi,

Two days ago while playing with MK I think I noticed misreporting of width and height by 0.x mm. But I wasn't smart and saved that svg files.  Yesterday I tried to recreate that "bug" with rectangle, and when I thought I found it I saw it's stroke width, so I deleted post. 

Now I think I have managed to create svg which have inkscape_width=282,659, and MK is reporting 282.5 mm. Height is OK.
(note: all stroke widths are set to zero
[box_width_rc12.zip](https://github.com/meerk40t/meerk40t/files/7140133/box_width_rc12.zip)
)

Sophist-UK, 2021-09-09:
> I agree that this is the correct definition, but MK nevertheless needs to take the stroke width into account when determining e.g. raster box sizes.

SVGelements needs a separate set of raster bbox functions which take into account the stroke width. (Though you could probably replicate the Group bbox function in MK and make it do that.

tatarize, 2021-09-11:
I believe this to be accurate. The issue at the left most side with the little curves there. See path2128. This may be a bounding where the cubic bounds are being miscalculated. 

As svg does not include stroke-width in the bounds calculation, svgelements does not include this either. This seems to be a somewhat independent bug to anything dealing with stroke-width which is only used in MeerK40t as a display property and to match the WC3 tests for SVG.

tatarize, 2021-09-11:
I tried isolating this little section and it looked to basically be rounding error.

13.850 in inkscape 13.854947199464277 in svgelements. That's an error at the hundreds place for pixels and needs to be at the 1/10ths for mm. Length('3.665786133667mm') vs 3.664 in inkscape.

It looks like the spot since the line at the left side of the box_width seems like it goes through path2128.

Sophist-UK, 2021-09-12:
Suppose I have two Rect elements, with both stroke and fill, all of the same colour. One is 19mm in size with stroke-width of 2mm and the other is 12mm in width with a stroke size of 16mm. They look identical.

But if I align their left sides, they don't align visually as a user might expect.

Perhaps MK should do as Inkscape does, and use the external dimensions (including stroke) for purposes of bounding boxes and alignment.

dpticar, 2021-09-12:
I like more current type of dimension calculation. There is only one problem, for example if you have circle d=10mm with stroke 4mm, and you convert that to raster (in MK) it will crop it to 10x10 raster, while it should be 14x14, but you can avoid that by converting stroke to path (in inkscape).

tatarize, 2021-09-13:
@Sophist-UK if you wanted you could make align work in that way. But, svg in general gives the pure geometry and does not give the drawn render values.

@dpticar that is a solid point, if you render something with a fat stroke-width it could actually crop at the edge. It would, in that case need to correctly calculate the width ± stroke-width in order to be sure it captures the entire represented object as it appears on screen.

Sophist-UK, 2021-09-13:
@dpticar Fixing that at the moment.

@tatarize I understand the purity of the SVG spec, but MK is about practicality.

tatarize, 2021-09-16:
Took a bit and some changes like `transform` and `with_strokes` having some interactions since stroke width depends on the scale size, but was pretty solid upgrade overall.

tatarize, 2021-09-16:
```
MeerK40t crash log. Version: 0.7.0 RC-11 on win32
Traceback (most recent call last):
  File "meerk40t\svgelements.py", line 8028, in load_file
  File "PIL\Image.py", line 2968, in open
FileNotFoundError: [Errno 2] No such file or directory: 'data:image/bmp;base64, <image of car>
```
It's weird that it would try to open this since data:image/bmp is clearly a base64 bitmap. But maybe the hook is wrong and doesn't match the routine correctly. Also fun this crash log was like 3 megs.

```
MeerK40t crash log. Version: 0.7.0 RC-10s on win32
Traceback (most recent call last):
...
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```
Jos managed to run source and finished RC-10s and crashed with the local error.

* 1x normal local crash

```
MeerK40t crash log. Version: 0.7.0 RC-9 on win32
Traceback (most recent call last):
  File "meerk40t\gui\executejob.py", line 647, in on_button_start
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
  File "meerk40t\core\planner.py", line 757, in plan_optimize
  File "meerk40t\core\planner.py", line 71, in execute
  File "meerk40t\core\planner.py", line 151, in optimize_travel
  File "meerk40t\core\planner.py", line 1133, in inner_first_cutcode
  File "meerk40t\core\planner.py", line 1066, in is_inside
  File "meerk40t\core\planner.py", line 1066, in <listcomp>
  File "meerk40t\svgelements.py", line 3652, in point
  File "meerk40t\svgelements.py", line 3916, in point
  File "meerk40t\svgelements.py", line 4092, in npoint
AttributeError: 'NoneType' object has no attribute 'x'
```
X2 - Old version, though I don't really recall seeing much of this bug.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\mwindow.py", line 72, in on_close
  File "meerk40t\kernel.py", line 545, in close
  File "meerk40t\gui\mwindow.py", line 92, in finalize
  File "meerk40t\gui\wxmeerk40t.py", line 2173, in window_close
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\kernel.py", line 2456, in shutdown
  File "meerk40t\kernel.py", line 917, in shutdown
  File "meerk40t\kernel.py", line 1684, in process_queue
  File "meerk40t\gui\navigation.py", line 574, in on_emphasized_elements_changed
  File "meerk40t\gui\navigation.py", line 616, in select_ready
RuntimeError: wrapped C/C++ object of type BitmapButton has been deleted
```
I've seen this several times, not sure how it happens.

```
MeerK40t crash log. Version: 0.7.0 RC-11 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 1213, in open_simulator
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1788, in console
  File "meerk40t\kernel.py", line 1874, in _console_parse
  File "meerk40t\kernel.py", line 1227, in inner
  File "meerk40t\core\planner.py", line 717, in plan_validate
[... 239 lines cut ...]
  File "PIL\ImageOps.py", line 58, in _lut
OSError: not supported for this image mode
```
Must be a raster wizard mode error for autocontrast.

```
MeerK40t crash log. Version: 0.7.0 RC-11 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3780, in specific
  File "meerk40t\core\elements.py", line 1624, in inner
  File "meerk40t\core\elements.py", line 4366, in clone_single_element_op
TypeError: inner() takes 1 positional argument but 2 were given
```
Unsure. I don't recall fixing anything there. RC11

```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3793, in specific
  File "meerk40t\core\elements.py", line 1648, in inner
  File "meerk40t\core\elements.py", line 4707, in duplicate_element_1
TypeError: inner() takes 1 positional argument but 2 were given
```

2X - RC12 same sort of crash.

dpticar, 2021-09-17:
Hi,

There is some bug when loading black png images and using raster wizard. Full black is crashing raster wizard, and second one in attachment is cropped when using raster wizard.
[image]
[image]

tatarize, 2021-09-17:
Confirmed. Bounds box for pure black or white is None. And subscription fails.

```
MeerK40t crash log. Version: 0.7.1s on win32
Traceback (most recent call last):
  File "C:\Users\Tat\PycharmProjects\meerk40t\venv\lib\site-packages\wx\core.py", line 3407, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "C:\Users\Tat\PycharmProjects\meerk40t\meerk40t\kernel.py", line 1705, in process_queue
    funct(path, *last_message)
  File "C:\Users\Tat\PycharmProjects\meerk40t\meerk40t\gui\rasterwizard.py", line 225, in on_emphasis_change
    self.panel_select_op()
  File "C:\Users\Tat\PycharmProjects\meerk40t\meerk40t\gui\rasterwizard.py", line 266, in panel_select_op
    panel.set_operation(self.context, op, svg_image=self.svg_image)
  File "C:\Users\Tat\PycharmProjects\meerk40t\meerk40t\gui\rasterwizard.py", line 818, in set_operation
    self.refresh_dims()
  File "C:\Users\Tat\PycharmProjects\meerk40t\meerk40t\gui\rasterwizard.py", line 835, in refresh_dims
    top_left = matrix.point_in_matrix_space([box[0], box[1]])
TypeError: 'NoneType' object is not subscriptable
```

tatarize, 2021-09-17:
The getbbox code for a blank image is pretty easy but the other one is also of note. Namely we're dealing with applying the bbox to something that is *mostly* black and we want the margin. In that case we need to fully avoid using bbox at all. I'll have to do some research on this to see what other effects disabling that would have.

tatarize, 2021-09-18:
```
Image.getbbox()[source]
Calculates the bounding box of the non-zero regions in the image.

Returns
The bounding box is returned as a 4-tuple defining the left, upper, right, and lower pixel coordinate. See Coordinate System. If the image is completely empty, this method returns None.

This helps to get the bounding box coordinates of the input image:
```

0 is black and doesn't do this with white. So it might need flipping to use that function or dropping the use since it doesn't really do what it would be assumed to do. Might still need to look into this, though `.invert()` `.getbbox()` and `.invert()` would not be too absurd.

tatarize, 2021-09-18:
I must say @dpticar that is some top notch beta testing. Yeah, the black as 0 color is pretty common. You can take your MK example image and do an actualize on it and it will clip off the black, when really since white is non-op it should strongly prefer clipping white background. I'm still wrangling the code but it's pretty clear my use of that code was intended to crop off the white edges and did not do that, and could very much result in crops that were neither expected or desired. Such as doing a 1 inch square stamp and getting the box you were expecting to raster shrunk in error.

I think I'm just down to the imagetools malfunction, but there's a couple of these to get this right.

tatarize, 2021-09-18:
This set of fairly interesting errors should be fixed in #570. 

https://github.com/meerk40t/meerk40t/commit/0e76b38cff5b2d63e2679e3b7ca162d13facf95b

Unified the actualization code between planner and imagetools, and updated the code to the most modern version of both.

--
Also, corrected the label not forced updating on altered call.

dpticar, 2021-09-18:
Hi,

Now I'm doing some research on Lhymicro code so I wanted to see how engrave code will look like on black raster image and found that bug. Last week I was cutting/engraving  wooden stamps for school where I work and haven't noticed any bugs, but I used svg (with embedded images), not pure images.

Anyways I must say that MeerK40t is really fantastic piece of software.

tatarize, 2021-09-18:
I did see you found a pretty obscure but in `laserspeed.py`. Bugs are getting rarer. Hunting down a couple other small ones that have churned up. The 0.7.x should just be bug fixes from here on out. Enhancements and other changes will go in the 0.8.x line.

Sophist-UK, 2021-09-22:
# Bug
1. Remove operation for disabled operation is `Remove ''`

tatarize, 2021-09-22:
1. yes, that should be fixed.
2. yeah, that seems like it might be in error
3. I dunno what you mean by removes the existing file and adds a new one, are we talking looking at what happened on disk or in the tree or what?

Sophist-UK, 2021-09-23:
> 3. I dunno what you mean by removes the existing file and adds a new one

Suppose I load an Inkscape SVG file. When I click Ctrl-S MK asks for a filename because it is not an MK SVG file.

If I do File/New and then reload this saved SVG file, and click Ctrl-S, it saves without asking for a filename.

If I then remove the file-node and all elements (because I want to reuse the operations with another file, and then load a new file, when I do Ctrl-S, it overwrites the previous saved file rather than asking for a filename which I would like it to do.

tatarize, 2021-09-24:
```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\device\ch341\ch341libusbdriver.py", line 332, in CH341EppWrite
  File "usb\core.py", line 994, in write
  File "usb\backend\libusb0.py", line 549, in bulk_write
  File "usb\backend\libusb0.py", line 686, in __write
  File "usb\backend\libusb0.py", line 447, in _check
usb.core.USBError: [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: The device does not recognize the command.\r\n'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1478, in run
  File "meerk40t\core\drivers.py", line 124, in _driver_threaded
  File "meerk40t\core\drivers.py", line 141, in _process_spooled_item
  File "meerk40t\core\drivers.py", line 238, in command
  File "meerk40t\device\moshi\moshidevice.py", line 611, in unlock_rail
  File "meerk40t\device\moshi\moshidevice.py", line 822, in unlock_rail
  File "meerk40t\device\moshi\moshidevice.py", line 764, in realtime_freemotor
  File "meerk40t\device\moshi\moshidevice.py", line 784, in realtime_pipe
  File "meerk40t\device\ch341\libusb.py", line 82, in write_addr
  File "meerk40t\device\ch341\ch341libusbdriver.py", line 411, in CH341EppWriteAddr
  File "meerk40t\device\ch341\ch341libusbdriver.py", line 338, in CH341EppWrite
ConnectionError

```

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3709, in specific
  File "meerk40t\core\elements.py", line 1416, in inner
  File "meerk40t\core\elements.py", line 3387, in operation_property
  File "meerk40t\kernel.py", line 472, in open
  File "meerk40t\kernel.py", line 508, in open_as
  File "meerk40t\gui\mwindow.py", line 88, in initialize
  File "meerk40t\gui\operationproperty.py", line 172, in window_open
  File "meerk40t\gui\operationproperty.py", line 452, in on_size
  File "meerk40t\gui\operationproperty.py", line 458, in refresh_display
  File "meerk40t\gui\operationproperty.py", line 505, in refresh_in_ui
  File "meerk40t\gui\operationproperty.py", line 469, in calculate_raster_lines
ZeroDivisionError: division by zero
```
4x RC6.

```
MeerK40t crash log. Version: 0.7.1 on win32
Traceback (most recent call last):
  File "meerk40t\gui\panes\consolepanel.py", line 117, in on_entry
  File "meerk40t\kernel.py", line 135, in __call__
  File "meerk40t\kernel.py", line 1791, in console
  File "meerk40t\kernel.py", line 1877, in _console_parse
  File "meerk40t\kernel.py", line 1230, in inner
  File "camera\camera.py", line 184, in set_camera
cv2.error: OpenCV(4.5.3) C:\Users\runneradmin\AppData\Local\Temp\pip-req-build-hi4qs98u\opencv\modules\videoio\src\cap.cpp:413: error: (-2:Unspecified error) in function 'bool __thiscall cv::VideoCapture::set(int,double)'
> Can't set read-only property (expected: 'propId != (int)CAP_PROP_BACKEND'), where
>     'propId' is 42
> must be not equal to
>     '(int)CAP_PROP_BACKEND' is 42

```
Pretty sure I did this with a weird set.

```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "wx\core.py", line 3285, in <lambda>
  File "meerk40t\gui\scene\scene.py", line 426, in refresh_scene
  File "meerk40t\gui\scene\scene.py", line 454, in update_buffer_ui_thread
[... 107 lines cut ...]
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3793, in specific
  File "meerk40t\core\elements.py", line 1648, in inner
  File "meerk40t\core\elements.py", line 4429, in clone_single_element_op
TypeError: inner() takes 1 positional argument but 2 were given
```
X6

```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1478, in run
  File "camera\camera.py", line 554, in threaded_image_fetcher
  File "camera\camera.py", line 459, in _attempt_recovery
cv2.error: Unknown C++ exception from OpenCV code
```
X2

```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\gui\panes\camerapanel.py", line 327, in <lambda>
TypeError: %d format: a number is required, not builtin_function_or_method
```

tatarize, 2021-09-25:
@Sophist-UK I tracked down the blank remove bug:

label_truncate_re = re.compile("((:|\().*)") actually takes "(Disabled) Cut 10mm/s 1000ppi #ff0000" and makes it "".

It strips everything.

tatarize, 2021-09-26:
* Implement #560 for 0.7.x. Disable sleep while running.
* All contexts are getting a `path` value that is blank. For alias this means there is an alias path that returns blank. This blocks the call to the command `path`.
* Typing `alias path` does not clear the path, it sets that path to blank.
* Typing alias path path` sets the alias `path` to `path` which causes an infinite loop if you then type `path`.

tatarize, 2021-09-28:
Review Review.
Searching all the previous reviews for versions.

A couple things (very minor) fell through the cracks and were not addressed at all.

Most of the remaining list of potential todos, were already moved to Project 1.

# Might not have been correctly addressed

ampostata commented on Jun 17
I dont know :) I just draw (and see) a single line and saw 2 cuts, so I inform :)

This was not fully tracked down and solved. It was dismissed, perhaps in error. We have another report of a similar thing coming in from Facebook.

---

Never Reviewed:

> Execute Job - Merge options need overheating warnings in tooltips e.g.
> For Merge Operations: "Operations will be merged before optimisation so that e.g. raster burns may be interspersed with vector burns. If you have Duplicated Operations (with identical paths), please be aware that these paths may be burned in rapid succession and this may cause much higher material temperatures than if operations were executed sequentially."
> 
> For Merge Passes: "Rather than executing one pass followed by a second separate pass, this allows e.g. the same path to be burned repeatedly in rapid succession, and this may cause much higher material temperatures than if operations were executed sequentially."

# Moved to Project/1  or pushed to later versions

---

- [ ] Optimization: If greedy optimization did not save time, it should be undone. Prevent Optimization from making slower solutions.
- [ ] Optimization: Rasters can be cut out of order but they should count as always inside anything any cut.
- [ ] Simulation: Simulation speed is unrelated to Burn Speed.
- [ ] 0.6.xx: Permits you to recreates items in order when you toggle automatic settings without requiring reopening ExecuteJob.
- [ ] Estop: Estop in 0.7.x seems to prevent homing immediately afterwards with as much success as with 0.6.xx
- [ ] Estop. TCP Estop does not purge the TCP buffer, it's entirely possible to estop and then create additional lhymicro-gl code which is then wrongly sent to the laser, while in the wrong mode.
- [ ] RasterWizard: Upgrade RasterWizard main view to using Scene/Widget.

 ---

Laser apparently not communicating - turned out that MK was paused but there was not any indication in the UI that this was the case.

---

> Can we have an option to save the compiled/optimized/you-name-it generated in the spooler? (and load it also in the spooler window), so my so long "optimizations" can be saved optimizated :) something like a "raw" file.

Extremely cursory work done towards this, put into developer mode. Might fix it if there's time. plan return as a command is intended to do this work.

---

> Bed dimensions should be part of the device configuration IMO not a global Setting

Joe agrees with you. And I also tend to agree I'm planning on switching it after the full release. It needs some slight related changes and switch to steps from mm.

---

> SVG files are still not saved with groups and friendly names.

Yeah, not sure if that will make this version or end up in the next. Requires a number of saving changes.

---

> We probably need an ability to interrupt / cancel a long running optimisation (either for Execute or Simulation).
Likely push to later version.

---

> (Fix > 0.7.0) Need to be able to change Execution Optimisation etc. options from the simulate window and replan/show new simulation. At present you have to exit simulation, open Execute, change options, then reopen simulate.

Entirely accurate but likely should be pushed for a bit. The two things for execution and simulation themselves likely need to be combined.

---

> Simulation - close simulation window when maximised, next time you open it then it is not maximised but instead is Restored slightly bigger than maximised.
This would actually be true for most every window. It preserves size and does not preserve state.

---

> Tree/Operations/Add Operation adds an unknown operation. UI would be better with a submenu for Cut/Engrave/Raster/Image.

Still not fixed.

---

> Incompatible endpoint types should not be permitted. Grbl device being hooked to a Moshi backend crashes. But, sort of should crash. That's not a device that makes any sense.

Really they should have larger device integration and have lhystudios as a full device. This isn't fully fleshed out even for the longer-term goals etc.

---

> Dots are not shown on Scene - they need a special symbol IMO.

This is probably still rather desirable. Not sure how often people are using dots though.

tatarize, 2021-09-28:
# Crash Review
```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\svgelements.py", line 8028, in load_file
  File "PIL\Image.py", line 2968, in open
FileNotFoundError: [Errno 2] No such file or directory: 'kun4_obr_Images\\kun4_obr_ImgID1.png'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2866, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2573, in load
  File "meerk40t\core\elements.py", line 5742, in load
  File "meerk40t\core\svg_io.py", line 293, in load
  File "meerk40t\core\svg_io.py", line 355, in parse
  File "meerk40t\core\svg_io.py", line 355, in parse
  File "meerk40t\core\svg_io.py", line 338, in parse
  File "meerk40t\svgelements.py", line 8001, in load
  File "meerk40t\svgelements.py", line 8035, in load_file
AttributeError: type object 'Image' has no attribute 'open'
```

An SVG with a linked file that did not exist caused a FileNotFoundError. 

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3693, in specific
  File "meerk40t\core\elements.py", line 1459, in inner
  File "meerk40t\core\elements.py", line 3876, in break_subpath_elem
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\core\elements.py", line 1996, in element_subpath
  File "meerk40t\core\elements.py", line 595, in replace_node
  File "meerk40t\core\elements.py", line 405, in add
  File "meerk40t\core\elements.py", line 703, in __init__
AttributeError: 'NoneType' object has no attribute 'node'
```
Old.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1450, in run
  File "meerk40t\core\drivers.py", line 122, in _driver_threaded
  File "meerk40t\core\drivers.py", line 133, in _process_spooled_item
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 812, in plotplanner_process
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 989, in goto_octent_abs
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1006, in goto_octent
ValueError: Not a valid diagonal or orthogonal movement. (dx=-1, dy=2)
```

Old.

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 685, in <lambda>
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\core\spoolers.py", line 434, in home
  File "meerk40t\core\spoolers.py", line 150, in default_spooler
AttributeError: 'Context' object has no attribute 'active'
```

Old.

```
MeerK40t crash log. Version: 0.7.0 RC-12 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3793, in specific
  File "meerk40t\core\elements.py", line 1648, in inner
  File "meerk40t\core\elements.py", line 4707, in duplicate_element_1
TypeError: inner() takes 1 positional argument but 2 were given
```
X6 - Fixed but unpublished. -- Six of these. Should probably push the newer version out.

tatarize, 2021-10-01:
Add wx.version() (if it exists it might not) to the crash log.
`wx.version()` can be useful in things like #242 where it's properly a wx issue. Also it might be needed to see if the snapshot fixes work to correct things.