meerk40t/meerk40t #560: 0.6.x: Prevent Sleep During Data Send

tatarize, 2021-09-12:
In windows, `ctypes.windll.kernel32.SetThreadExecutionState(0x80000001)` prevents the computer from sleeping and `ctypes.windll.kernel32.SetThreadExecutionState(0x80000000)` restores the ability. These should put into 0.6.x during the start and end of data sending to prevent and restore sleepmode. So a laptop doesn't turn off and wreck a job.

See Chris Provazek's comment (https://www.facebook.com/groups/888394261237059/permalink/4257387297671055/ ): "I had a few cases of it failing half way in a job on me but It may have been my laptop falling asleep and I've been hesitant to try anything else with it again"

Sophist-UK, 2021-09-12:
Also needs to be done in 0.7...

tatarize, 2021-09-12:
Sure, but things in 0.7 are a dime a dozen. It's rare something might be needed for 0.6.x

Sophist-UK, 2021-09-12:
LoL. Version 0.7 is literally a list of all the things that were needed for 0.6.x - a list that was so long and extensive, that you decided you needed to rearchitect and start again from scratch!!

tatarize, 2021-09-12:
Not really, there's a lot of overlap with version 7 and version 6 had some notable architectural differences.  It used to have a sort of kernel for the main part of the program and each of the devices. And each device would associate with a new window so if you had two devices you'd get two main windows and they'd weirdly sync together since they were showing the same thing. And the console rather than stored in a centralized place was its own file, and needed direct access to all the stuff it was messing with. And some of the gui windows like for planning did actual work making it impossible to use the CLI to do those things.

There was a pretty significant reengineering of how the central bits of code worked.

tatarize, 2021-09-12:
Also, in 6 it would just call `ctypes.windll.kernel32.SetThreadExecutionState(0x80000001)` and in 7 it would get put in meerk40t/extras and called with and as a console command. Look all tidy and give the user access if they somehow wanted to call those things themselves for some bizarre reason. `sleep_suspend` and `sleep_restore` as hidden functions. This would also let them be accessible for pure command line access, whereas in 6 it would only be a gui feature since they are more tightly stuck together.

tatarize, 2021-09-18:
https://github.com/meerk40t/meerk40t/pull/571

Should be done. I still need to test it though.

tatarize, 2021-09-19:
Tested. Worked.

0.6.24 is fine for a merge.

closed 2021-10-04 by tatarize