meerk40t/meerk40t #593: Why can I just do a raster operation?

tomislav17, 2021-10-04:
Why can I just do a raster operation? v071

Sophist-UK, 2021-10-04:
I think a bit more information than this might be helpful. Like screen shots and possibly an upload of your SVG file.

Sophist-UK, 2021-10-04:
And do you mean "can't execute just a raster operation"?

tomislav17, 2021-10-04:
> And do you mean "can't execute just a raster operation"?

https://user-images.githubusercontent.com/61805701/135907797-3c0b5a73-6195-4c6b-8457-9798df58d061.mp4

[kocka 10x10.zip](https://github.com/meerk40t/meerk40t/files/7280616/kocka.10x10.zip)

Sophist-UK, 2021-10-04:
You need to keep clicking the big button until the window closes.

If you want to Execute just the Raster operation, then select just the Raster Operation in the tree, then Right Click and select Execute.

@Taterize: Do you think we should reduce the number of clicks needed to Execute?

tatarize, 2021-10-05:
In panes there's a go button. You can get that and if you're confident that everything will go fine hit it and it'll just go.

tatarize, 2021-10-05:
There are some questions as to what the best interface might be and what you'd be best off using. There's a lot of panes and this might hide things that maybe users would want to use more frequently but that sort of UX and UI work isn't clear as to what kind of default would be best.