meerk40t/meerk40t #594: Datastructures and laser-rendered objects such as slot-tab.

tatarize, 2021-10-06:
There has been some discussion of non-svg objects in Meerk40t previously: see #24 for a multi-pass raster object, and #218 for gradients for bevelled edges. There are other laser specific object types like living-hinges that could be useful for design objects. The slot-tab edges for laser projects are based, in part, on material thickness. Which means the edge needs to re-generated for different thicknesses of material. If this information was stored and processed we could define a edge as a slot-tab edge and re-render that edge based on a material thickness entered.

Currently all the main objects within the tree are svg style objects. These are svg-shapes and svg-image objects with slight increases in metadata.  There are some cutcode objects that get added processed there which also contain laser settings. Some lasercode objects which provide even broader data structures, and blob data which is pre-compiled data for a particular laser type.

Implementing renderable data that would be converted to a specific type of path or provide particular laser-ready functions would require that our loading and saving format provide this additional information. In accordance with the need to implement fullsave these topics certainly call for discussion. Should we perform these sorts of laser rendered objects. These rendering objects exist in the realm of embroidery and a long term goal is to spin off some of the functionality here and make an embroidery application, so the difficulty here aligns *really* closely with other goals, such as defining an object as having an embroidered fill or defining an object as a satin-stitched shape or a particular fill motif. While different, there is notable overlap with the required concepts.

So, what sorts of planning steps should be made to make this possible? Should this be adapted into more-or-less SVG as current meerk40t files are processed currently and expected to be processed within `Project: FullSave`. Should this be deemed outside the realm of features and ignored. Does this require too much design and creation infrastructure (which is currently in the long term plans)?

Sophist-UK, 2021-10-06:
This seems to me to be another example of something that needs to be generalised.

One of the real strengths of Meerk40t v0.7 is the generalisation of everything into plugins - with a lifecycle and a way to register functionality. We need to extend this into the Elements and Operations parts of the tree - with objects representing leaves and generalised methods for MK to get what it needs from these objects. So for example we can have Laser or Special objects in the Operations part of the tree, yet these are not subclasses of a common leaf object and are treated differently using conditional code rather than by calling the same method in different subclasses of an object. (Not sure how well I have explained this.)

Similarly we need to generalise all the stuff about PPI and bed size and speeds etc. into the Driver objects in the same way that converting CutCode to actual code is generalised.

In other words, do more of what MK already does.

tatarize, 2021-10-08:
The problem with those elements is that some of them do not actually fit well within SVG as a spec.

I think #598 offers some reasonable reasons for the shortsided problems here with svgelements being used natively in that space. While it works really nicely to load things and it saved a lot of development time it's a bit of an albatross around our necks in places. While this stuff can have some overlap with other code elements, I'm thinking `FullSave` might want to consider saving to its own format and merely `export to SVG`. Not that things don't work as they currently are, but there's no real restriction on what sorts of objects our nodes tree nodes can have and some stuff just needs to keep a lot more information than it currently does and that information is *absolutely* not a part of the SVG spec, and trying to shoehorn it in there may be in error.

Cutcode currently includes a few different elements that don't match spec either like Raw which merely consists of raw pixel movement values. Not lines or anything but a list of points to go and whether that point needs a laser strike or not. It might be better to just write element types that properly correspond to our needs rather than trying to make SVG serve as that purpose.

Fullsave would currently require things like laser-data-blobs and special-commands, and loading and saving of cutcode, and currently has operations save and load. These might be more easily done with more native saving and loading of objects, in ways that better reflect the modern direction of the program, but also directions that help with #577 etc. Since there's a future goal of some kinds of embroidery in a sister application and things like closed objects with fills and properties and satin stitches might also be better stored as their own data-classes too.

These wouldn't need to replace element nodes either, you could still have more simplistic versions of those nodes. But, it might be better to also have more robust objects if we could do that in a reasonable and consistent manner. So if you wanted a tab-slot pathsegment you would need to use a CutVector (or whatever the name ends up being) instead of a path. And just easily convert between the two. Since the types of data don't matter as much when there are plenty of simplified methods to make these bits easier. 

It's built to be super-hackable and this is a way we could sort of hacking it ourselves. We do not need svgelements in cases where we're left to hack the data to try to shoehorn it into something we care about.

jpirnay, 2024-10-04:
Too old to revisit

closed 2024-10-04 by jpirnay