meerk40t/meerk40t #599: Raster wizard introduces black lines at right hand / bottom edge

Sophist-UK, 2021-10-08:


Sophist-UK, 2021-10-08:
I tried it on my laptop running from source and I didn't get this - so probably fixed by #597.

But bad image processing is for me getting to the point where 0.7.1 is unusable.

tatarize, 2021-10-09:
The PR #600 and #601 both seek to address this problem. I think there are some merit to both. But, some clear interference. #601 seeks to make rasters uniform. All the raster stuff is transitioned to using dewhite to make the preview images. So all images loaded and viewed in meerk40t are converted to black/transparent images. Actualize makes the images actually black and transparent. It should also reduce bounding box if needed and allow some other nice changes. But, the most important thing is that it should get rid of any and all whack-a-mole style errors here.

* Has not been extensively checked with raster wizard.
* May avoid applying to an image from execute-job if it matches the old criteria but should actually trim and modify the image.
* Has not been tested for inverted images and inverted images might prefer being white/transparent images
* Does harder alterations of the current errors.

#600 
* fixes the actualize pixel code call, but may be overly harsh in recalculating the bounding box.
* includes a burn value in actualize that is not triggered.
* Burn should not function differently than not-burn. 
* Mode P for pallette might have an alpha. I forget exactly how that works. This would mean new_mode.endswith("A") could fail.
* Comment is wrong, Pillow does allow to fill in new areas created by transform with transparent but only when that mode is RGBA and is the default in that case. The problem seems to be that it's transparent black so converting it to L converts that color to black.
* Probably slower since it holds the alpha and image and applies the same to both, likely bbox is also slower.

Sophist-UK, 2021-10-09:
As stated above, #597 seems to fix this problem. However responding to these comments:

1. Both #600 and #601 seek to make rasters uniform. I just wasn't brave enough in #600 to go to greyscale "L" on actualise rather than keep the image coloured "RGB". But on balance I think that going to grey scale is a good idea on actualize. But I think my approach to keeping alpha i.e. end up with "LA" is a better UI approach for user Actualization of source elements - whereas for simulation we should ideally switch to black with alpha for greyscale because that visibly simulates burns, and for burns we should switch to greyscale with no alpha because that is simplest for processing.

> Burn should not function differently than not-burn.

If we go to greyscale uniformly, then we don't need to treat burn as different.

I was thinking last night about how to do inverted images for the RasterScript greyscale invert. The issue here is how to handle alpha / white when these pixels could be a background border or an internal area. 

On reflection as I type this, I am beginning to wonder whether we shouldn't settle on black plus alpha as greyscale as the uniform standard because this is the best simulation of laser action.

2. Neither #600 nor #601 yet address inconsistencies in use of `actualize` for e.g. simulation / Raster Wizard.

> Comment is wrong, Pillow does allow to fill in new areas created by transform with transparent but only when that mode is RGBA and is the default in that case. The problem seems to be that it's transparent black so converting it to L converts that color to black. 

Agreed - though the concept is the same, and the solution is the same.

> fixes the actualize pixel code call, but may be overly harsh in recalculating the bounding box.

I definitely considered whether the bounding box was overly harsh, but I couldn't find an example where this was not the case.

Sophist-UK, 2021-10-09:
P.S. I think we need to have a single place to comment on #600 / #601 as having three places is confusing as heck.

tatarize, 2021-10-09:
Five if we count #598 which has very considerable overlap with #601. Since #601 is a simplified version of the idea in #598 without the special class. In fact, it might be even simpler, since it converts the mask and image into a single value. Harping on black-transparent values.

Sophist-UK, 2021-10-09:
Yes. And I could open another PR which doesn't split out the alpha channel at all, but instead does a transform with an RGBA colour of `#ffffff00` if there is already an alpha channel.

I think if we combine all the comments we can now come up with a great solution.

tatarize, 2021-10-09:
I think #601 has a lot of great things. And I'm fixing up the last bugs now.

tatarize, 2021-10-09:
The actual great thing about #601 is that it doesn't use an alpha channel. It uses the whiteness of the image as the alpha display value. But, the actual thumbnail treats white like alpha. So you see images as black and transparent. But, they are 8 bit white and black images. This does what you had suggested a long time ago, that I implemented in simulation. Even things that are full RGBA look like black and transparent. They aren't actually converted.

The processing of the image will necessarily run it through through the actualize routine which will force it into grayscale at that point but not before. So actualizing it with a right click, or actualizing it when you need it actualized for ExecuteJob or when actualized in raster wizard. It will end up grayscale at some point. The only time this might not happen is if you have an image already at the native resolution and it doesn't require actualization to send it to the laser. But, in that case rasterplotter can deal with those other image values without issue.

tatarize, 2021-10-09:
[image]

tatarize, 2021-10-09:
oof, it's a but in actualize. Really shouldn't have messed that up. Oh well. Likely easy fix, probably screwing up the default inversion value.

tatarize, 2021-10-09:
Found a knock-on error, fixed it. Added in the alphablack as a view setting.

Sophist-UK, 2021-10-09:
> The actual great thing about #601 is that it doesn't use an alpha channel. It uses the whiteness of the image as the alpha display value. But, the actual thumbnail treats white like alpha. So you [...]

I don't see why we cant do that for images - obviously we cannot do that for Shape fill.

> The processing of the image will necessarily run it through through the actualize routine which will force it into grayscale at that point but not before. So actualizing it with a right click, or [...]

That sounds good.

> I think #601 might require additional testing here and there.

I will give it a test later this evening.

> Selecting RasterWizard, DPI does not shift the DPI value. Only the Step in Resample forces the change. (I think this predates the changes)

DPI is raw K40 DPI / step - I am unsure why we have a dropdown (though I think it is worth calculating and displaying the DPI for the user's info.

> Simulation appears to crop things wrongly. This might be actualization creating a bug or something.

I will take a special look at that when I test.

> Images with white inset (non-edge touching) in a black field got cropped to just the black pixels when simulated. This also went _very_ slow. (Fixed the inset, can't duplicate the slow)

> I might actually want regular tests for this, though it's a critical error and we might want to push things out sooner rather than later. I also sort of want to get ruidabounce working too. It's really close and I think I only need to force queue some of the lightburn moves to get it working. Modern RDWorks requires knowing laser response that I do not know.

You need a ruida board to eavesdrop when driven by RDWorks. I assume you have researched the reverse engineering already done by others.

tatarize, 2021-10-09:
@Sophist-UK 
> Do all image actualises (inc. RasterScript and Simulate) use actualize (because I don't see make_thumbnail using it yet)?

Thumbnail doesn't. But, it doesn't use anything else either. The job of thumbnail is supposed to create a wxBitmap of the class to view. This is not a modification of the image, it's actually a customized view of the image.

> I think we should definitely aim to have a single routine to actualize images and not try to keep code aligned.

Yes. I think everything is using actualize. There's no other method that gets you from non-processed image to processed image.

> Also, scene background will always be white so that burns show as black on a white background, however e.g. tree mini-me icons could be against a dark background if we implement a dark-mode. Does your new code plan for this eventuality?

It might be possible to switch to a white/transparent view. The important thing is thumbnail does not actually modify the image it only views the image. It's turning image into a wxBitmap. So that can be whatever sorts of views you want.

---

There is also a turn the blackalpha off for views, so you can see what they look like (images cache early and you might need to reload them) but you can tell they aren't modifying things in any weird ways or anything *most* of the time. And when they do the images are actually black&white.

I added a bunch of tests to prove the lack of flawed edge pixels.

tatarize, 2021-10-09:
[image]
[image]
[image]
[image]

Test images. Pure black, pure white, MKBlack, MKWhite. The goal here is to show that we can correctly crop the images to the right size. These are small, not sure why they scaled up so weirdly.

tatarize, 2021-10-09:
> I will give it a test later this evening.

Thank you. I think it should be close but it's a bit spooky. I added in some tests so I can show that it's not developing edge lines. Though also it now has an `inverted` flag that will make sure the edge lines are the right color if it needs to add in an edge line (sometimes due to math this is required, it's the scanline equivalent of a remainder).

> DPI is raw K40 DPI / step - I am unsure why we have a dropdown (though I think it is worth calculating and displaying the DPI for the user's info.

I should have it fixed. The combo box just wasn't altering stuff.

> I will take a special look at that when I test.

I should also have that fixed too.

> You need a ruida board to eavesdrop when driven by RDWorks. I assume you have researched the reverse engineering already done by others.

I know, inpain has a ruida so it should be able to be determined. Also, I'm talking with folks using it for a commercial project and they are doing some reverse engineering and talked with me as I've done the most extensive reverse engineering of anybody. Most of the other reverse engineering is much weaker than mine. They were going to give me an answer on Monday or so. Inpain did a wireshark to explain how you respond with a filename on the board, which was needed to get those files working. (this actually works in the ruida emulator too, it queries what .rd files you have locally and streams one to your device if it's correct). Without the missing response, RDWorks gets a communication error and only Lightburn or older RDWorks versions will work. Which is a bit meh, but good enough. I think mostly Lightburn just needs a small tweak to do "rubberband outline". It sends a series of moves in quick succession, due to the implementation of `move` in console this moves to the first location then says it's busy and refuses the other movements. It should be rather minor. The rastering works but does a sort of "hard-raster" which isn't great but works.

Sophist-UK, 2021-10-09:
I recommend that we do NOT pass un-actualized images to the device. The whole point of switching to "L" was to reduce the number of different types of images that we need to handle downstream in order to reduce the bug-surface-area, and letting un-actualized images through to the device to handle does exactly what we are trying to eliminate.

tatarize, 2021-10-10:
Yeah, it's notably a bit riskier, though to pass to the backend they would basically need to qualify which means they have skewing and they are already the perfect size. So scaled exactly in accordance with the step-size. There are a few unknowns which is the problem. 

It *should* be fine, but you're right. Those aren't unknowns worth dealing with. It isn't just a matter of switching to L for example if we have an alpha channel we need to treat that like white and some conversions strip the alpha channel which means the color of the pixels being affected matters. I think those are all dealt with correctly, but again, applying actualize consistently would generally correct it. And so long as actualize doesn't do unneeded work, everything should be perfectly safe. We could even specify that images as part of cutcode are type only of type `L` or type `1`.

This has the knock-on effect of making sure that if something else is needed for a different type of laser it will be well defined.

tatarize, 2021-10-11:
* Rightclick on image "Actualize" does not correct bounding box. (May have been fixed by now closed #600)
* Rightclick on image, initially gives commands for group requiring ungrouping before manipulating the image.
* Rightclick on image, does not allow image remove.