meerk40t/meerk40t #610: Other Images incorrectly removed from Image Op

Sophist-UK, 2021-10-16:
When I delete an image from the attached file, several images are removed from the classification to Image Op.

[Foam tag LASER7.zip](https://github.com/meerk40t/meerk40t/files/7357957/Foam.tag.LASER7.zip)
1. Load the file, deselect everything. 
2. Select one image on the right using the blue selection rectangle. 
3. Click delete. 
4. Check which images are still in the Image Op.

tatarize, 2021-10-17:
Confirmed.
Deleting a single image from the scene can remove opnodes for related but non-selfsame image.

tatarize, 2021-10-19:
I'll debug it, since proper bugs found should sort of prevent things from going from beta to a full release.

Sophist-UK, 2021-10-19:
I guess python's `in` using `==` makes sense in general terms (for example comparison of floating point numbers).

I have checked the SVG file - and every image is individually provided, but identical, but in this case with different transforms. But suppose two images were on top of each other and had the same transform - should they be considered equal or different? This is a bit of a philosophical question, but it does impact the implementation. I do wonder whether SVGImage.__eq__ should simply compare object ids rather than data or whether an SVGImage object should be assigned a random internal code, which is copied when an image is copied, so we can distinguish between an image which is duplicated (internal codes equal, data equal, transform equal) and identical images from the SVG file (internal codes different, data equal, transform equal).

Perhaps python should allow `object is in list` as an alternative to `object in list` so that you can check using `is` and not `==`.

tatarize, 2021-10-19:
When you sign off on #614, just go ahead and merge it. I think we're probably good for another release either a beta or 0.7.3 (assuming none of my other changes to RasterCut are still problematic).