meerk40t/meerk40t #651: [0.7.4b1 Windows] Crash on startup

Sophist-UK, 2021-10-30:
```
MeerK40t crash log. Version: 0.7.4-beta1 on win32 - 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "meerk40t\gui\scene\scene.py", line 280, in on_paint
  File "meerk40t\gui\scene\scene.py", line 456, in update_buffer_ui_thread
  File "meerk40t\gui\scene\scene.py", line 539, in draw
  File "meerk40t\gui\scene\scene.py", line 795, in draw
  File "meerk40t\gui\scene\scene.py", line 795, in draw
  File "meerk40t\gui\scene\scene.py", line 792, in draw
  File "meerk40t\gui\scene\scenewidgets.py", line 962, in process_draw
AttributeError: 'Context' object has no attribute 'units_convert'
```

tatarize, 2021-10-30:
Drat.

Sophist-UK, 2021-10-30:
At first sight, doesn't look like a hard one to fix.

But if you had an executable from each PR / commit, then you could have run a quick test on the executable before creating the beta. **Just saying.**

tatarize, 2021-10-31:
It's not it's just a little silly. When you turn your grid off it uses the guide as first thing to check thing and it doesn't intro units_convert and therefore crashes. I moved the units to the main part of scene to a Choices so they all get registered correctly for all subelements there.

tatarize, 2021-10-31:
Fixed in #652

Sophist-UK, 2021-10-31:
Ah. Well thanks for fixing anyways.

tatarize, 2021-11-01:
Merged and added all good.

closed 2021-11-01 by tatarize