meerk40t/meerk40t #655: Optimise greedy improvement idea: Merge operation passes

Sophist-UK, 2021-10-31:
1. Rather than have permitted as a boolean, have burns_remaining as an integer
2. When Merge Passes is **not** checked, work as currently, i.e. iterate into separate cutcode objects, with burns_remaining = 1.
3. When Merge Passes **is** checked, only put cutcode-objects in once but with burns_remaining = passes.

The idea here is that merge passes does NOT increase the number of objects to sort / search through.

I will give this a go tonight.

Sophist-UK, 2021-10-31:
P.S. I don't think this will impact anything if greedy is on (because it will produce multiple copies) but not sure what happens if user sets greedy off in console with a `set`.

tatarize, 2021-10-31:
This actually quite clever and quite actionable. This will also not duplicate out the data in memory or cost any of that time which is beginning to be the larger timesuck. Candidate() just means that everything inside the the routine has reached zero, the number of next and previous links will be reduced as will the value of N for anything of any time complexity.

This will also speed things up in another way that might not be as desirable but would certainly be the case. If you're going to 10 circles as your job, and these all perfectly overlap because they are 1 circle with 10 passes. We'd store this as a linked set of 10 circles and the cuts they break down into namely some cubic rasters, then the object themselves will be 1 object and consequently the loops the next and prev connection have will be used for additional passes bypassing the distance check altogether. Making the fastest method, basically by definition doing all 10 passes in constant time and no searching.

This is *not* what you wanted with your leather since it gave things the chance to burn however, this tweak opens up *another* change. We can implement merge passes into something closer to what you originally wanted without any any issues. What we can do is level the cutcode. We do not permit elements to be candidates *unless* their burns_remaining is the largest number of burn remaining.

So if there are anything with 5 burns left, and nothing with 6 only things with 5 burns left are candidates for inclusion into the plan.

Sophist-UK, 2021-11-17:
Implemented in #677.

closed 2021-11-17 by Sophist-UK