meerk40t/meerk40t #667: Stroke-width is not accounted for in make-raster size of images

tatarize, 2021-11-05:
https://www.facebook.com/groups/716000085655097/permalink/952890345299402/

Seems to arise in the wild where people would like to raster their shapes correctly.

Sophist-UK, 2021-11-05:
I will get around to fixing this when I get the time - it is not difficult, esp. since I added the functionality to SVGelements to include the stroke width in the bbox.

This needs to be done for:
a. Burn prep (simulation and actual burn)
b. Right click and convert raster elements to image
c. Console command to create raster image.

There should also be a consistent way of determining the bbox of selected elements (from memory several ways are used in different places).

Because it is in the same code, I was also saving this up for doing with optimisation of raster elements to break raster elements into multiple images if they include large amounts of whitespace and would be quicker done in several smaller images.

tatarize, 2021-11-05:
The make_raster command in line 504 in planner and 3068 in elements is being fed the bounds to make the rasters at. Usually by some Group.union thing.

You made group union accept with_stroke=False.

Sophist-UK, 2021-11-05:
It was about 10 minutes to code and test - including fixing the classify_reverse issue.

tatarize, 2021-11-05:
Approved. Though mixing and matching different checks took a bit longer. But, yeah, that stuff would feed in there and I'll assume you tested that since it tends to matter only with obfuscation objects.

commit 2b8fe1b3 (2021-11-05): Merge pull request #668 from meerk40t/include_stroke_width_in_raster-

Fix #667 - include stroke width when making image of raster elements
commit dc936c95 (2021-11-05): Fix #667 - include stroke width when making image of raster elements

This also fixes an issue when classify_reverse is True whereby images are rastered in the wrong sequence.
closed 2021-11-05 by Sophist-UK