meerk40t/meerk40t #680: [Enhancement] In Execute window, warn user if stuttering is likely

Sophist-UK, 2021-11-17:
The idea here is that when you open the Execute Window, we analyse any images in the operations to be executed based on operation speed and PPI, step etc. to determine whether there will be occassional or frequent stuttering due to USB bandwidth, and if so then we give the user a warning and "expert" recommendations:

a. "Warning: You are likely to experience heavy stuttering when burning the following image elements: image1234. Use the Raster Wizard to increase the step to 2 or 3."

b. "Advice: You may experience occasional stuttering at these settings when burning the following image elements: image1234. You can reduce or eliminate these by: Turning on pulse grouping, using the Raster Wizard with a slightly higher step, increasing / reducing the operation speed to xxxmm/s, changing  the PPI to 1000."

tatarize, 2021-11-18:
Stuttering is possible at about 100mm/s if everything is wrong. So step 2 might stutter at 200mm/s. Step 3 at 300mm/s. If step 1 and group pulses is on the stutter ceiling should be about 300mm/s. This is for step-1 with a pure gray image. If the image is either 25% or 75% gray it will go X000X000 or XXX0XXX0 respectively which has a stutter ceiling of 400mm/s. At step 2 with 25% gray the stutter ceiling is 800 mm/s which exceeds the 670 mm/s possible with the card.

@joerlane we might need your snooping at ticks powers, we need to measure the actual distance it travels between stutters in a worst case scenario. Specifically if we're in stuttering at an exact speed is it possible to know exactly the relationship between stutters at a speed. If you do `rect 1in 1in 5in 5in fill gray raster --speed 150` in console and raster that exactly how long is it between stutters. And compare this to --speed 200. Pure gray should have a stutter ceiling of 104mm/s. So from a pure protocol standard what is the time between gasps for air. If we set the raster to step 1, pure gray and run it at 600mm/s exactly how long is it between stutters (these should happen nearly instantly).

We would be then able to calculate in simulation not just a warning about stuttering but potentially where it might stutter. The math here isn't too terrible. And really knowing the stutter ceiling might give the right answer. We're going pure gray at 104mm/s and hit stutters. This means we're sending DaUaDaUaDaUa at 2 characters per mil. 4094.49 mils/sec speed which means in about 1 second we process 8189 characters. This means that our stutter ceiling should be anytime processing per second must exceed 8k/s.  We have `characters-required-to-go-a-distance / speed` > 8k/s. So if our code is set to go DzUz that's (4/510) * speed > 8k/s. So we have speed > 1020,000 mils a second or 25,908 mm/s to exceed the stutter ceiling.

Stuttering happens when speed > 8189 * (distance / chars). So 8189 / (chars/distance). So if 2/1 which is the worst we stutter at 4094.5 mils/s . At step 2 our chars/distance = 2/2 which gives us 208.0006 mm/s (2/2) or 208.

The stutter ceiling is located at: (average distance per character) * 208mm/s. So if `speed > 208 * (distance/character)` we will stutter. So if we're averaging 500 units of distance over 200 characters we get 208 * 500/200 which is 520mm/s as our stutter ceiling.

We can calculate this for an image if we check how often does the pixel next to this pixel equal that pixel. (That percent) * 104 gives the speed at which stuttering will occur. though this is average. You could try this in clumps or scanlines. Though we need to know the time between data accepting gasps to know how long you need to exceed this value before stuttering will occur, or rather how quickly we end up outstripping out buffer. We could probably just calculate the raw data flow with the busy commands etc. If we send more than 273 packets a second stuttering is likely to occur. Though the stuttering here is very likely to screw that up as a raw calculation since we can't really exceed that value and not stutter, so we can't exceed that value.

Sophist-UK, 2021-11-19:
It is likely that this can only be an estimate of whether it will stutter or not, both because of the cost of extensive analysis and because of variability of hardware. 

Rasters (of constant greyscale) are easier to calculate taking into account speed, PPI, greyscale level and pulse grouping. Images are harder because they are variable - but we can probably use the knowledge above to decide how to break each image raster line into chunks of pixels that we can easily calculate the verbosity of.

I suspect that we can classify images into the following:

a. Almost certainly no stutters
b. Possibly occasional stutters
c. Highly likely frequent stutters

tatarize, 2021-11-19:
Well, we could calculate the other properties quite easily, and just not calculate directly the theoretical value for (distance/character) based on step size. If we exceed the 100 for step 1. We have possible stuttering. If we exceed 200 for step 2 we have possible stuttering. If we exceed 300 for step 3 we have possible stuttering. From this alone we could give a worst case estimate for when stuttering may happen.

```
0: 0.002000
25: 0.122000
50: 0.198000
75: 0.298000
100: 0.398000
125: 0.498000
150: 0.598000
175: 0.698000
200: 0.798000
225: 0.898000
250: 0.998000
275: 1.098000
300: 1.198000
325: 1.298000
350: 1.398000
375: 1.498000
400: 1.598000
425: 1.698000
450: 1.798000
475: 1.898000
500: 1.998000
525: 1.902000
550: 1.802000
575: 1.702000
600: 1.602000
625: 1.502000
650: 1.402000
675: 1.302000
700: 1.202000
725: 1.102000
750: 1.002000
775: 0.902000
800: 0.802000
825: 0.702000
850: 0.602000
875: 0.502000
900: 0.402000
925: 0.302000
950: 0.202000
975: 0.127000
1000: 0.004000
```

Gives the various values. This is matched against the PPI/Gray values out of a 1000. In all cases you can divide this value by the step value. Though not if partial grays are used. Only step values against pure black and white (grays will use a PPI like factor if they are partially gray). Stepping by 25 PPI each time. I didn't really do the math I just plugged it into a calculation script that applied PPI and told me how many characters that would take.

In reality the values for 0 and 1000 are a bit too high since we could use 255 of 1 character there rather than 3. We would do Dzzz235 rather than D255255255235 which would certainly work. But, we're talking about ranges that will never stutter anyway.

I'm guessing this is going be pretty consistent. We lose 100 characters each 25 PPI we take with the worst being 1 inch in 500PPI which would take ~2000 characters. So if our gray is 75% we've moved 250 so we've lost 1000 characters, meaning. This basically is the same as going with a step of 2. 

I somehow thought this would be more parabolic but it's apparently a constant rate. 20% characters per 10% gray. Just straight lines maxing at 50% gray. So distance from `abs(grey - 50%) * 2`.

This gives us a stutter ceiling relative to localized PPI (at step 1):

```
0: 104000.000000
25: 1704.918033
50: 1050.505051
75: 697.986577
100: 522.613065
125: 417.670683
150: 347.826087
175: 297.994269
200: 260.651629
225: 231.625835
250: 208.416834
275: 189.435337
300: 173.622705
325: 160.246533
350: 148.783977
375: 138.851802
400: 130.162703
425: 122.497055
450: 115.684093
475: 109.589041
500: 104.104104
525: 109.358570
550: 115.427303
575: 122.209166
600: 129.837703
625: 138.482024
650: 148.359486
675: 159.754224
700: 173.044925
725: 188.747731
750: 207.584830
775: 230.598670
800: 259.351621
825: 296.296296
850: 345.514950
875: 414.342629
900: 517.412935
925: 688.741722
950: 1029.702970
975: 1637.795276
1000: 52000.000000
```

tatarize, 2021-11-19:
In `K40 Whisperer`, without counting the double-gated bug, we'd have a stutter ceiling chart:

```
0: 69333.333333
25: 1216.374269
50: 700.336700
75: 465.324385
100: 348.408710
125: 278.447122
150: 231.884058
175: 198.662846
200: 173.767753
225: 154.417223
250: 138.944556
275: 126.290225
300: 115.748470
325: 106.831022
350: 99.189318
375: 92.567868
400: 86.775136
425: 81.664704
450: 77.122729
475: 73.059361
500: 69.402736
525: 72.905713
550: 76.951535
575: 81.472777
600: 86.558469
625: 92.321349
650: 98.906324
675: 106.502816
700: 115.363283
725: 125.831821
750: 138.389887
775: 153.732446
800: 172.901081
825: 197.530864
850: 230.343300
875: 276.228420
900: 344.941957
925: 459.161148
950: 686.468647
975: 1168.539326
1000: 34666.666667
```

tatarize, 2021-11-19:
[image]

Here's the proper chart. Though it's fairly easy for me to calculate the ceiling now. It's a confluence of how gray it is and the step distance.

Sophist-UK, 2021-11-19:
Hmmm ... funny how the focus is on the techy aspect of this rather than the usability perspective...

tatarize, 2021-11-19:
[image]

tatarize, 2021-11-19:
[image]

[image]

closed 2024-10-04 by jpirnay