meerk40t/meerk40t #682: Utilize serial number challenge to identify laser

tatarize, 2021-11-18:
Challenge coding has been solved. Enter your laser serial number and identify it.

tatarize, 2022-07-20:
This has a spot in the GUI but is not currently implemented.

Sophist-UK, 2022-07-20:
To make this as simple as possible for the user, I would suggest that MK autocreates a device if one does not already exist with this serial number.

So, from a user perspective, the first time they run MK with a laser plugged in, they get a correct M2 device created, and that would then be reused every time MK connects to the same laser.

tatarize, 2022-07-20:
The Serial Number code for M2 Nano is a lot dumber than that. You can't read it, you need to challenge it. You send it the serial number and it replies whether that's correct or not.

tiger12506, 2022-07-20:
To further elaborate on tatarize, the only way to interact with an m2:nano to determine it's serial number is to check that it matches. It seems to be meant by the original hardware designers as a method for determining a dongle protection match.  We managed to "break" it by recognizing the hash algorithm that was protecting it. There is no way to know what a new K40's serial number is when you attach it, by hardware design, we can only hash the serial number and ask the m2:nano if it matches.

Sophist-UK, 2022-07-20:
Ah. My bad.

closed 2023-04-22 by tatarize