meerk40t/meerk40t #683: [Suggestion] Make scene look more like Inkscape

Sophist-UK, 2021-11-18:
1. Fix the rulers to the surround.

jpirnay, 2022-06-01:
I believe we have done quite a bit for the GUI. Need to have a more detailled spec for changes to be applied.

Sophist-UK, 2022-06-02:
This is absolutely NOT completed.

This is a request for a GUI that groups functionality in a way which accords with the task that the user is doing. If you don't understand the requirement you should ask, not just close the issue.

jpirnay, 2022-06-02:
Happy to reopen it - but the description in the suggestion was only referring to the ruler and did not include anything you've provided now in the comment. It would help us all if you could elaborate on the concepts you have in mind.

Sophist-UK, 2022-06-02:
Well I did open this 6 months ago, so my memories are a bit hazy. And as you quite rightly pointed out, the ideas were not well thought out or described.

But I guess the guiding principle I had in mind was - for the design / drawing part of MK - to give it as much synergy to Inkscape as possible in order to make it as familiar to typical K40 users as possible.

In essence, Inkscape has a button bar (running down the left side) which controls a different set of layouts for each button. So the pointer button's layout shows the Position pane, and other buttons show specialised panes for the specific functionality for that button (nodes, rectangle, ellipse, text etc.).

Of course, you would want to maintain flexibility on layout so you might e.g. have two layout modes - a simple layout mode where the layouts are fixed(as close to Inkscape as you can make it), and an expert mode where each button's layouts can be tweaked by the user. 

I am not sure how well I have explained this. Perhaps there needs to be an online meeting to see what consensus there might be on a new UI design.

closed 2024-10-04 by jpirnay

Sophist-UK, 2024-10-04:
I have no issues with this being closed, but it was not completed.