meerk40t/meerk40t #692: [Bug] Raster elements are combined and then fail Burn inner first

Sophist-UK, 2021-11-22:
If you have various raster elements inside multiple closed cut paths, then because these Rasters are combined into a single image which cuts across these several closed cuts, the individual rasters are not considered to be inside and are burned entirely separately.

### Proposed solution

1. Change test for actualised rasters from checking operation contains only a single image to checking that it contains only images and that they are all actualised.

2. Change the routine that vectorises raster elements as follows:
     a. If Cut Inner first **is** selected then group all elements that share an overlap (similar to the algorithm used in the classification routines) into separate images.
     b. If Cut inner first is **not** selected, then group all elements using the same grouping routine but adding the acceleration distance onto either side of the bbox in the raster direction (which will separate into separate images those rasters that are sufficiently far apart horizontally to be faster if rastered separately).

tatarize, 2021-11-24:
If any part of a raster is inside a vector the entire raster is inside the vector.

So a circle inside a square is really the square inside the circle if that's a raster object.

Rather than outlines we are dealing with filled shapes and the entire inside of the filled shape is inside any shape inside that shape.

Sophist-UK, 2021-11-25:
Ah .. ok.

tatarize, 2021-11-25:
It's the way things need to be thought of, and currently aren't. It's not saying that that is not a problem but that is the way filling should work.

Sophist-UK, 2021-11-25:
My solution is still a positive one though as it will improve optimisation.

tatarize, 2021-11-25:
I'm not sure it won't make things more complex for a smaller amount of help. It sounds like specialty grouping based on position etc. Which may run the risk of making the algorithm we're using more complex and less understandable from the outset if it has a bunch of edge cases built in rather than a better overarching algorithm that accounts for the edge cases more holistically.

Sophist-UK, 2021-11-25:
1. My proposed solution (or at least the one I have in miond rather than my poor description) **is** holistic in nature rather than an edge case.

2. If you have full width burn with a small raster at left edge and another small raster at right edge, then the head is going to traverse the whole width of the bed for each row, rather than do each separately. Thus the time savings for 2b. above would be substantial.

Sophist-UK, 2022-02-15:
Fixed by #795

closed 2022-07-20 by tatarize