meerk40t/meerk40t #697: 0.8.0 Report Thread.

tatarize, 2021-11-25:
So it's that time again, 0.8.0 may or may not have a bunch of small problems but should be the next version. But, to spare as much time as we can patching things though there might be easier ways of doing that. I'm hoping we can switch the version branches up somewhat soonish.

https://github.com/meerk40t/meerk40t/pull/675

* Kernel Services work. These can be checked with `service` in console.
* You can add additional services with `service <domain> start <name> <index>` for example `service device start moshi 1 will begin a new moshi device.
* The registered.dict() along with the registered.dict() of the active services is generally called the lookup.
* Services and Modules now have a much more overt lifecycle. The lifecycle of a Service is that it's activated and deactivated which calls `attach()` and `detach()` repeatedly and `shutdown()` once.. And the lifecycle of a module is that it's opened and then closed at which point it's deleted.
* Lifecycle elements `Service`, `Module` and the `Kernel` can have delegates, these delegates get their `@lookup_listener` and `@signal_listener` flags methods attached in the kernel until the lifecycle of the module is either `shutdown()`, `deactivated()` or `closed()`
* Lifecycle delegates also have lifecycle events called on them. If you `add_module_delegate` to a module it will have the `open()` and `closed()` lifecycle events called on those objects to match the lifecycle of the object. This will be done at registration of the delegate if the object lifecycle is already at a mature stage.
* Windows which are a type of Module can merely flag methods with `@signal_listener` to have them listen and unlisten to those signals during their lifecycle.
* Ribbonbar buttons listen for changes to the registered buttons and these are changed when the lookup changed. This is done with a `@lookup_listener` which announces when things are changed, this could be a new button registered dynamically or it could be a service is switched to a different one with different buttons.
* Legacy Device was used and discarded. There are only the four drivers currently. Lihuiyu, Ruida, Moshi, and GRBL.
* Devices have their own settings for bedsize and xy-scale. Setting these in preferences results in setting that for the driver. Then switching the device (currently done in LaserPanel) will switch the bed size based on the device being used.
* Most gui elements can be registered within a service and are dynamically built. This means we can have different buttons and windows for different device.
* Camera was added back into the code. This is because `Modifier` as a kernel object was dropped and consequently the Modifier.camera didn't work. There wasn't an easy way to transition this stuff over so it was briefly removed while I solve the kernel change. When it's spun back off it can have gui elements too, since the camera gui is now registered by the plugin more directly.
* DeviceManager was built around the legacy device swapping and no longer is needed.

---

The m2nano driver should still cut correctly.
The tcp driver for the m2nano likely won't currently work.
There is an unsightly lag starting the program up where it flickers for the OS being built.

I tend to need to know where the weak spots are. I can fix TCP and maybe test GRBL TCP a little bit. Build a NewDeviceWizard but then I'm out of rough edges that need to be fixed.

tatarize, 2021-12-23:
[image]

[image]

Some preferences that are now device specific should be in device or configuration rather than preferences.

Look into maintaining device settings between 0.7.0 and 0.8.0

tatarize, 2021-12-23:
Does not cleanly shutdown.

tatarize, 2021-12-23:
PM: Debug: ClientToScreen cannot work when toplevel window is not shown --- until hangs

tatarize, 2022-01-03:
Plot thickens. In Linux if you have a window stuck open from previous instance and it launches in that way it doesn't get the weird matrix effect. It loads that window normally for the duration.

tatarize, 2022-01-06:
Shutdown is now clean.
Matrix effect is fixed for linux.

Sophist-UK, 2022-01-07:
Testing in mock m2nano on my dev machine at commit 48c4be341b96424b10aa40d4ebff455ee3c116bb plus fix for #724...

* Device Config - Pulse Grouping->Pulse Shifting - I have no issues with a change of terminology, but if so then tooltip needs to be updated and made consistent with Advanced Operations Properties which has reverted to Group Pulses (after I previously made it consistent).
* m2nano vs moshi - pause and stop buttons are reversed. Grbl and Ruida have neither pause nor stop buttons.
* m2nano config - no tooltips for USB and Networked
* moshi config has way way way way too much empty space.
* window sizes not reset (or at least preferences isn't) after Reset Windows.
* few seconds delay between screen first appearing and being finalised and ready for accepting input
* execute window - closed distance is only meaningful (I think) when Cut Inner First is enabled. If this is the case, we need to grey it out when Cut Inner First is not enabled.
* when changing device (which is rare) ribbon flickers
* tools/scene seems to go into some sort of loop then crashes crashes crashes (16x) with the following:
```
MeerK40t crash log. Version: 0.8.0-beta1 git on Windows:AMD64 - 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\scene\scene.py", line 312, in on_paint
    wx.BufferedPaintDC(self.scene_panel, self._Buffer)
TypeError: BufferedPaintDC(): arguments did not match any overloaded call:
  overload 1: argument 2 has unexpected type 'NoneType'
  overload 2: argument 2 has unexpected type 'NoneType'
```

Sophist-UK, 2022-02-18:
OK - I have started testing this version on my dev machine i.e. without a real K40 and using mock. Here are my comments (limited to anything that would stop me using this in production rather than minor issues for future development):

* Need for pyserial - I think we need a better error reporting mechanism for people running MK from source who decide to upgrade. Rather than simply get a crash we should trap the `ImportError` and write a friendly instruction to run `pip install pyserial`.
* The v/h scales on my scene appear to be measured in furlongs or something. The nearest grid mark to 320mm is 0.0126 somethings. Sheesh.
* When I load my element classification test file, the tree is populated but the scene is empty.
* Right click on image element, steps menu every item, has checks against it, Raster Wizards / Scripts are missing.
* Right click on Raster op and select Make Raster Image causes a crash.
```
Traceback (most recent call last):
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\wxutils.py", line 146, in specific
    f(node, **func_dict)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\elements.py", line 4343, in inner
    returned = func(node, **ik, **kwargs)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\elements.py", line 3605, in make_raster_image
    image = make_raster(
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\laserrender.py", line 497, in make_raster
    gc = wx.GraphicsContext.Create(dc)
wx._core.wxAssertionError: C++ assertion "bmp.IsOk()" failed at ..\..\src\msw\graphics.cpp(2575) in wxGDIPlusRenderer::CreateContext(): Should select a bitmap before creating wxGCDC
```
* Simulation crashes
```
Traceback (most recent call last):
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\simulation.py", line 510, in open_simulator
    kernel.console(
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 2942, in console
    self._console_parse(command, channel=self._console_channel)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 3021, in _console_parse
    data, remainder, input_type = command_funct(
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 4336, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 1098, in plan_optimize
    data.execute()
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 335, in execute
    cmd()
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 652, in actualize
    if op.type == "op raster":
AttributeError: 'CutCode' object has no attribute 'type'
```
* Same error on execute of image op.
* Right click on Op node, select add special operation - is added at very end rather than after (or before?) this particular op.
* Crash on executing a Dot operation
```
Traceback (most recent call last):
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 2538, in run
    func_result = func(*args)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\spoolers.py", line 404, in run
    self._execute_program(program)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\spoolers.py", line 373, in _execute_program
    for p in program():
TypeError: 'RefElemNode' object is not callable
```
* Crash on execute of Engrave op
```
Traceback (most recent call last):
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\gui\executejob.py", line 299, in on_button_start
    self.context("plan%s optimize\n" % self.plan_name)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 193, in __call__
    return self._kernel.console(data)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 2942, in console
    self._console_parse(command, channel=self._console_channel)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 3021, in _console_parse
    data, remainder, input_type = command_funct(
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\kernel.py", line 4336, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 1098, in plan_optimize
    data.execute()
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 335, in execute
    cmd()
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 579, in optimize_travel
    self.plan[i] = short_travel_cutcode(
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\planner.py", line 1598, in short_travel_cutcode
    start_length = context.length_travel(True)
  File "D:\MeerK40t-64\Source\meerk40t_meerk40t\meerk40t\core\cutcode.py", line 415, in length_travel
    distance += abs(complex(self.start) - complex(cutcode[0].start))
TypeError: complex() first argument must be a string or a number, not 'tuple'
```
* Execute window - lists are even more cryptic than before - as a user I don't want to see one line per element - I want to see one line per Opnode, and later one line per spool object. Way way way too nerdy even for me.

closed 2022-07-20 by tatarize