meerk40t/meerk40t #704: Files not opening

chefmellis, 2021-12-12:
So about a week ago after using meerk49t for 6 months I developed an issue where all files load extremely slow.  Even files thst previously loaded quickly.  I uninstalled and reinstalled and issues continued….any ideas

Sophist-UK, 2021-12-13:
What version are you running? What O/S? Did this start after you changed versions or on the same version?

chefmellis, 2021-12-14:
> There's a list of older version (https://github.com/meerk40t/meerk40t/releases) do any of those open in this older faster methodology.

Running windows 10 hone 64 bit.  Issue is on the original download.

Sophist-UK, 2021-12-14:
What version of Meerk40t? And did this problem start when you updated Meerk40t to a new version or did it just start on the existing version of meerk40t?

tatarize, 2022-07-20:
not enough info.