meerk40t/meerk40t #719: Allow scene selection iof elements whilst avoiding unwanted moves.

Sophist-UK, 2021-12-27:
As I understand it there has been a deliberate decision to only select the entire scene with a mouse click to avoid unwanted selections and moves. Individual elements (or multiples) can now only be selected using a selection drag-box.

The problem is that this makes it much more difficult to select elements and use right-click functions to change them - and this is one of MK's most powerful features. Ideally we would find another way to prevent moving subsets of the whole without this being a deliberate choice by the user.

With `Undo` functionality this may become unnecessary.

Another possible approach is that if only part of the design is chosen, moves are disabled by default and you have to select Move Enable in a right-click menu to set a flag to enable moves.

tiger12506, 2022-02-16:
jpirnay's #825 should help this situation, but it's not enough. It's not possible to select more than one item from the scene that are not adjacent, even with #825. Other programs handle this with holding shift or control (Windows general is ctrl, Inkscape uses shift, etc.)

With #825, and above paragraph fixed, it would be possible to actually use selection in scene, but it would still be nice to see the "click = select all" gone. It should become a preference item default on.

jpirnay, 2022-06-02:
I believe this can be closed now: with [PR1058](https://github.com/meerk40t/meerk40t/pull/1058) you can select a single element with a single click and maintain the existing selection if you are holding down the shift-key.

closed 2022-06-03 by jpirnay