meerk40t/meerk40t #739: Catch ezdxf structure errors and tell the user parsing failed.

tatarize, 2022-01-06:
MeerK40t crash log. Version: 0.7.4-beta2 on win32 - 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "meerk40t\gui\wxmribbon.py", line 189, in <lambda>
  File "meerk40t\kernel.py", line 136, in __call__
  File "meerk40t\kernel.py", line 1822, in console
  File "meerk40t\kernel.py", line 1908, in _console_parse
  File "meerk40t\kernel.py", line 1260, in inner
  File "meerk40t\gui\wxmmain.py", line 320, in load_dialog
  File "meerk40t\gui\wxmmain.py", line 1558, in load
  File "meerk40t\core\elements.py", line 6489, in load
  File "meerk40t\dxf\dxf_io.py", line 66, in load
  File "ezdxf\filemanagement.py", line 149, in readfile
  File "ezdxf\filemanagement.py", line 92, in read
  File "ezdxf\lldxf\tags.py", line 288, in group_tags
ezdxf.lldxf.const.DXFStructureError: Missing required y coordinate near line: 18.

tatarize, 2022-01-06:
https://github.com/mozman/ezdxf/issues/601

Harden the ezdxf loading and issue correct fail afterwards.

tatarize, 2022-01-06:
https://ezdxf.readthedocs.io/en/stable/drawing/recover.html

will-janz, 2022-09-23:
If it helps, ~~I've attached a DXF file~~ where can I send a DXF file that I got an error on trying to import. Version 0.8.0027. I was, however, able to save the project as an SVG and reopen it without issue (so far).

tatarize, 2022-09-26:
`meanwell-uhp-500-hole-template` didn't suffer any error requiring hardening.

closed 2024-10-04 by jpirnay