meerk40t/meerk40t #745: [Bug] Move to corner of selected elements does not work on rotary

Sophist-UK, 2022-01-13:
Facebook user [@Brian Catano](https://www.facebook.com/groups/716000085655097/user/561442555/) reports that the Navigation buttons to move the laser to the corner of the selected elements does not work when he has rotary enabled. (See https://www.facebook.com/groups/716000085655097/permalink/991805394741230/ for a video.)

As @tiger12506 pointed out it looks like the scaling is out - he says his rotary scale is c. 0.5 something.

Sophist-UK, 2022-01-13:
A quick look at the code for Navigation pane does not seem to have any noticeable differences for Centre vs. corners, so the fact that corners work may just be a coincidence.

However I wonder whether it is because these buttons use console "move_absolute" rather than "move".

@tatarize Do you think this is the cause?

Sophist-UK, 2022-01-13:
Hmmm..... perhaps not - these seem to be synonyms at line 401 in spoolers.py.

tatarize, 2022-01-19:
Wow. 5 days. I must be way too deep in this galvo stuff.

Yeah, move and move_absolute is the same thing. Though really both might want to do move absolute. The odds are pretty good the movement between the corner and the object is done in actual units rather than scaled units. So it just doesn't account for that. In theory the newer version in 0.8.0 will account for that. It does the scaling x and y correctly.

Sophist-UK, 2022-01-19:
We should probably try to fix this in 0.7.5 too.

tatarize, 2022-01-19:
The problem might be a little deep in the code. But, here's how it works. The rotary is implemented in the cut planner. There's an extra bit for rotary that what it does is scales the rotary based on the current position of the laser head. So if you are at the upper left hand corner where the project will start it should be fine. The rotary currently works by stretching the field. So if each tick moves you down a different amount in the y-axis than usual ticks of the board are you are supposed to scale that so that you get a 1:1 step distance. This gets applied before the things get turned into cutcode. So the end result is a squished or stretched set of objects needed to create the correctly sized end result.

Now, in 0.8.0, there was a clear need to support other devices required being able to have the device native units and mk units completely different. So rather than relying on 1 pixel, 1 step, 1 mil being the universal size things got converted so that we can have all sizes in nanometers (doesn't have to be the final choice but makes all issues very big and can store all practical distances in integers) so in the config when you say that you have a stretched belt and your step x is 1.05 mils wide or you switched your stepper motor and you now have a 2 mil step it will take that into account and when it converts from mk units to device specific units it will adjust things by that factor. So jogs are sent to the driver in actual units like `jog 10mm to the right`. And it uses its understanding of the native units to perform that operation. This tended to result from issues where galvos would have lens that would effectively change their bed size, and literally native numbers had no bearing on distances. This also means that raster-wizard script should probably need to be applied during the cut planning stage at the same place that actualization currently occurs in cases where it's required, since we can't really convert units into native units except during cut planning. Everything else has actual distance units which is unrelated to steps, and this applies to steps in x and y as well.

Now, a couple things crop up when trying to support rotaries more robustly. The first is, not all rotaries replace an axis. You could stretch the x and y and use the x and y steps to do the rotaries and odds are pretty good you'll do fairly okay. But, a lot of boards have additional axis and can step the rotary independent of the x and y. In many cases there's still access to the y axis even so you have your axis x, y, and r. So there's a concept of splits. Basically how big of an area should your raster before you need to move the raster axis to raster the next section. While this isn't common on the M2 Nano. It is fairly common for some other types of rotaries. Likewise some people can have a z bed for 3d or 2.5d work. But, here we need to know if our rotary is replacing an axis, we'd need to know how that affects our real world distances along the axises in question.

The issue with doing this in 0.7.5 is that it's a *very* breaking set of changes. The raster image stuff doesn't even work correctly still since there's question of how big a pixel is without a very clear answer under the new scheme. And much like some problems never got fixed in 0.6.24 since there wasn't really a way to do that under that architecture the same might be true here.  If you're going to do rotary work, you may want to unlock the rail, move to the corner of your work and call that (0,0). While the changes in 0.8.0 do actually seem like they would largely fix this set of problems they require a pretty large set of non-trivial changes. 

If there's a minimally invasive way to do this, I could see that being added but I think the correct and best answer for now is to manually move your laserhead when doing a rotary.  In 0.8.0 the hope is the rotary support can be a lot more robust even allowing other axis control of those locations. So that if you just hooked your rotary to a step-stick hooked to your rPi rather than used it to replace your y-axis you could just move the laser head normally and step the rotary independently and still be able to do raster jobs.

closed 2024-10-04 by jpirnay