meerk40t/meerk40t #748: Loading SVG with text messes up alignment and size of text

arminth, 2022-01-14:
Hi, I have created a simple SVG in Inkscape 1.1. It's a circle (blue) and text in KunstlerScript which is centered in the middle of the circle. (For engraving wooden, round lids.
When I load the file in Meerk40t 0.7.5beta , it resizes the text bigger and puts it out of original alignment (shifted up and parts are outside the circle).

[image]
[image]
Here the original SVG:
[Deckel_mehl.zip](https://github.com/meerk40t/meerk40t/files/7870669/Deckel_mehl.zip)

This is a problem as you would normal assume that all mesaurements stay the same after importing.

Hope, there is a solution as I really love Meerk40t!

Thanks and best regards!

Armin

Sophist-UK, 2022-01-14:
This is a known issue with SVG Text elements not loading correctly.

SVG Text element specification is complicated, and difficult to implement. It is on the To Do list to fix, but nowhere near the top.

The solution is to convert the text to paths in Inkscape before you save it. If you want the text to remain editable, then duplicate it beforehand and hide the duplicate.

arminth, 2022-01-14:
@Sophist-UK thanks a lot. I was searching the issues but couldn't find this. With converting to path, it should work. I have to check on the infill though. Will try!

Cheers
Armin

Sophist-UK, 2022-01-14:
You might want to join the [Facebook group](https://www.facebook.com/groups/716000085655097) as a place to ask for future help as responses are likely to be faster there (though it might well still be me who answers 😉).

closed 2022-01-14 by Sophist-UK