meerk40t/meerk40t #750: Idea: Saveable presets for different materials

jaredly, 2022-01-15:
Next to my k40, I have a piece of paper where I've written down the settings that I've found to work for various materials that I use, for engrave, cut, and raster.
I'd love to be able to put these into meerk40t, somehow.
One dream-interface would be: instead of specifying mm/s & ppi for each operation, I just select the material that I'm using, and meerk40t knows what to use for each operation (engrave, cut, and raster). I could imagine there being a couple of options for engrave and raster (dark vs light etc.), but I'd be able to select from just a couple of darkness options instead of fiddling with specific parameters.

Now, that might be something of a drastic change -- the much more incremental version would be to have this configuration stored out of sight somewhere, and allow me to right-click an operation and select a configuration from a dropdown, and it would set the parameters accordingly. So people that don't use this wouldn't have to change their workflow at all.

It would also be super cool to be able to export/import this config, to share them between users.

I'm imagining something like

#### Thin Cardboard
| Operation | Power | Speed | PPI |
| --- | --- | --- | --- |
| Cut | 11% | 20mm/s | 1000 |
| Engrave dark | 10% | 30mm/s | 1000 |
| Engrave light | 10% | 30mm/s | 750 |
| Raster dark | 12% | 100mm/s | 1000 |
| Raster light | 12% | 150mm/s | 1000 |

#### HDF
| Operation | Power | Speed | PPI |
| --- | --- | --- | --- |
| Cut | 15% | 10mm/s | 1000 |
| Engrave dark | 13% | 25mm/s | 1000 |
| Engrave light | 13% | 30mm/s | 500 |
| Raster dark | 12% | 100mm/s | 1000 |
| Raster light | 12% | 150mm/s | 750 |

etc.

It would be lovely then to have meerk40t recognize when I need to adjust the power between operations, and automatically add a "beep/interrupt" :)

Does this sound appealing at all? I'd be happy to dive into implementation if folks think this would be welcome.

Sophist-UK, 2022-01-15:
@jaredly Thanks for this great suggestion - however it seems that you may not have found this existing functionality (or perhaps you have and I have misunderstood your suggestion.) You can already save and restore normal operations if you right-click on the `Operations` node right at the top of the tree, you will see Use and Save. 

The current version of MK has some limitations - it has fixed slots and doesn't save special operations. The next version saves special operations like beep or interrupt (#727) and there has been some discussion about having slots you can name yourself but I don't think this has been coded yet.

Looking at your tables...

Personally I find it better to try to stick with 12mA on the hardware power, and use speed, steps, PPI, etc. to get the depth of burn I want. This gives the fastest burn, and because I use the same hardware power for every operation, I can run them all at the same time. As an example, if we assume that 30% on the hardware is 12mA, then I would have a table like this (this is an example - obviously you will need to test these out for yourself):

**HDF - Power 12mA**
| Operation | Speed | PPI | Passes | Steps |
| --- | --- | --- | --- | --- |
| Cut | 7mm/s | 1000 | 1 | n/a |
| or Cut | 15mm/s | 1000 | 2 | n/a |
| Engrave dark | 35mm/s | 666 | 1 | n/a |
| Engrave light | 35mm/s | 300 | 1 | n/a |
| Raster dark | 250mm/s | 1000 | 1 | 1 |
| Raster light | 250mm/s | 1000 | 1 | 2 |

😃 **|Finally a **BIG** thank you for your offer to put time and effort into coding etc.** There is never anywhere near enough resource to do everything we want to do and Meerk40t can do with all the help it can get. Can I suggest that you come chat with us on IRC or Discord and discuss what you can help with? 😃

jaredly, 2022-01-15:
Oh neat, yeah I hadn't noticed that feature 🙃. Thanks for your detailed response! I'll have to play around with ppi; I've been extremely conservative with my power setting, because I haven't gotten around to installing an ampmeter, and saw people saying that setting the power too high would compromise the lifetime of the tube 😅. Thanks for the tips! I've joined the discord, happy to chat there about current dev directions!

closed 2022-01-15 by jaredly

Sophist-UK, 2022-01-15:
Do not set the hardware power above 25% until you have a mA meter installed. My k409 varies between 27.5% and 31% for 12mA, and the lowest I have heard is 25%. Installing a mA meter is not difficult. The most time consuming part is cutting the hole for it.

(When I first got my k40, the tube lasted about 1hr of cutting time before it lost power - because I had been cutting on 80%. Was replaced under warranty because I had used it in accordance with the operating instructions.)

jaredly, 2022-01-17:
Yeah I've been keeping it below 15% to be extra careful, and having a great
time with it 😅 ammeter should come tomorrow though, so then I can go full
bore.