meerk40t/meerk40t #764: Add a note to Interrupts?

jaredly, 2022-01-22:
I'd love to be able to add a note next to interrupts. I'm using them to adjust the power settings, and it'd be great to have listed the power that I want to set it to.

tatarize, 2022-01-22:
I'd think there'd be pretty limited use there but it wouldn't be too hard to do. Not sure how to interface it either. I guess just interrupt with some arguments and catch it for the cli and gui interrupt code there.

Sophist-UK, 2022-01-22:
As I commented on [Discord](https://discord.com/channels/910979180970278922/910983847435448420/934368134322847764):

You would IMO be better off adjusting your Operation speeds (and for rasters, steps - and if absolutely necessary PPI) so that all operations are burned with the same hardware power settings (ideally full power i.e. 12mA).

tatarize, 2022-01-22:
Yeah, but that doesn't detract from the note, there might be other messages that you'd like to send with interrupts and they totally can provide messages. If somebody made a meerk40t file that was pre-set you could very well use interrupt-messages to make sure your instructions are followed. "At this point turn the wood over, the rest of these cuts are for the back side." Like that's be some weird things but should be completely possible to do with a little tweaking.

Sophist-UK, 2022-01-22:
True - and it would certainly make more sense when we are able to save and restore special operation in an svg file.

tatarize, 2022-01-22:
And the discussion in discord seems to have gone in a very useful direction, namely just make a different node type that runs a line of console command and can be edited. Then modify the commands to allow an interrupt command. In theory, you could just interject console commands at different points in the spooling process. Assuming the commands are generally rather global this would be a pretty versatile solution. Admittedly a couple commands in the cmdop nodes don't have clear console command parallels but that could be rectified. You could maybe change the display label of the node type depending on the command a little bit to hardcode a sort of look and feel in there.

PR #770 (2022-01-23): Add "ConsoleOperation" to allow interrupts to have notes and custom labels
Fixes #764 

Here's my first stab at it! I left in the original interrupt menu item so I could compare behavior, I'm happy to take them out once this is in good shape.

https://user-images.githubusercontent.com/112170/150663972-47504b97-5f61-4295-9001-da3a20b6c793.mp4
closed 2022-01-23 by tatarize