meerk40t/meerk40t #766: [Suggestion] Raster Wizard can dither upwards as well as downwards

Sophist-UK, 2022-01-22:
See https://www.imag-r.com/enhance-img and Paint.Net for various examples of algorithms.

tatarize, 2022-03-20:
@Sophist-UK Please describe what this did since the pointed to prior art does not work anymore. Is this merely density raising algorithms or something. Show them a bunch of pictures and they upsample or dither being applied as a scanline bottom-to-top?

Sophist-UK, 2022-03-20:
There was a markdown error in the link which I have now corrected.

Bu in essence, Raster Wizards/Scripts are currently designed to reduce the number of pixels. I am suggesting that we find and implement algorithms that work the other way, increasing the number of pixels instead.

tatarize, 2022-03-21:
The current code would absolutely upsample the image if that was required.

The algorithms are the same. Resize very much does work both directions.

tatarize, 2022-03-21:
Intelligently upsampling an image where you get new pixels that are relevant rather than more pixels being created out of the intersection of fewer pixels are well beyond the scope. I'd point out that it's like saying "enhance" in a TV show 50 times until you get 4k resolution out of a liquor market's VHS tape. --- I would also call it absurd but it's actually possible with some AI to do rather coherently especially with cameras where you have many frames of roughly the same thing. The imag-r probably does super-resolution algorithms ( https://learnopencv.com/super-resolution-in-opencv/ )

Sophist-UK, 2022-03-21:
That may be how imag-r does it. It is possible that imag-r had proprietary algorithms, but I suspect that much of it is good commercialised packaging of open source code.

closed 2022-07-20 by tatarize