meerk40t/meerk40t #769: [Bug] Raster is broken when "Burn complete subpaths" is checked

jaredly, 2022-01-23:
This SVG reproduces, load it up, make sure "Burn complete subpaths" is selected, and click "simulate"
[blank-1 (4).svg.zip](https://github.com/meerk40t/meerk40t/files/7919590/blank-1.4.svg.zip)

I think this was @Sophist-UK's work
```
MeerK40t crash log. Version: 0.7.5-beta2 git on Darwin:x86_64 - 4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/gui/panes/laserpanel.py", line 322, in on_button_simulate
    self.context(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel.py", line 152, in __call__
    return self._kernel.console(data)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel.py", line 1892, in console
    self._console_parse(command, channel=self._console_channel)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel.py", line 1972, in _console_parse
    data, remainder, input_type = command_funct(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/kernel.py", line 1324, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/planner.py", line 1147, in plan_optimize
    data.execute()
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/planner.py", line 354, in execute
    cmd()
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/planner.py", line 603, in optimize_travel
    self.plan[i] = short_travel_cutcode(
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/planner.py", line 1676, in short_travel_cutcode
    for cut in context.candidate(complete_path=complete_path, grouped_inner=grouped_inner):
  File "/Users/jared/clone/fork/meerk40t/meerk40t/core/cutcode.py", line 373, in candidate
    if grp[0].burns_done < grp[0].passes:
TypeError: 'RasterCut' object is not subscriptable

```

tatarize, 2022-01-23:
@Sophist-UK #771 fixes this.

Sophist-UK, 2022-01-23:
It was indeed my (insufficiently tested) work. 😢

tatarize, 2022-01-23:
Well, I didn't test it with rasters either though I tested the vector part of it.

The fix was to skip that section for RasterCut by checking to make sure we were actually dealing with a CutGroup and to mark RasterCuts and `first=True` which causes them to get included in consideration for the distance routine in the cases when they wouldn't have. 

Merely fixing the Candidate code would still exclude the rasters because they didn't qualify for distance considerations even though they actually *were* now correctly marked as candidates.

Sophist-UK, 2022-01-23:
I hope that there are no other similar issues with other cut types.

tatarize, 2022-01-23:
All the other cut types get made into cut groups with objects in them. I think Dots doesn't get transformed and produces lasercode directly but it might be something to check. Rasters and Image Ops became RasterCut which aren't put in a group as such.

tatarize, 2022-01-23:
It occurs to me that Dot might not work in 0.8.0 with the switch in native units as such. Might need to check through that when it comes time of RCs.

Sophist-UK, 2022-01-23:
I have a test file with dots I can use. When I fix the other stuff I will check this.

closed 2022-01-26 by tatarize