meerk40t/meerk40t #772: [Sticky] Anti-twitching reports

Sophist-UK, 2022-01-23:
If you experience any issues after turning on "Reduce Vector Twitching" in the Lhystudios driver Config window, we would appreciate it if you could log these here.

Sophist-UK, 2022-01-23:
I have turned this on and it definitely reduces / eliminates twitching and increases the speed burning lines of small leather stitch holes.

However, I have experienced greater frequency of USB hangs - unplugging and replugging recovers without any loss of burns, but this is happening twice in the one hour of burns to date when it normally only happens once a month or so.

tatarize, 2022-01-26:
Should in theory have limited effects on usb connections, but I guess keeping an eye out is fine. This is why it's not default yet.

Sophist-UK, 2022-02-18:
Works great for me.

tatarize, 2022-02-18:
I've not seen anything but it's not on by default. I suppose I should make a video about using it and get people to turn it on. But, after @joerlane analysis of the twitches and my research with different angles and what causes the twitches, I'm pretty sure it's a goof in the Chinese software that it's not written like this from the start. That or some other versions have different values or whatever. Since it really seems to be done in error. That or we should more consistently use reset which seems to more properly twitch in the correct direction. In fact, it might be the case that that is what @ means. Since really there's not a difference between "NSE" and "@NSE" other than kicking it fully back rapid mode.

Sophist-UK, 2022-02-18:
No - vector anti-twitching is not on by default. I will make a Facebook group post about these two beta options, hence my suggestion on Discord that this vector twitch option is either turned on by default or made the permanent way of doing things.

As for chinese software, it wouldn't surprise me if that did stuff poorly too - and that you (and Joe) are now literally the world experts on the Lhy controllers.

closed 2022-05-22 by tatarize