meerk40t/meerk40t #789: [Feature] Drag laser position using mouse

jaredly, 2022-01-31:
I'd love to be able to drag the laser with my mouse instead of just using the jogger to get it to a certain position

tatarize, 2022-02-04:
Yeah, that's clearly doable. Some of the tools in 0.8.0 would in theory be set to allow that if they got coded up the rest of the way.

tiger12506, 2022-02-06:
Oh.... Or even just a "set laser position at mouse position in scene" Wow, that would be crazy useful.

tatarize, 2022-02-07:
As useful as it would be easy to write.

tatarize, 2022-02-14:
[image]

`tool relocate` added to 0.8.0

https://github.com/meerk40t/meerk40t/commit/fb295ad88d9ce75b745cc10fe20dc6b194978971

Didn't add an icon for it, or ellipse which I also quickly added. I figure the hard part of tools there is making the icons and figuring out where to put the toolbar ultimately.

tatarize, 2022-02-15:
Ok. Added the icon for it and such. It's limited to the bed size so it shouldn't be allowed outside the bed. Turns out I still had the older icons I just needed to set the tool name to equal the right checks. Also added the circle, ellipse, polyline, vector, polygon, text tools too. And added in some constructive geometry functionality though that's currently only in the console.

closed 2022-04-29 by jpirnay