meerk40t/meerk40t #803: [Bug]: Tree Selection/Delete Key issues

tiger12506, 2022-02-06:
### Summary Description

This is for tracking any remaining tree selection/delete key issues remaining before next release. Probably any fixes should be applied to main branch, and merged also into tatarize-services.

### MeerK40t Version

main

### MeerK40t Type

Git

### Your Operating System

Linux

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Sophist-UK, 2022-02-06:
#793 was the PR that was supposed to fix the delete key issues.

Have there been any examples of it not working as expected after this was merged. If so please document them here as @tiger12506 suggests.

tatarize, 2022-02-14:
I cannot delete text objects.

Sophist-UK, 2022-02-14:
I can't see anything directly in #793 code that would cause this. My first thought was it was checking for Shape and SVGImage and I had omitted to list SVGText but it turns out that it deletes everything.

I just tested it on SVGText nodes (both in Ops and Elems) and it works just fine for me.

Sophist-UK, 2022-02-24:
Closing as the SVGText issue cannot be replicated and there have been no reported issues since this was released in a full release.

closed 2022-02-24 by Sophist-UK