meerk40t/meerk40t #810: [Suggestion] 3x and 4x pulse grouping.

Sophist-UK, 2022-02-09:
Make pulse grouping a choice of grouping 2, 3 or 4 pulses together.

2 as current. 3 or 4 using a similar technique but doing it over 8 bits instead of 4.

tatarize, 2022-02-10:
I'm unsure the as to how effective this would be. But, a general solution to pulse grouping wouldn't be terrible. Currently it's just a static set of calls. We could certainly buffer any amount of bits and move them around. But, I could certainly see this being rather effective, since you could, if you went fast enough still stutter. Though with just the 4 bit window we end up code bunches already at 3. Since `X_X_X` becomes `_XXX_`. If we increased the max move to 2 then `X_X_X_X_X` could become `__XXXXX___` though it might become `XXX_____XX` or so, since I didn't really predict what would happen with the shifting window at moves of size 1.  And I don't know what the algorithm would be for this. But, yeah, if the algorithm were generalized you could set the pulse grouping up to any level and completely get rid of stuttering.

Sophist-UK, 2022-02-10:
The algorithm in code would be  to have a 256 character bytestring and index into it. Here's a bit of analysis for a 3x fix...

Bytes can have 0-4 zeros or ones - the analysis is symmetric, so I'll only look at the first 4 examples.

For 3x the objective is to have no more than 2 transitions from zero to 1 inside (and possibly one on a change to the next byte) and to move bits no more than 2 bits to the side.

Similar for 4x.
 
I am trying to create a spreadsheet to calculate the values as an example.

Sophist-UK, 2022-02-10:
Ok - here is a (i.e. one possible) bytestring that can be used for such a translation where you move a bit up to 2 positions and have at most 3 on/off transitions per byte (2 inside the byte and possibly one on the transition from byte to byte). I believe this will be A) not visible on the burn and B) pretty much avoid stuttering altogether and C) very quick to execute.
```
b'\x00\x01\x02\x03\x04\x06\x06\x07\x08\x06\x06\x07\x0C\x0E\x0E\x0F\x10\x06\x0C\x0E\x0C\x0E\x0E\x0F\x18\x0E\x1C\x0F\x1C\x1E\x1E\x1F\x20\x0C\x0C\x0E\x18\x0E\x0E\x1E\x18\x1C\x1C\x1E\x1C\x1E\x1E\x1F\x30\x1C\x1C\x1E\x1C\x1E\x1E\x1F\x38\x1E\x3C\x1F\x3C\x3E\x3E\x3F\x40\x81\x81\x83\x81\x83\x83\x87\x30\x1C\x1C\x87\x1C\x1E\x1E\x1F\x30\xC1\x38\x3C\x38\xC3\xC3\xC7\x38\x3C\x3C\x3E\x3C\x3E\x3E\x3F\x60\xC1\xC1\xC3\xC1\xC3\xC3\xC7\x38\xC3\x3C\x3E\x3C\x3E\x3E\x3F\x70\xE1\xE1\xE3\x78\xE3\x7C\xE7\x78\xF1\xF1\xF3\x7C\x7E\x7E\x7F\x80\x81\x81\x83\x81\x83\x83\x87\x30\x83\x83\x87\x83\x87\x87\x8F\xC0\xC1\xC1\xC3\xC1\xC3\xC3\x8F\xE0\xE1\x3C\x3E\x3C\x3E\x3E\x9F\xC0\xC1\xC1\xC3\xC1\xC3\xC3\xC7\x70\xC3\xC3\xC7\xC3\xC7\xC7\xCF\xE0\xE1\xE1\xE3\x78\xE3\xE3\xE7\x78\x7C\x7C\x7E\x7C\x7E\x7E\x7F\xC0\xC1\xC1\xC3\xC1\xC3\xC3\xC7\xE0\xC3\xC3\xC7\xC3\xC7\xC7\xCF\xE0\xE1\xE1\xE3\x78\xE3\xE3\xE7\xF0\xF1\xF1\xE7\xE3\xE7\xE7\xDF\xE0\xE1\xE1\xE3\xE1\xE3\xE3\xE7\xF0\xF1\xF1\xE7\xE3\xE7\xE7\xEF\xF0\xF1\xF1\xF3\xF8\xF3\xF3\xF7\xF8\xF9\xFC\xFB\xFC\xFD\xFE\xFF'
```
This is created from the following mapping:
From |    | To
-- | -- | --
00000000 | -> | 00000000
00000001 | -> | 00000001
00000010 | -> | 00000010
00000011 | -> | 00000011
00000100 | -> | 00000100
00000101 | -> | 00000110
00000110 | -> | 00000110
00000111 | -> | 00000111
00001000 | -> | 00001000
00001001 | -> | 00000110
00001010 | -> | 00000110
00001011 | -> | 00000111
00001100 | -> | 00001100
00001101 | -> | 00001110
00001110 | -> | 00001110
00001111 | -> | 00001111
00010000 | -> | 00010000
00010001 | -> | 00000110
00010010 | -> | 00001100
00010011 | -> | 00001110
00010100 | -> | 00001100
00010101 | -> | 00001110
00010110 | -> | 00001110
00010111 | -> | 00001111
00011000 | -> | 00011000
00011001 | -> | 00001110
00011010 | -> | 00011100
00011011 | -> | 00001111
00011100 | -> | 00011100
00011101 | -> | 00011110
00011110 | -> | 00011110
00011111 | -> | 00011111
00100000 | -> | 00100000
00100001 | -> | 00001100
00100010 | -> | 00001100
00100011 | -> | 00001110
00100100 | -> | 00011000
00100101 | -> | 00001110
00100110 | -> | 00001110
00100111 | -> | 00011110
00101000 | -> | 00011000
00101001 | -> | 00011100
00101010 | -> | 00011100
00101011 | -> | 00011110
00101100 | -> | 00011100
00101101 | -> | 00011110
00101110 | -> | 00011110
00101111 | -> | 00011111
00110000 | -> | 00110000
00110001 | -> | 00011100
00110010 | -> | 00011100
00110011 | -> | 00011110
00110100 | -> | 00011100
00110101 | -> | 00011110
00110110 | -> | 00011110
00110111 | -> | 00011111
00111000 | -> | 00111000
00111001 | -> | 00011110
00111010 | -> | 00111100
00111011 | -> | 00011111
00111100 | -> | 00111100
00111101 | -> | 00111110
00111110 | -> | 00111110
[... 174 lines cut ...]
11101101 | -> | 11100111
11101110 | -> | 11100111
11101111 | -> | 11101111
11110000 | -> | 11110000
11110001 | -> | 11110001
11110010 | -> | 11110001
11110011 | -> | 11110011
11110100 | -> | 11111000
11110101 | -> | 11110011
11110110 | -> | 11110011
11110111 | -> | 11110111
11111000 | -> | 11111000
11111001 | -> | 11111001
11111010 | -> | 11111100
11111011 | -> | 11111011
11111100 | -> | 11111100
11111101 | -> | 11111101
11111110 | -> | 11111110
11111111 | -> | 11111111

</body>

</html>

```

tatarize, 2022-02-11:
11111101 -> 11111101

This seems a bit off. But, I guess it's just a table update. That or you are manipulating the bits that early since the bitwindow would have plenty of time given the look ahead to do that. That seems entirely reasonable.

Sophist-UK, 2022-02-11:
Eh? That is essentially mapping the bit pattern to itself i.e. not changing it. Why would this be an issue. The idea is simply to reduce the number of laser on/off transitions so that you have fewer U/D instructions and less verbosity in the instructions stream - so if there are 2 or fewer transitions in the byte (this example has two) I didn't change it. That said...

In this example, you could do `11111101 -> 11111110` and save one transition inside the byte. If the following translated byte starts with a 0 you would then save another transition, but if it starts with a 1 you would be back to where you started. On average you would still save one transition.

On this basis, in the above table, any translation ending in '101' should be changed to '110', ending in '010' should be changed to '001', starting in '101' changed to '011' and starting '010' changed to '100'. Since all these start/end sequences have two transitions, it implies that the remaining bits need to be the same, so there will be at most 4 such examples.

Sophist-UK, 2022-02-11:
Actually, there are probably more optimisations that this. For example, `00000101` was translated to `00000110` but would be better translated to `00000011`.

tatarize, 2022-02-11:
I think if you push them forward rather than back you might end up moving them more than 2 places since they could travel forward as the window moves forward

Sophist-UK, 2022-02-17:
Yes - you have to be very careful that when you do a substitution, the pattern created can never get shifted to the left and match again in a way that the same bits move again. That is why we needed to look at 6 bits when we were only ever moving a single bit to the right - and it did my head in for just a single bit, for a double bit move you might need 8 or 10 bits and be very careful not to have a matching bit pattern repeat.

tatarize, 2022-02-17:
See the results in #821. While not fully realized that is actually improved results for a loss in quality which is what we want here. Your results don't seem to be outperforming regular pulse grouping. It causes clumps of 3 or so which is *already* what pulse grouping did. While I'm not recommending this, the results from my table at least did make things more clumpy and would raster with less buffer underruns.

Sophist-UK, 2022-02-17:
#821 fixes the 1-bit grouping to handle 101101->110011 which was not previously handled. That is pretty much all it does. It does not attempt to do 2-bit groupings.

closed 2022-05-22 by tatarize