meerk40t/meerk40t #826: [Bug]: Reify on a text-element does not save changed font attributes

jpirnay, 2022-02-13:
### Summary Description

If you modify the font-properties of a text element and try to reify the changes, then these will be ignored.

### MeerK40t Version

0.7.5.b.4

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Sophist-UK, 2022-02-13:
SVG text elements are handled incorrectly by MK and / or svgelements, which IMO currentoy makes them unusuable.

If you read the SVG 1.1 text element specification it is complex because it supports text along a path and sub-strings with different attributes, and complex means difficult (or very difficult or extremely difficult) to implement.

It may be that the basic issue(s) with basic SVGText elements is as simple as a reify issue in the first place, or this may simply be an additional issue on top of one or more other basic issues.

tatarize, 2022-02-14:
Reify with regard to text objects is currently meaningless. 

https://github.com/meerk40t/svgelements/blob/345b10e430ecccc6b1b2e8d3ae31367cbad8b749/svgelements/svgelements.py#L7902

It calls to code for the subobjects 
https://github.com/meerk40t/svgelements/blob/345b10e430ecccc6b1b2e8d3ae31367cbad8b749/svgelements/svgelements.py#L3475
Which reifies the stroke width. (unused by text)
https://github.com/meerk40t/svgelements/blob/345b10e430ecccc6b1b2e8d3ae31367cbad8b749/svgelements/svgelements.py#L3372
Which invalidates any `._length` values that might have been set. (unused by text)

Reification makes an object that can utilize those properties as part of it's real properties to do so. This would actually seems like you might be right. There very well could be a font-size property that could be used rather than the matrix size as such and thus reification of a text object would take the matrix size properties and make them into font-size values. Assuming font-size can be allowed to scale to very large amounts.

tatarize, 2022-02-14:
Riffing off this idea I added `-s size` to the text object in the above PR. We can see that:
 * `text -s 2000 Hello`
 * `text "Hello" scale 124`

 End up being about equal.  I would expect a default 20 pt font to be 100x larger than a 2000 pt font.

Also the textproperties window are not made for fonts of that size. They start getting *very* iffy.

tatarize, 2022-02-14:
If we assume pt fonts are correct and correspond to size it seems like a scale factor of scale(7.831622, 7.831622) is actually correct for this. I know 72 pt fonts should be about 1 inch high and scaling the text to an inch gave me that as the scale-factor. Though it is somewhat clear there's a couple different ways to do these fonts sizes and I'm not sure if the above PR isn't mostly correct for that stuff, it would make sense if we could scale our font size based on our scale factor and that is what reify might need to correctly do in that situation.

Sophist-UK, 2022-02-15:
I note in the SVG file I use that the bbox for text includes margins that are not there in Inkscape (so I assume that this is also a bug - not sure whether it is related to size / matrix issues or not).

a. It will cause raster optimisations based on bbox to be incorrect;
b. It does allow a user to specify a white background that will mask out a black area underneath. But IMO this only really makes sense if the margins can be specified by the user and AFAIK they can't.

In my test file, I have two elements, each an identical sized letter R, but one is text and the other has been converted in Inkscape to a path. The Path one is 8.52mm x 9.67mm in size, the Text one is 12.52mm x 19.4mm in size - the letter itself is c. 10.88mm x 12.35mm, with the remainder being horizontal and vertical margins which look by eye to be symmetric.

jpirnay, 2022-05-22:
Most of it should be fixed, the advanced features like text on a path remain unsupported

tatarize, 2022-05-22:
You should have the ability to close whatever issues here that are settled. Doubly so for this once, since you opened it. The meaning of reify didn't quite apply as a thing for text objects other than maybe setting up the font point size based on the scale. The newer font stuff passes a lot more tests for the WC3_SVG_11 with notable exceptions:

* changed baselines
* text on path
* vertical text (there's some vertical svg commands that aren't really a thing).
* the size is set with differently because you're using left-to-right fonts like hebrew,
* some stuff in calculation of which font to use based on a cascade of different attempted values
* strike through and underline which apparently do exist in wx, but not as attributes exactly but are buttons in the fontdialog.

But, generally font engine stuff is *hard*, harder still getting something like freetype imported and used to do the direct vector conversion from the font to vector stuff which is certainly possible and I've run a couple tests on it (there's a couple example scripts that draw fonts in matplot to show it accessed the geometry), and other things like hershey fonts (which are really useful for lasering) but would require their own parser.

closed 2022-05-29 by jpirnay