meerk40t/meerk40t #832: [Suggestion]: Sync property panels with Tree-Selection

jpirnay, 2022-02-14:
### Summary Description

If you have opened a property window for an element (i.e. by double click) this window remains open and is still linked to the original element, even if the user selects another element (of the same type or another). Chnages will still be possible and propagate to the very first element. This is very confusing and the property window should follow the focus to the newly selected element.

IMHO: the property panel class needs to listen to 
>self.context.listen("emphasized", self.on_emphasized_elements_changed)

and derive from there which node in the tree has been clicked. Unfortunately I can't figure out how to establish this. If you have a hint I would be willing to implement the changes in the different property panels.

### Current functionality

see above

### Future functionality

Link the property windows to the selection in the Operations/Element-Tree

### MeerK40t Version

0.75b4

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tatarize, 2022-02-14:
This could be done. And you're right it would need to change on the emphasis change there. So when they change the emphasis we would check and make sure we are properties for the emphasized objects. I completely see what you're getting at here. There are, however, some windows that might not apply, and they are currently somewhat hard-coded. The pathproperties window, the operation property window, text properties window, etc. These are currently hard-coded and it wouldn't make sense to show them, or update them with the selection.

This seems like it might want to be deferred to 0.8.x code. Since within that code base we are using more notebooks. And the best solution I can come up with for this would be to have the property window be basically dynamic. There's currently a requirement for this since 0.8.x is actively doing this for some things like OperationProperties. We have some properties in OperationProperties that currently do not apply correctly to all drivers. For example a Galvo driver might want a wobble attribute whereas d_ratio is fine property to have for M2 Nano drivers.  So the properties have pages:

[image]

Now here we would have our same properties window but we'd want to dynamically create, switch, and remove panels. So we'd call the window Property and have Operation-Property as one of the windows. We could then, dynamically add the Operation-Properties and Path-Properties when they highlight an operation that has paths in it. And these could be swapped out based on whatever objects are in the current lookup. 

---

We might be able to class objects correctly where the the OperationProperty window just updates to the other operations when we change what operation is selected, and simply maintain the last valid operation when if we select a path.  If you type `channel open signals` in console you can see the signals bouncing around. And yes, so long as the `emphasized` signal is triggered on an object that is a LaserOperation type you should actually be perfectly fine to call the same sort of activation code we use when we swap over the operation.

In 0.8.0 you would flag any function with `@signal_listener("emphasized")` in 0.7.x you'd want to hook the lifecycle open and lifecycle close events (whatever those are depends on the type of thing we're dealing with) with `context.listen("emphasized", self.myfunction)` and the close with `context.unlisten("emphasized", self.myfunction)`. Then your myfunction needs to check that the emphasized content is a LaserOperation node type and then call the activation code.

tatarize, 2022-02-14:
The basic method for this would be edit OperationProperty.py

```python

    def finalize(self):
        pass

    def initialize(self):
        self.set_widgets()
        ...
```
To hook into the emphasized thing and do the basic checks you want.

```python

    def my_function(self, origin, emphasized_object, *args):
        from meerk40t.core.elements import LaserOperation
        if isinstance(emphasized_object, LaserOperation):
            self.operation = emphasized_object
            self.set_widgets()

    def finalize(self):
        self.context.unlisten("emphasized", self.my_function)

    def initialize(self):
        self.context.listen("emphasized", self.my_function)

        self.set_widgets()
        ...
```

This will give

tatarize, 2022-02-14:
I must admit quickly hacking that in, does show that you may be right. That seems really nice and responsive even without a more generalized solution for all the different things put in a notebook.

jpirnay, 2022-02-14:
I am currently implementing these changes (and started with and group and path) but have encountered an error if you are closing the panel and then continue selecting in the tree. I have caught the error for now but that needs a review... Will create a draft PR tomorrow for you to check.

Sophist-UK, 2022-02-15:
This would be good. AFAIK you can still cause a crash by opening an Operation Property window, delete the operation and then make a change in the properties window.

My take on this is that you would need to tap into a signal for tree selection changes, and if the operation currently associated with the window is deselected (or deleted) then
a. if another operation is selected you display that (or the first of these) instead; or
c. if no other operation is selected, then you either grey everything out or close the window (not sure which would be most appropriate).

closed 2022-05-18 by jpirnay