meerk40t/meerk40t #840: [Suggestion]: From Facebook - Operations File Node, Level in Tree

tatarize, 2022-02-17:
### Summary Description

The suggestion offered was to add into the operations a grouping for the particular file node this came out of or the grouping. So if we had multiple files and these got placed into the same operation we'd get another node within our references for the grouping within the references.

### Current functionality

Currently we display the reference nodes ("opnode") in a flat list of all of the elements.

### Future functionality

We would display the reference node but parallel the structure of the elements. So if they had any groupings we would also give those groupings.

### MeerK40t Version

0.7.5001

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

Sophist-UK, 2022-02-17:
1. This would likely complicate the existing code, and would be quite a big internal change which would need reasonable testing.
2. There are much higher priority items IMO.

tiger12506, 2022-02-18:
I think this just makes the interface messier. I don't see

closed 2022-02-18 by tiger12506