meerk40t/meerk40t #841: [Suggestion] Provide Lhy config values for acceleration boundaries

Sophist-UK, 2022-02-17:
Today I was rastering a large image at 250mm/s and the acceleration was too great for the steppers and it was shifting.

(At this stage I am assuming that this is not MK providing a twitch direction incorrectly - but the noise was very reminiscent of previous vector twitch noises so I do wonder.)

The default acceleration (distance) for raster at 250 is 3 so I reduced speed to 200 and set acceleration 4 to increase the distance it has to accelerate / decelerate. 

It occurs to me that we should provide a Device Config override for the standard acceleration break-points to automatically apply an override acceleration. The algorithm would then be:

1. Operation manual acceleration - use that
2. Device config table - if enabled calculate acceleration number from that
3. Otherwise let Lhymicro use its own default.

Indeed we could populate the table with the default values and provide a reset button or right click reset action.

Sophist-UK, 2022-07-20:
The issue as originally reported was a twitching issue due to alternative raster instructions - turning that off avoided the shifts.

But now that we have a hardware agnostic driver model, it would be generically useful for each actual driver to provide acceleration / speed values (max speeds, acceleration distances (m2) / or actual values (other)) etc. through a common API.

tatarize, 2022-07-20:
Well, other devices don't have these boundaries they tend to handle acceleration internally. Only M2 Nano would need that chart and to tweak it as such.

Sophist-UK, 2022-07-20:
Ideally it is still useful to know - e.g. for image margin optimisation, or to know if an image is so close to the end stops that the head will hit them during deceleration etc. 

That said, if we took a worst case situation we could probably e.g. use a fixed value of 1cm or 500mil or whatever.

closed 2024-10-04 by jpirnay