meerk40t/meerk40t #843: [Bug] Greedy optimisation of images doesn't account for direction

Sophist-UK, 2022-02-18:
The greedy optimisation of images (and possibly images generated from rasters) does not take into account the direction of the raster. This is (of course) only noticeable when doing Merge Operations.

I haven't yet checked the code but suspect that the start/end point is based on bbox whereas it needs to be set based on direction. I am unclear which end the rastering starts at (or whether it is clever enough to start the raster from the closest end to the head position, though in the big scheme an average of half a sweep is neither here nor there in overall optimisation).

Sophist-UK, 2022-07-20:
Perhaps we could do with extra raster / image direction options of "vertical" or "horizontal" where "vertical" means MK has a free choice between B2T and T2B and "horizontal" means MK has a free choice between L2R and R2L - and the greedy candidate routine then yields four choices for the start position (based on the first and last pixels in the first and last scan lines of the image).

tatarize, 2022-07-20:
Hm. I guess, but that's going to save what 6 seconds at the very worst. The reason this is somewhat wrong is that rasters are such a giant time suck that it sort of doesn't matter which side you start or end at since you're going to save a second or two and you're doing things like rastering for minutes on end.

Sophist-UK, 2022-07-20:
That is true. Vector optimisation is definitely worthwhile, but rasters not so much so.

closed 2023-07-04 by tatarize