meerk40t/meerk40t #844: [Bug]: "[WinError 193] %1 is not a valid Win32 application" on dev usb_connect

jds8086, 2022-02-19:
### Summary Description

When trying to connect to the M2, i'm getting [WinError 193] %1 is not a valid Win32 application. I do not get this behavior when running the precompiled  binary. I followed the steps listed on in https://github.com/meerk40t/meerk40t/wiki/Install:-Running-from-source

### Additional Details

Using CH341DLL driver

### Crash logs

```shell
MeerK40t crash log. Version: 0.7.5001 src on Windows:AMD64 - 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\device\ch341\windll.py", line 151, in __init__
    self.driver = windll.LoadLibrary("CH341DLL.dll")
  File "c:\users\user\appdata\local\programs\python\python38\lib\ctypes\__init__.py", line 451, in LoadLibrary
    return self._dlltype(name)
  File "c:\users\user\appdata\local\programs\python\python38\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\gui\panes\consolepanel.py", line 141, in on_entry
    self.context(command + "\n")
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\kernel.py", line 154, in __call__
    return self._kernel.console(data)
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\kernel.py", line 1896, in console
    self._console_parse(command, channel=self._console_channel)
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\kernel.py", line 1976, in _console_parse
    data, remainder, input_type = command_funct(
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\kernel.py", line 1326, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\device\lhystudios\lhystudiosdevice.py", line 396, in usb_connect
    output.open()
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\device\lhystudios\lhystudiosdevice.py", line 1664, in open
    self.connection = self.ch341.connect(
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\device\ch341\ch341.py", line 135, in connect
    handlers.append(WinHandler(channel=self.channel, status=self.status))
  File "C:\meerk40t\meerk40t-0.7.5001\meerk40t\device\ch341\windll.py", line 159, in __init__
    raise ConnectionRefusedError
ConnectionRefusedError
```

### MeerK40t Version

0.7.5001

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

jds8086, 2022-02-19:
> This is because of a mismatch between the CH341DLL.dll being 32 bit and the 64 bit version of python/windows. I think I've only seen this happen with compiled versions though. This is apparently run from source and kicking such an error. It might fix if you install the driver built into K40Whisperer since then meerk40t will use the LibUsb driver instead.

I installed the libusb0 driver via K40Whisperer install, had the same error. I then tried the driver from https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/ and got the same result. Would using 32 python be a possible solution/work around?

jds8086, 2022-02-19:
This is working as it should in a 32 bit venv (3.8).

tatarize, 2022-02-19:
Probably yes. Though I would think the libusb should come in the 64 bit version but admittedly I have not done extensive tests of this.

tatarize, 2022-02-19:
@tiger12506 are there going to be issues here going forward? Do we need to provide the same sort of code for K40Whisperer with the install for 64bit libusb or hunt down the 64 bit CH341 dll and include it (though it wouldn't be installed natively I think). We should totally be able to run *something* in pure 64 bit, right?

Sophist-UK, 2022-02-19:
IMO not only should we fix this for 64bit python run from source, but we should probably create 64 bit python versions of the executable. I don't think that CH341 was the reason we don't currently do this - I think that there was some other kind of issue.

tiger12506, 2022-02-19:
No, the 64-bit version of ch341 was pretty much the reason that we don't have 64-bit builds. That's just something that has fallen in the cracks, there's no good reason for not having 64-bit, except having 64-bit versions of those dlls on all platforms where it is necessary.

Sophist-UK, 2022-02-20:
http://www.wch-ic.com/downloads/CH341SER_ZIP.html

jds8086, 2022-02-20:
> http://www.wch-ic.com/downloads/CH341SER_ZIP.html

This driver does not appear to be compatible, the inf does not list PID 5512.

tiger12506, 2022-02-20:
It's easy enough for anyone to test. Just remove the x86 line from the build yaml and run a build. Test it out to see if it makes the connection using the dll, or if it gives the error.

Sophist-UK, 2022-03-12:
Hmm - as far as I can tell from internet research (as I am nowhere close to a K40 ATM), the windows device id is `USB\VID_1a86&PID_5512`. I will see if I can find 64-bit drivers or otherwise research how to call a 32-bit driver from 64-bit python.

Sophist-UK, 2022-03-12:
> > http://www.wch-ic.com/downloads/CH341SER_ZIP.html
> 
> This driver does not appear to be compatible, the inf does not list PID 5512.

It is a CH341 to serial driver - hence it not being the right one. There is no parallel driver on the wch-ic.com website but I did find one elsewhere which had 32 and 64 bit versions of the drivers (*.sys) but a single DLL to interface to them. So I guess we need to work out how to interface python64 to a 32 bit dll.

Sophist-UK, 2022-03-12:
There is no 64-bit DLL that I can find, and as far as I can see a 64bit executable cannot load a 32-bit DLL. We therefore have the following choices for delivering a 64-bit version of MK:

a. Support only LibUSB on 64-bit python - this is the simplest option as does not require any re-architecture;
b. Put the drivers into a separate 32-bit executable and use interprocess communication to link the two - although this will be significantly more work, it does have a few advantages: A) it allows us to have a 64-bit address space for growth in functionality i.e. to hold multiple projects open at one time and process larer images; B) it avoids issues of GIL and multi-threading whereby e.g. a long running simulation holds the CPU and the driver gets insufficient CPU time to process the CutCode into lhy instructions, and this will be esascerbated as we support spooling to multiple simultaneous devices; and C) it allows the drivers to be 32-bit and thus support CH341.

My gut reaction is that we should leave rearchitecture to a later date, and at this stage we should:

1. Support CH341 only from Windows 32-bit Python, with all other versions only supporting LibUSB; and
2. We should bundle the LibUSB driver installers in with the (Windows?) executable and provide console and GUI commands to run the installer.

Sophist-UK, 2022-03-12:
Actually there  may be a third option - we create a proxy 64-bit CH341A.dll using [LegacyWrapper](https://github.com/CodefoundryDE/LegacyWrapper). However I am unconvinced that this is worth the effort.

Sophist-UK, 2022-03-12:
I am going to code up a PR on the basis of option 1 above, and I will try to rework how the DLL is included so that it is accessible when running from source (as source download from Github or from Git or from Pypi) and also try to make it possible to have both 32 and 64 bit executables.

closed 2022-07-20 by tatarize