meerk40t/meerk40t #846: [Suggestion]: Allow user to enable / disable elements / groups etc.

Sophist-UK, 2022-02-19:
Title says most of it.

At present if I have an SVG with several separate pieces to cut contained in it that I wish to burn on separate (offcut) pieces of material, I need to delete some, burn, then reload and delete others, and burn again.

It would be nice to be able to disable some elements and then reenable them and disable others.

Also, this would allow elements hidden in Inkscape to be loaded disabled.

I am guessing that this would not be too difficult to achieve. (Probably limited to elements.py and svgio.py.)

jpirnay, 2022-04-29:
Can be solved with moving stuff into regmarks

tatarize, 2022-05-09:
This is mostly settled now, but might need some testing to ensure it works fairly well. Anything that is marked `visibility: hidden` should actually be added only to regmarks which will save this data back out on save. But, elements in regmarks are not able to be selected or manipulated in many ways.

This should allow you to use these marks as a guide for the data but not burn them etc, in fact they can't be referenced within the regular tree structure. You'd need to add them to elements. I guess these changes happened after Feb 19th. I'll hold off closing this until things get a bit more testing for robustness.

jpirnay, 2022-07-29:
You can lock elements and even prevent moving locked elements around (preference setting) in the latest main

Sophist-UK, 2022-07-29:
The use case I had in mind was for designs where you have e.g. various images, only one of which you want to select at burn time.

At present you load the file and delete the images you don't want from the tree or remove them from the Image operation.

The purpose of this request was to make it easier to e.g. burn one image and then switch to a different image by enabling / disabling the relevant elements or groups (rather than reload and then delete a different set).

I haven't yet had a chance to look at these enhancements in the app, but from the descriptions here they do **not** appear to meet the requirement as stated.

jpirnay, 2022-07-30:
Lock will not prevent an object from being cut, if it's in an operation and that that operation is enabled. But that could be the way to go: assign these images to different operations and enable/disable them at will

jpirnay, 2022-07-30:
I overlooked though your reference to inkscape hidden Tag, that's a  valid point, so I will pick this one up

closed 2024-10-04 by jpirnay