meerk40t/meerk40t #850: [Bug?] wxmscene registers a console. Is this correct?

Sophist-UK, 2022-02-21:
https://github.com/meerk40t/meerk40t/blob/3dd762b0a33d5889487ce00748b4c71b698d103d/meerk40t/gui/wxmscene.py#L36 reads:
```
    context.register("pane/console", pane)
```

This is the same registration as https://github.com/meerk40t/meerk40t/blob/3dd762b0a33d5889487ce00748b4c71b698d103d/meerk40t/gui/panes/consolepanel.py#L27 . Is this correct or a bug?

tatarize, 2022-02-21:
Bug. It should likely be pane/scene. It doesn't matter as much since the latter code likely hits later and you usually can't disable the scene anyway.

Sophist-UK, 2022-02-21:
I tried changing it to pane/scene and MK crashed on load with some sort of wxwidget assert. Gonna have to leave this one to you to sort.

tatarize, 2022-02-21:
```python
def register_panel_scene(window, context):
    # self.notebook = wx.aui.AuiNotebook(self, -1, size=(200, 150))
    # self._mgr.AddPane(self.notebook, aui.AuiPaneInfo().CenterPane().Name("scene"))
    # self.notebook.AddPage(self.scene, "scene")

    panel = MeerK40tScenePanel(window, wx.ID_ANY, context=context)
    pane = aui.AuiPaneInfo().CenterPane().Name("scene").Caption("Scene")
    pane.dock_proportion = 600
    pane.control = panel

    window.on_pane_add(pane)
    context.register("pane/scene", pane)
```

The problem is the code elsewhere uses the set Caption, which doesn't exist for pane to make a menubar value.

Corrected like that and you can toggle the scene on and off, which is a little weird.

Sophist-UK, 2022-02-22:
I fixed this a different way - by not adding to a menu anything without a caption.

commit 9a69c57f (2022-02-23): Merge pull request #851 from meerk40t/fix_pane-scene

Fix #850 - extra code is a simplification to avoid taking a circular route.
commit bc82caaa (2022-02-23): Fix #850
closed 2022-02-23 by Sophist-UK