meerk40t/meerk40t #862: [Bug]: Wrong raster position

dorsi90, 2022-02-27:
### Summary Description

To replicate the problem:
In inkscape I made a rectangle with a red stroke (cut) and a black text, the latter converted into "Object to path" because otherwise on meerk40t the text is in the wrong position and results with larger font size. After importing the file on meerk40t the rectangle is correctly recognized as cut and the text as raster but running the job the raster is done in the wrong position. However, if I click on make raster image before doing the job, it works correctly.

I would like to point out that my machine has a moshiboard MS10105 v4.85 and despite some other small bugs I found  you have done a really good job.

In the second pictures below i show the test without clicking "make raster image" and after do that.

[image]
[image]

### MeerK40t Version

0.7.5000

### MeerK40t Type

Executable

### Your Operating System

Windows

Sophist-UK, 2022-02-27:
@dorsi90 Guiseppe: Thanks for such a clear description of the issue. Can you please open the Device, Config and Execute Windows (and position them so they don't overlap) and attach a further screenshot?

Thanks.

dorsi90, 2022-02-27:
@Sophist-UK thank you for responding so quickly.
In the first picture the execute is before clicking on start, update or simulate buttons and the second is after.

[image]
[image]

Sophist-UK, 2022-02-28:
Thx. @tatarize  may be the only Moshi expert who can look at this.

tatarize, 2022-02-28:
I think it's an iteration of the problems with text objects and has nothing to do with Moshi. In Inkscape, convert text to paths before exported.

dorsi90, 2022-02-28:
Thanks @Sophist-UK.
I would also like to point out that by clicking "make raster image" and then after the raster image is created as an element I do not move it to the raster or image I simply start the job and it works perfectly but if I delete the element just created and I start the job again the laser head moves on the x-axis out of the working area and bumps against the rail.
The machine is almost 10 years old and i never replaced anything not even the laser tube but if the problems are related to moshiboard maybe it's time to buy the M2 Nano which has more support.

dorsi90, 2022-02-28:
Hi @tatarize thanks for your reply.
As I said in the post above I noticed the problem with the text object that have font size bigger and wrong position so i converted it to path and the render on meerk40t is perfect but when start the job without click on make raster image the laser burns in wrong place.
As soon as possible i do more tests to be clearer

Sophist-UK, 2022-02-28:
When you click Simulate, is it in the correct place or in the wrong place?

Sophist-UK, 2022-02-28:
Also, do you do an automatic home at the beginning of every executed job? In the Execute Window, select Automatic/Before/Home.

dorsi90, 2022-02-28:
Hi @Sophist-UK in the simulation is in correct place as the picture below
[image]

Also with before->home enabled the raster is in wrong place.

After other tests i saw that when i click start the first time the raster is in wrong place but if i click start a second time the raster is in the right place as the picture below
[image]

Sophist-UK, 2022-02-28:
Since the simulation looks OK, I am guessing that the issue is in the moshi driver rather than at a higher level in the stack. This area of the code is not something I can help with, and I think @tatarize is the person to help here.

dorsi90, 2022-02-28:
@Sophist-UK i tried the text with blue stroke and with engrave mode the position is in the right place so only raster is affected. Do you have any tested svg files with cut and raster or cut and image inside to test with my board? Just to be sure that is moshi related problem or maybe my os

Sophist-UK, 2022-02-28:
Highly unlikely to be your os. I am not at the computer right now - it's bed time here - but you can find a decent test file in one of my recent closed PRs.

dorsi90, 2022-02-28:
Thanks @Sophist-UK tomorrow I see if I can find it.
Also in italy is bed time since few hours ago but i go to sleep very later. I'm a developer and designer of boards so I spend a lot of time on the computer, almost the whole day, but unfortunately I don't know python language :(

tatarize, 2022-03-01:
Okay. Begrudgingly I must say there's some a chance this *is* Moshi related. There's some references in Moshi to a sort of coordinate offset and if you set that you tend to get a different location than you otherwise would get. In this case it *could* be the case that running the raster twice causes the offset in the Moshi card itself to offset in the code offset *and* to also have an additional offset. 

Since it doesn't show up in Simulation then it basically *should* be driver related. The simulation is setup to be very explicit about what exactly will be run on the laser and if what matches between the simulation and driver is mismatched then it really *should* be the case that the driver or physical hardware is at fault.

tatarize, 2022-03-01:
Regretfully I hooked the makeshift wires I have for the moshiboard up wrongly and a little puff of smoke drifted up from the board as it sat there doing nothing. 

I am somewhat sure I know generally where the bug reported here would be. Likely homing the device between runs would avoid that. And I do have a lot of caps of the board and I do have a fake board that I can hook up to pretend to be a moshiboard and capture more caps from the Moshidraw. However without a functional board on hand I am unsure how successful I could be supporting Moshi going forward.

tatarize, 2022-03-01:
I at least went out of my way to be highly descriptive in the code:

```
    The protocol for sending rasters is as follows:
    Check processing-state of board, seeking 205
    Send Preamble.
    Check processing-state of board, seeking 205
    Send bulk data of moshiblob. No checks between packets.
    Send Epilogue.
    While Check processing-state is 207:
        wait 0.2 seconds
    Send Preamble
    Send 0,0 offset 0,0 move.
    Send Epilogue
```

So does the raster code do the Send 0,0 offset 0,0 between the two rasters? This would be effectively just a home command. It should do a raster, home, raster. If it's correctly adhering to the protocol I cited there. If not, does doing that actually fix it. Home the device between rasters to avoid a double offset?

dorsi90, 2022-03-01:
Hi @tatarize after having made other tests I have seen that it does not always work by start the job twice i just got lucky and it does not matter if I set home after and before or clicking the home button before start the job, actually the raster is almost always in the wrong position, i also tried install the libusb driver but it is always the same.
I finally understand where is the problem but later i want make much more test to be sure.
however if you see my last picture posted here the raster has 4 paths basically one for each letter and if i click make raster image and transfer the raster image generated in image operation it works perfectly but also if i click on "merge items" as in the picture below

[image]
[image]

So with one path element it works without any problem.

tatarize, 2022-03-01:
Hm. This sounds a bit like it might be dividing up the raster into multiple pieces as both of the cited methodologies would prevent that from happening. While that code shouldn't have that effect, it might be able to. I think I'm going to need to write an emulator for moshiboards to test that the code is being consistent.  It does occur to me that @Sophist-UK  didn't make an off switch for the raster splitting functionality. To test if there's some weird interaction between the moshi driver and the code that creates multiple rasters. 

Though you should tell whether it rasters each letter individually or as a whole in the case where it screws up.  It really seems like the offset should be doubling though in that case the letters would be offset by about 2x the regular distance and your examples didn't quite show that to be the case.

Sophist-UK, 2022-03-01:
I see that you are running the 0.7.5000 executable version of MK. I have a feeling, but haven't yet checked, that my raster optimization PR was merged **after** 0.7.5000 was created, in which case that would not seem to be the cause.

Before my optimisation PR - i.e. how MK 0.7.5000 deals with rasters - all the elements in a Raster operation are converted into a single image pretty much by exactly the same code that Make Raster Image uses. This happens way before it gets to the moshi driver, so there really should be no difference between a single path and multiple paths and an image you have created using "make raster image".

If my code has been merged, then it is quite possible that it cannot determine the acceleration/deceleration distance for a moshi board and uses zero, in which case it might well raster each letter separately, but a careful look at the simulation shows that there is only one travel move from the single raster to the cut path. You can of course verify this by watching to see whether it rasters each letter separately or rasters them as a single word.

With hindsight, I think @tatarize is correct that there should have been a hidden or visible setting to enable this optimisation to be disabled. I will add that to the list of minor final PRs I plan to write for 0.7.x.

Sophist-UK, 2022-03-01:
I am also wondering whether moving the part somewhere else on the bed causes the offset to change.

Sophist-UK, 2022-03-01:
I have just checked - my raster optimization PR is **not** included in 0.7.5000,

dorsi90, 2022-03-01:
@Sophist-UK i've already tried to move the part to other position on the bed nothing happen.
The rasters is done as single word so your PR in not included.

@tatarize the offset is 2x the regular distance but only in vertical axis

tatarize, 2022-03-02:
The 2x offset of the regular distance only being in the Y direction is *very* weird. I looked at the x and y values and didn't see anywhere where it would give a zero rather than the given value. I might have to audit the output of this by the driver with some solid emulation. I checked some of my pcaps for the moshiboard and they seemed to conform to my current understanding there.

It's likely something to do with the driver itself or a misunderstanding of the moshi driver itself.

While my fried moshiboard makes harder to check if it's hardware related. But, I can still double-check the driver data send and emulate the moshidraw well enough to get it to send additional data. I'll certainly do the former at a minimum.

dorsi90, 2022-03-02:
Hi @tatarize I bought the m2 nano board and it should arrive within 10 days :)
I would investigate a little bit so I compiled meerk40t from the source and after an initial compilation error due to a conversion of floating number to integer in the file kernel.py already reported to Sophist-UK now I have the same problem of another user on the loading the dll ch341 that is located in the system32 folder but for some reason does not find it.
Are you applied a reverse voltage on your moshi?

tatarize, 2022-03-02:
The M2 Nano has 4 pins of connectors. These are like `24v-G-5v-L` on the M2-Nano and on the Moshi board there's `G-L-5V-24v-G` as a 6 pin power things. So they have the same pins but they are in different order and in different numbers. I plugged them in wrong which caused it to pop one little connector D12 on the board. And stop working. Finding the connector thing might actually mean it's fixable.

Regardless whether the board could be fixed or not, I still need to check the blobs it makes to send to the board and see if the re-raster bug is easily fixable. All the connections with an M2 port over really easy but you need to wire the power connection itself (and not hook it up in the wrong order).

dorsi90, 2022-03-02:
@tatarize on my moshiboard D12 is just a led diode that certainly does not create problems in the operation, I think you have damaged other components.

I've tested the raster optimization PR of @Sophist-UK but won't solve the issue and in my case has the problem that rastering with high speed small parts makes the laser perform strange movements and takes much longer.

Sophist-UK, 2022-03-02:
I do not have the metrics for the moshi acceleration / deceleration and so it probably breaks the rasters into very small pieces on a moshi. I guess we need to have a different default other than zero for devices where we do not have a metric. Leave that one with me.

Sophist-UK, 2022-03-02:
> pop one little connector D12 on the board

I am guessing that D stands for diode.

dorsi90, 2022-03-02:
> I am guessing that D stands for diode.

Yes it is. I think he mean C13 which is near D12 and is an electrolytic capacitor that if connected with reverse polarity it explodes and you die from fear :)

tatarize, 2022-03-02:
[image]

The now white bit there, pretty sure it's a little tiny thing that is now blown up. It says D12+ or something.

dorsi90, 2022-03-02:
@tatarize you have different version of moshiboard i think my board is newer. i've version 4.85!
Anyway D12 is a diode led and is gone, no matter, you need to test D11 on the right. i see it's a little bit brown.

dorsi90, 2022-03-08:
Tested the same svg file on lightburn with ruida emulator and the position of the raster is perfect however with emulator i noticed a slow engraving speed and i raise the speed

dorsi90, 2022-03-08:
I tested the same svg file on lightburn with ruida emulator and i noticed a slow speed when engraving and if i raise the speed it is more slower but the raster position is in the right place.
Yesterday i disassembled my power supply and added another connector so when i receive the m2 nano board i can easy switch from moshi to m2 and back to understand which issues are hardware related.

tatarize, 2022-03-11:
Hm. That sort of rules out the M2 since there are weird speed code things in M2 the Moshi has no such compunction. The faster speed is a bigger number full stop. And I did see reports of this elsewhere where the speed didn't match. I did an audit of the speed values and found nothing amiss but it seems there is likely a bug in there somewhere with regard to speed at least.

dorsi90, 2022-03-11:
Yes, i see just now on facebook group that another user report the issue related to the speed in lightburn. However this was just a test to see if in lightburn the raster was in the right position i don't want use lightburn anymore.
So as i can see your moshi code works like a charme i think there is someting wrong in the blob generated from meerk40t. There is some way to export the blob generated?

tatarize, 2022-03-12:
I put off doing the work on this for a bit too long so I ported in the code I used to debug the pcaps of Moshi and found something interesting. While I'm not absolutely convinced it is what's causing it, there is clearly a problem in the protocol when looking at the debug of the blobs. I don't have a way to save the blobs out but I made a branch that does debugging of the program if `mock` is turned on. In the events channels `channel open 1/events` (my moshi is 1 rather than 0)

```
[02:46:57] channel open 1/events
[02:46:57]     Watching Channel: 1/events
[02:47:21] rect 1in 1in 0.2in 0.2in fill black raster
...
[02:50:07]     Raster Mode
[02:50:07]     Raster Header Speed: 150 cm/s
[02:50:07]     Set Location z: 0, x: 999, y: 999
[02:50:07]     Move x: 0 y: 200
[02:50:07]     Move Horizontal x: 1
[02:50:07]     Cut Horizontal x: 11
...
long bunch of raster moves.
...
[02:50:08]     Cut Horizontal x: 200
[02:50:08]     Finished Mode
[02:50:08]     Final Raster Home
[02:50:08]     Rapid Mode
[02:50:08]     Termination.
[02:50:08]     Pushed program to output...
[02:50:08]     Pushed: bytearray(b'\x02\x0e\x00\x00\x00\xe7\x03\xe7\x03\x8a\x00\x00\xc8\x00\x82\x01\x00\xa2\x0b\x00\x82\x0c\x00\xa2\x1b\x00\x82\x1c\x00\xa2+\x00\x82,\x00\xa2;\x00\x82<\x00\xa2K\x00\x82L\x00\xa2[\x00\x82\\\x00\xa2k\x00\x82l\x00\xa2{\x00\x82|\x00\xa2\x8b\x00\x82\x8c\x00\xa2\x9b\x00\x82\x9c\x00\xa2\xab\x00\x82\xac\x00\xa2\xbb\x00\x82\xbc\x00\xa2\xca\x00\x82\xdc\x00\x88\xc6\x00\x82\xc8\x00\xa2\x00\x00\x82\xff\xff\xa2\xfe\xff\x82\xec\xff\x88\xc4\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xc2\x00\x82\xc8\x00\xa2\x00\x00\x82\xff\xff\xa2\xfe\xff\x82\xec\xff\x88\xc0\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xbe\x00\x82\xc8\x00\xa2\x00\x00\x82\xff\xff\xa2\xfe\xff\x82\xec\xff\x88\xbc\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xba\x00\x82\xc8\x00\xa2\x00\x00\x82\xff\xff\xa2\xfe\xff\x82\xec\xff\x88\xb8\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xb6\x00\x82\xc8\x00\xa2\x00\x00\x82\xff\xff\xa2\xfe\xff\x82\xec\xff\x88\xb4\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xb2\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\xb0\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xae\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\xac\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xaa\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\xa8\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xa6\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\xa4\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\xa2\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\xa0\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x9e\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x9c\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x9a\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x98\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x96\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x94\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x92\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x90\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x8e\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x8c\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x8a\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x88\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x86\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x84\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88\x82\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88\x80\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88~\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88|\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88z\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88x\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88v\x00\x82\xc8\x00\xa2\xc7\x00\x82\xc6\x00\xa2\xfe\xff\x82\xec\xff\x88t\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88r\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88p\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88n\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88l\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88j\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88h\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88f\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88d\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88b\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88`\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88^\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88\\\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88Z\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88X\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88V\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88T\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88R\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88P\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88N\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88L\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88J\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88H\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88F\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88D\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88B\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88@\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88>\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x88<\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x88:\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x888\x00\x82\x00\x00\xa2\xca\x00\x82\xdc\x00\x886\x00\x82\xc7\x00\xa2\xfe\xff\x82\xec\xff\x884\x00\x82\x00\x00\xa2\xc9\x00\x82\xdc\x00\x882\x00\x82\xc8\x00\xa2\xfe\xff\x82\xec\xff\x880\x00\x82\x00\x00\xa2\xc9\x00\x82\xdc\x00\x88.\x0 [... cut ...]
[02:55:37]     Pushed: bytearray(b'\x02\x03\x00\x00\x00\xe7\x03\xaf\x04\x8a\x00\x00\x00\x00\x8a\x00\x00\x00\x00\x80\x80\x80\x80\x80\x80\x80')
[02:55:37]     Raster Header Speed: 4 cm/s (03)
[02:55:37]     Set Location unknown: 0, x: 999, y: 1199 (0000,e703,af04) 
```

Now in theory this could totally do the raster issue and get it wrong. I would entirely expect to have returned to the correct position in *vector* mode. It does seem to home but you can't really do absolute moves in raster mode. And there's a couple other differences. If I had to guess this is the reason for the bug.

tatarize, 2022-03-12:
Can you check [moshi-debug](https://github.com/meerk40t/meerk40t/tree/moshi-debug) branch. It should be good but it seems like it should correct the weird moshi bug, and we can close these issues down and merge it in if it does fix them.

dorsi90, 2022-03-12:
@tatarize tested and not working.
I upload some videos to show you what happens.

In the video below as you can see on the first start (head in home position) the raster is in the wrong position, then without homing and clicking start again the head moves to the right position and make the raster in the right place. Every time i click start after the job was finished it works but if i click home button before start again then on start the head moves out of working area and bumps into the rail.
https://user-images.githubusercontent.com/44925234/158020768-31b42636-085b-45a3-8879-e4b92d129929.mp4

In the video below i added Automatic->Before->Home so after the first start that is already in home position the head go to home position and the raster is not in the right place.
https://user-images.githubusercontent.com/44925234/158021142-fba58eb4-ec30-4f1f-82f9-6140c502e787.mp4

I make the first 2 videos with Meerk40t version 0.7.5000
the last one below is with version 0.7.5001 that include your moshi changes and @Sophist-UK path engraving separation. This is with Automatic->Before->Home disabled as the first video.
https://user-images.githubusercontent.com/44925234/158021564-e3890b7c-d176-4b21-9389-14c9b7a9006e.mp4
As you can see for the first start nothing is changed but when start again the head move outside the area until it bumps into the rail.

I think the problem is the homing maybe when it's in home position position meerk40t use an absolute position to move the head and start the job and later use a relative position.

P.S. i don't know why "ST" letters are made together with @Sophist-UK path separation, as i can see in the software all letters are different paths but anyway  this is not the cause for this issue.

tatarize, 2022-03-13:
Hm. Seems even more off. I was hoping Moshi-Debug would fix it. There must be a different issue with the Raster in moshi. Also, I note that the divided lines between the T and E and E and ST weirdly fired the laser in a cut-like fashion. That seems a fairly critical bug. 

If you run that test in Mock with the `0/events` channel open to view the debug code does the log suggest it should be doing this bit of weirdness? That or send me the exact file. So I can test it myself. Seems like if the broken raster-call isn't correct maybe the raster's work differently and is messing up the offset somehow. And I could check moshiboard caps to see if I goofed on those calls.

dorsi90, 2022-03-13:
>Also, I note that the divided lines between the T and E and E and ST weirdly fired the laser in a cut-like fashion. That seems a fairly critical bug.

This issue is only in the version 0.7.5001 after the raster separation but if you see the picture below only the letters printed separately are affected, maybe because they are too small.
[image]

>If you run that test in Mock with the 0/events channel open to view the debug code does the log suggest it should be doing this bit of weirdness?

I'm sorry but i don't know how to do that.

This is the file I always use
[image]

dorsi90, 2022-03-14:
@tatarize i see that if i change the position also the red dot inside the working area in meerk40t moves to the right xy coordinate but if i click home button the red dot not move to xy 0.
This happen only for moshi, if i set lihuiyu home button move the red dot to xy 0 correctly. This is can be tested also without k40 connected.
Is possible that meerk40t not update the position and then generate the coordinate not from xy 0 so introduce an offset?

tiger12506, 2022-03-14:
> > If you run that test in Mock with the 0/events channel open to view the debug code does the log suggest it should be doing this bit of weirdness?
> 
> I'm sorry but i don't know how to do that.

[image]

tatarize, 2022-03-14:
Hm. @dorsi90 Try the latest moshi-debug push in the branch. I tweaked it so that it will not move during the raster with a normal movement. At least that *should* prevent the laser on move over happening or something. And hopefully work.

If there's still a lack of WinDll maybe that just got messed up. But the moshi-debug branch doesn't include Sophist's WinDll change which might have resulted in that.

dorsi90, 2022-03-14:
>Try the latest moshi-debug push in the branch.

Tested and now the T letter is in wrong position while others letters in the right place but the T is engraved with correct speed while others letters very very slow like cut mode
[image]

tatarize, 2022-03-15:
Hm. I may actually need to buy a replacement Moshiboard. I feel like my tweaks should have made things better, but they do not appear to, though I think it may be a requirement to home the device between each use and that may not be happening. I'll check the power routines though that should be fairly easy to check/correct either way.

dorsi90, 2022-03-15:
@tatarize if i run the job with lightburn that works and send you the events, can help you to understand where is located the issue?
I have uninstalled lightburn but if can help i can install it again.

tatarize, 2022-03-16:
Maybe. In that if the same events result in the same result. But, Lightburn there might just be doing hard-rastering which isn't wildly effective. It just does the raster events like a vector since there's no correction for the ruida rastering given that device. Which is something you could already just turn on anyway in settings.

In console, type: `set -p moshi/driver/0 enable_raster False`
This will, in theory, turn all your rasters into hard rasters and correct the problem. But, it wouldn't be using correct raster mode. But, the mode that is quite similar to how lightburn would run the moshi.

dorsi90, 2022-03-16:
>In console, type: set -p moshi/driver/0 enable_raster False

Tested and you are right, the text is in the right place but the it's done very very slow as lightburn so like vector engraving.

tatarize, 2022-03-17:
It's excactly vector engraving. But, it also tends to mean there's something weird about raster mode that I don't know since it really should have positioned things about the same way regardless of the mode change.

I purchased a new moshiboard since that's basically what I need to correct this.

dorsi90, 2022-03-17:
@tatarize if you get a new moshiboard just connect +24v GND and L
In the picture below how is connected my board
[image]

I see that m2 nano also require +5v but moshiboard regulate the voltage itself

tatarize, 2022-03-21:
My board doesn't have that UU1045 thing. Or the blue whatever that is there. Still waiting on hardware to figure this out.

dorsi90, 2022-03-21:
It's a LC filter that filter out any spikes. My board 4.85 version has an inductor also where il located the big white resistor of 150ohm 5w and i think it's used to generate the 5v in switching (i need to take off the board to be sure).
The k40 power supply only generate +24v in switching while +12v and +5v are generated from linear regulator which is not efficient.

tatarize, 2022-03-21:
My version is 4.5. Well, I'm glad this sort of thing isn't the Lihuiyu boards since there's some extra emphasis to figure out those boards to the stage that we track the particular versions over time to see if things are strange apparently the V4 boards react weirdly to some inputs and have these giant gnarly stepper motors rather than the middle of the line steppers you find on the later versions. Though the earlier steppers on the moshi are 2 chips with heat sinks, I know @joerlane's done some work tapping into some of the Lihuiyu boards with a step-stick lately. 

--- Ali Express shipped my package, so now it's down to a long boat from China.

closed 2024-10-04 by jpirnay