meerk40t/meerk40t #870: [Suggestion] Auto-reload on file change

Sophist-UK, 2022-03-01:
Have an ability to watch the time-stamp on an SVG file already loaded and reload the file if there is a change in the timestamp because e.g. you have been editing it in Inkscape and have just saved a new version.

It seems easy to have a timer that would do this, just a bit more challenging to provide a decent UI for it.

tatarize, 2022-03-09:
This seems a bit invasive and requires constant pinging. If they ask for the reload that's fine but just constantly checking and replacing work seems a bit goofy, especially since MK does some clear edits or whatever to files as needed.

Sophist-UK, 2022-03-09:
On Windows you can request a callback when it changes. Not sure about other platforms. However if you are editing in Inkscape because there are minor issues with the design an auto-reload **option** might be useful.

tiger12506, 2022-03-09:
Other platforms have similar mechanisms for "watch" of a file to get that callback like sophist is describing, but they are wildly different across platforms. It would be a situation similar to the notification thing we tried to deal with -- looking for a cross-platform library to handle all cases, not finding it, settling for os-specific implementations of the feature... I could be mistaken, but that was my impression last time I looked into it.

I don't think the idea is terrible, as long as it defaults off, of course.

tatarize, 2022-03-10:
At that rate we could just have a command that does "reload_files_if_changed" search for any files in elements and reload them if they are different than their timestamps. Then if you wanted you could set that as a timer. Since it won't reload them if nothing changes.

tatarize, 2022-03-10:
`timer 0 10 .reload_files_if_changed`

And you could put that in your batch to load on start.

closed 2024-10-04 by jpirnay