meerk40t/meerk40t #873: [Bug]: Not responding on close

dorsi90, 2022-03-03:
### Summary Description

The application not responding on close when it tries to connect over usb.
If i click some button like home the application tries to connect to the k40 but if it's unplugged or for other reasons the connection failed then the app tries to connect in loop. At this point if i close the app then not responding maybe because waiting for the loop connection terminate your job but if before close the app i click stop button i can close the app as expected.
So a solution can be just terminate the usb connection on close event like call the code of stop button.
Note: clicking reset or release usb before close the app wont solve the problem, only stop button works

### MeerK40t Version

0.7.5001

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

tatarize, 2022-03-05:
Hm.  I seem to recall having this happen. The check for shutdown=True occurs during the sending process. So if you aren't connected or sending it doesn't detect that or die gracefully. The connect cycles are a different loop than the send cycles and the send cycles are the only ones that check for the shutdown check.

dorsi90, 2022-03-05:
Well, then clicking stop button before close abort the send cycle and there isn't the problem on close the app.
The solution can be just call the code of stop button on close event. I havent read the code but i can suppose the send loop is running from another thread and the application waiting for no threads running.
If you allow me i would also like to suggest of adding a dialog on close that if there are some process runnings notify the user that some process still running and if he's sure to close the app.

tatarize, 2022-03-06:
I believe this may be fixed in 0.8.0 but wasn't fixed in the 0.7.x version. The reconnect loop doesn't detect the attempt to shutdown the program whereas the regular send loops (different loops for different devices) does detect that. It also could be moshi related if that's the routine since the reconnect loops are handled in the devices themselves.

dorsi90, 2022-03-06:
I see that clicking stop also interrupt the reconnect loop. In this case calling automatically the stop routine on shutdown stops the reconnect loop and solves the issue related at least for moshi.

tatarize, 2022-03-12:
This should be fixed in #878.

closed 2023-05-01 by dorsi90