meerk40t/meerk40t #876: [Suggestion]: Manual regmarks for contour cut proyect.

JJMorales20, 2022-03-09:
### Summary Description

I suggest to add the option to have a wizard to do contour cut.
The flow process can be little bit easy.

At this point, we are able to tell our laser to move the head to any point on the workspace, so, we can start the process by making 1 button to send the head to the upper left corner of our design(wich will have to have a printed rectangle as a reference point), once we are on the upper left corner of our job, Meerk40t migh show a message and a button to suggest to "fire" a pulse for 200ms(or a custom value) so we can clearly see where is supposed to be our printed media.
Then, once the upper left corner is placed, this is the tricky part; we might be able to keep with Meerk40t controlling the laser with the help of an interface that allows us to move in up/down/left/right direction with the option of advances of 1inch, 1/2inch, and mils, or 3cm, 1cm, 1mm, and microns or steps for a fine tune, in this step, Meerk40t will keep a track of all the movements until we(should fire few laser short pulses) have aligned our laser to the lower right corner of our design.

Once Meerk40t have taken note about the 2 corner points(as a coordenates) we can think in a code to do an automatic scale and rotation adjustment(in case the print is nor perfectly vertical/horizontal) so Meerk40t can do this for us.

Once the print is placed, and Meerk40t have measured 2 points, rotation and scale adjusted, the job can start with the parameters to cut/mark that material.

This would be a complete manual way to do contour cut, but a very efficient one.
I can help you to design all the steps, and I can show you a short video explaining all the steps to follow the main idea.

Hope this issue can be done, because will be a huge help for a lot of users.
At this time, I already do it completely manual, and it's quite easy for me, but I think this feature might be a must on Meerk40t to improve it little more than how good is at this time.

I will be totally glad to help if its needed.
And sorry for my english, is not my native lang!

### MeerK40t Version

0.7.4

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2022-03-09:
I am quite able to program this code this process and I am not sure this is entirely sure this workflow is what would be the best for users. Workflows that work for you do not necessarily work for others and it's a bit harder to figure out which workflow would be the best one for users.

I was thinking about a somewhat similar process since I recently was lining up some material to do a pencil in a jig and found it somewhat annoying. What the laser moving and scaling was somewhat combined? Imagine we have a tool/mode that lets you select elements in the scene.  So you select an image and the laser moves to the upper-left hand corner of the image. You could be given on-screen controls that let you select the a corner and fire pulses a bit like navigation. So you select the object and it goes to the selected corner (if that corner is within the laser bed). When you resize the image by either moving it or by resizing it grabbing the corner, the laser also moves. So basically we turn on a "corner lock" feature. So everything you manipulate in the scene also moves the laser to that object's position automatically.

While it's not quite a wizard asking you questions or anything it would get you to the right result. You need to line up your object in the scene. So rather than clicking around with navigator you'd just move things around The selection box in this mode would give show you the corners and the corner the laser movement is currently locked to. You could also lock it to the center. When you switch selected objects it would move the laser to the new center of the other selected objects.

In 0.8.x there's also a feature which is intended for galvo lasers but is coded generally so all spoolers have this ability. Where you create idle jobs that just execute on the machine over and over again, this is, in galvo, for lighting around the edges of the project.  For example:

[image]

And while plotters-style lasers are not super-speed galvo style lasers. We could load an idle job to do something like outline the edge of the selected object forever. I'm not sure how this would work perfectly but if you had a laser-pointer for your laser as many people do. We could also us this laser-position lock to keep the laser cycling in a loop. And while it wouldn't be anywhere near as fast as a galvo to produce a seemingly solid laser design. You could keep adjusting the material without going back to the machine to tweak the position. I'm not quite as concerned with merely moving the laser while somebody has their hands inside the thing adjusting the position of the material.

---

In my head this sounds like it would provide a richer experience with less overhead. We'd simply put the laser into a mode that depends on the selection within the scene. This could be done in the most basic sense with a couple lines of code. Basically when the selection changes and we have this mode set we move the laser to a particular corner (or center) of the project and keep it there adjusting the position when the position of that corner is changed. We already detect these changes to adjust the drawing of the selection box. We'd simply need a little bit of code that would send move commands to put the laser at the selection box corner. And do this each time we detect a change.

Rather than fix the location of the laser in the field and span the element between those corners we'd merge the process of adjusting the laser position with adjusting the elements in the scene. We'd could do top-left, top-right, bottom-left, bottom-right, center, simple trace (perform a trace each time the selection is changed) and looped simple trace (keep performing the trace over and over again until told to stop). --- Codewise most developers here could whip up the basic corner-lock thing a fairly short period of time.

JJMorales20, 2022-03-09:
Hi! Thanks for the answer, hope this could be a future feature, if its needed I will be testing this as much as possible to fine tune it and help with feedback to improve it.

The methodology I suggested comes from the internal process from manual or semiautomatic contour cut for cutting plotters.
In a cutting plotter software, you are able to put 4 registration marks, each mark have a center point which is the reference point for the software.

So, once we have printed our design with the 4 regmarks, the tip of the blade have to be kind of near from the 1st reg mark, the software will show a digital move pad, and with that move pad(you can also use physical keyboard arrow keys) you can adjust the tip of the blade until makes contact at the centre of the reg mark.
At this time, if you hit ENTER the software will record this firs low-right cordenate, then the plotter moves the material until the blade is near or aligned with the reg mark 2, in case is not perfectly aligned, then we can still move with the digital pad, until the blade is perfectly aligned, once done you can hit enter and the software will take note about the top-right cordenate, then the process its repeated for the 3rd and 4th reg marks, at the end of the 4th reg mark alignment, the software takes the vector for the cut job and stretches using those 4 points as a mask or reference.

So, in this case, if the print is smaller, the software will calculate the new scale, rotation and possible skew/tilt of the print(which is fairly uncommon to see on most prints), once the vector is recalculated or adjusted, user is asked to start the cutting process.

I will stay alert if you need help, I don't code but I could run all the needed tests or help with the dummy design for it.

Thanks again for the quick response!

tiger12506, 2022-03-09:
I think the auto-scaling makes this a useful feature. How to expose it in the UI is another question. Somewhere in the positioning pane, you'd have coords for two points, click buttons to set them to where laser current position is, button for apply which scales/moves currently selected elements to fit within that bounding box.... Just thinking.

tatarize, 2022-03-10:
Well, my counter-suggestion was largely because of the ease of coding that. Corner lock as a feature would be pretty straight-forward but also just a few lines of code.

I could see defining a series of points. We could just make registration marks a thing probably in elements, maybe in device. We could save and load and keep some persistent positions around. This wouldn't be hard. It would be a more hard-coded version of the number-key binds for positions. We define positions and then add a command for goto_position 1 etc. We save the array of positions on open and close. These could be useful for other purposes, mostly goto that position stuff. We display these points in the main scene.

And let you drag an edge to snap-to these points. Then you could get the bulk of the work done. You define the registration points then you snap the corners to these points when you're resizing at a specific corner.

* You need positions that get registered.
* Scene to display these positions.
* Snap-to for the resizing tools.

We'd be saving the positions so that you could set things like jig points or user origins or a series of jig points (say if you wanted to make 10 such dog tags and have 20 jig points for the top-left and bottom-right points for these) or just go to various set positions. The scene would draw these points. And we'd snap-to probably requiring dragging with a modifier key pressed or something. It would take a lot more interactions to refresh the autoscaling on image rotations. But, the modifications here aren't too extreme or hard.

closed 2024-10-04 by jpirnay