meerk40t/meerk40t #881: L2R or R2L raster is currently broken

Sophist-UK, 2022-03-15:
L2R or R2L raster op is completely screwy (as is second half of cross hatch).

Have not tested image but probably similar.

I tried to upload a video but it wouldn't upload. So here is my attempt to describe Rastering a circle at very slow speed and with a step of 100 (so step should be visibly obvious).

Should start with tiny burn on first sweep - but instead does a 1cm-1.5cm burn. Then a series of twitches in both directions horizontally and a long move vertically, then burns a similar long burn in the opposite direction without any step at all.

tiger12506, 2022-03-16:
https://user-images.githubusercontent.com/1036922/158495771-2c1993db-7b8b-47cb-b267-a2f707cdce04.mp4

tiger12506, 2022-03-16:
It seems to produce a more correct result when "Alternative Rastering" is checked. I thought maybe it was an issue with too large of a step size -- indeed it seems that large step sizes cause greater incorrect vertical offsets... but seem to have no bearing on the complete failure shown in video. That is purely having "Alternative Rastering" unchecked.

tiger12506, 2022-03-16:
Also, I noticed that none of my tests actually started where my head was and went -x before attempting raster.

tiger12506, 2022-03-16:
In case it matters, all of my tests so far were done with Optimize checked. That is Rapid Moves, and Burn Complete Subpaths. I'd have to redo all the tests with optimize off if anyone were interested.

tatarize, 2022-03-16:
Hm. That seems like it doesn't work.

tatarize, 2022-03-20:
The aptly numbered #911 should correct this.

tatarize, 2022-04-17:
Fixed in 0.7.6000

closed 2022-04-17 by tatarize