meerk40t/meerk40t #884: [Bug]: Moshi not cutting rasters

TheGarageEngineer, 2022-03-16:
### Summary Description

First, I just got this old K40 with a moshi board10105 v4.87 .  I am new to git hub too so I am learning. Let me know if I am not following procedures correctly.
I appreciate the work this community is doing with this software.

I seem to be cutting ok. But The problem I am having is raster cutting paths.  I created the design in inkscape, converted text and paths to outlines.  saved as svg and inported to meerk40t,

When laser runs just the raster action, it cuts a few lines of the raster then either runs to the left or right and rams into the rail.  I was running at 150mm/sec but I read some where to run at 70 mm/sec but that did not help.   Then I started to home the machine before each time I executed the file.  still same problem.

First what is the correct steps to load a cut to the laser after it is imported to meerk40t?  Execute - validate - blob - spool?  or can I just arm - start the file to the laser.

Based on other post I am including some screen shots.  I do not know if it is important for your analysis.

here is the meerk40t screen and file
[image]
 
this is the the results of the job at 150 mm/sec.  ran a few lines then rammed straight into the right rail (blue marker)
[image]

this is the results at 69 mm/sec. more lines where cut but then rammed straight into left rail.
[image]

Thanks for all you help.

### MeerK40t Version

0.7.5001

### MeerK40t Type

Executable

### Your Operating System

Windows

tiger12506, 2022-03-16:
> First what is the correct steps to load a cut to the laser after it is imported to meerk40t? Execute - validate - blob - spool? or can I just arm - start the file to the laser.

Arm -> Start is just a shortcut for running through all the steps in the Execute Job window.

TheGarageEngineer, 2022-03-16:
> > First what is the correct steps to load a cut to the laser after it is imported to meerk40t? Execute - validate - blob - spool? or can I just arm - start the file to the laser.
> 
> Arm -> Start is just a shortcut for running through all the steps in the Execute Job window.

So it is an ok procedure to arm - start for the moshi board instead of running through all the steps in the execute window?

dorsi90, 2022-03-16:
>So it is an ok procedure to arm - start for the moshi board instead of running through all the steps in the execute window?

Yes it's the same. You can also disable remove the arm button in Settings.

Regars your issue that bumps into the rail on left side is because moshiboard need some margins in x-y so just move your object 10mm right and 10mm bottom

dorsi90, 2022-03-16:
@TheGarageEngineer can you test the svg file below and check if the raster is in the right place, please?
Just do right click on the picture below and save as, to get the file.
[image]

TheGarageEngineer, 2022-03-16:
> Yes it's the same. You can also disable remove the arm button in Settings.
>
>
>
> Regars your issue that bumps into the rail on left side is because moshiboard need some margins in x-y so just move your [...]

Thank you for info. Great ideas. I will try.  But it's not just bumping the rail it is hitting the rail and staying on the rail and looks like it is trying to continue the cut on the rail.  And not moving left or right. But let me try your suggestion

dorsi90, 2022-03-16:
>Thank you for info. Great ideas. I will try. But it's not just bumping the rail it is hitting the rail and staying on the rail and looks like it is trying to continue the cut on the rail. And not moving left or right. But let me try your suggestion

When it touch the end stop moshiboard not working as expected so just give some margins and don't forget that raster have overscan so i think you need add margins on x axis even considering this value.

if you can test the file i sent you before i would be grateful because i opened an issue of wrong raster position with moshiboard that we couldn't solve so it would be helpful if you also have the same problem or just me.

TheGarageEngineer, 2022-03-16:
[image]
So raster cut first and was off and low.  cut boarder was in correct location

TheGarageEngineer, 2022-03-16:
So I tried to run my logo as I did yesterday.  Yesterday I was at (0,0) but today I moved to start position to (20,20) and this is what I got.
[image]
#1 arrow - raster started low (offset), just like the 'test' raster. after a few lines it moved to location #2
#2 arrow - and continued the raster where it left off from #1. after a few more lines rastered
#3 arrow -  it moved to the right about 10mm and began the outline cut sequence, but instead of moving in a circular pattern as per the design it cut a straight line because the lens was headed to the right rail and would have crashed the rail if I had not turned off the machine.

I hope this helps with the diagnosis.

dorsi90, 2022-03-16:
@TheGarageEngineer have you tried with 0.7.5000 version instead of .5001?
In the last version the paths are engraved separatly that maybe can introduce more issues with moshiboard.
At the moment on the first start the raster is not inbox but after finish if you click start again without touching anything the text is inbox so in the right position.

dorsi90, 2022-03-16:
@tatarize as you can see here regards to moshi raster position the issue is in the object position maybe when calculate the offset.
In my tests the margins was 10,10 and the raster is just a little bit below the vector box, in this case with margins 20,20 the raster position is like double times below the box

TheGarageEngineer, 2022-03-16:
> 

@dorsi90 I have not tried 0.7.5000   I will look into that.  
I only ran file once.  I did not click start again without touching anything. I will try that procedure and respond with my results.

dorsi90, 2022-03-16:
Ok, you have my same issue.
If you can send your svg file i can test it for you as well.

TheGarageEngineer, 2022-03-16:
> Ok, you have my same issue. If you can send your svg file i can test it for you as well.

@dorsi90 
Here is my file.
Also, have you tried ver 0.7.500 yet?  does the same problem occur with the 'test' file?

[image]

dorsi90, 2022-03-16:
@TheGarageEngineer yes is the same but you use the executable so i think only has ruida control filx and not include the raster path separation maybe it was included later and usable only if you run the app from source code. Basically it engrave the letters separatly and in my case the T was in wrong place so outside the box while EST are in the right place inside the box.
Tomorrow i check try to engrave your file and tell you what happen.

tatarize, 2022-03-17:
In console, type: `set -p moshi/driver/0 enable_raster False`

This will turn your rasters into hard-rasters which will basically make these things correct. I ordered the new moshiboard because even with the added debug stuff I narrowed down some of the issues but it didn't see to correct things fully. I could also try extra move-to-origins after every raster since that seems to be the core protocol there. And may be a solid enough check for that branch.

dorsi90, 2022-03-17:
@TheGarageEngineer your file looks like my file so on first start the raster is in wrong position
[image]

While on second start it's perfect
[image]

TheGarageEngineer, 2022-03-17:
> @TheGarageEngineer your file looks like my file so on first start the raster is in wrong position [image]
> 
> While on second start it's perfect [image]

I wasn't even getting the consistency that you got.  But I will try what @tatarize recommended and see if that helps.
Thanks

dorsi90, 2022-03-17:
I put the object in position 10,10.
I have already tested setting enable_raster to false and the raster is in the right position but the engraving is done very slow like vector engraving so you burn too much.

TheGarageEngineer, 2022-03-17:
good to know about the 10,10.  That is probable why I was having problem.  I will try tonight.
too bad it is too slow.  
What is the difference between raster engrave and vector engrave?

dorsi90, 2022-03-17:
You can read the differences and others good information from meerk40t wiki [here](https://github.com/meerk40t/meerk40t/wiki)

TheGarageEngineer, 2022-03-17:
So I have not changed the settings yet but I ran a test of my logo

I positioned the logo at 10,10. It started the raster lower than it should be as we have been seeing. But it only ran a few lines.  Then it stopped.  Moved to the correct location on the computer screen and then continued the raster.  Then it did the proper cut out.  
Then with out doing anything.  I ran it again.  The lens took off directly to the right rail.  I turned the machine off so it would not crash.  
@dorsi90 

I beginning to think I need to change the board out. 
[image]

dorsi90, 2022-03-17:
With version 0.7.5001 this also happens to me, try the version 0.7.5000
I see also that you engrave from bottom to top while i set top to bottom in raster operation i don't know if it introduce more issues. However i think that with this board at the moment is better to use moshidraw that also has many many bugs but when start to engrave it complete the job without issues.

>I beginning to think I need to change the board out.

I already bought the M2 Nano board from AliExpress for 22 euros (only the mainboard) and i still waiting that arrive to me and i can start to use this beautiful software

TheGarageEngineer, 2022-03-18:
> With version 0.7.5001 this also happens to me, try the version 0.7.5000
>
> I see also that you engrave from bottom to top while i set top to bottom in raster operation i don't know if it introduce [...]

Where could I get moshi draw?  I went to their website but the only button I was on the page was buy.  And it did no action. Do you have a link to moshidraw?

I have a diode laser ortur laser master, I think it has a gbrl board. I think I am going to look at it to see if it can be rewired for the m40.

dorsi90, 2022-03-18:
Which software do you use for your k40?
Do you have the usb dongle? Is it red or orange?

TheGarageEngineer, 2022-03-18:
I have nothing.  I was given the machine and that was it.  No dongle either.  That is why I was hoping meerk40t would work better

Dennis Machida
Trail47.com

tatarize, 2022-03-18:
The software requires windows 7 or so, and the dongle which also requires some drivers. So it's a bit of a pain to get it working even if you had the dongle which would run like $110 dollars to get a replacement for. The best bet is to order a new M2 Nano. I need the hardware to figure out the issue it'll still be a while before it arrives. And a bit longer to test the stuff. I wish it worked better too. It seemed to mostly work before and the current PR trying to fix it just doesn't do enough though I think it *should* be better if I homed it between each raster event. Which I think I should be able able to make sure it actually does. And hopefully I'll find a little bit of time to home before each raster, and try that.

Even if you can find the Moshi software, it does absolutely require that dongle and it's a bit of a pain to get the dongle drivers to work. The M2 Nano board runs 25 bucks and has a lot of open source support today. And you can get pretty reasonable GRBL cards for 80 bucks or so.

dorsi90, 2022-03-18:
I agree with @tatarize that is better to buy the M2 Nano. I have the moshi software and i also can send you it but without usb dongle it's not working. In my case with windows 8.1 OS the red usb dongle not required drivers i don't know for orange dongle but the price is of course expensive then M2 Nano board If you bought k40 you can ask to the seller for the dongle that usually is included when buy this machines.

tatarize, 2022-03-18:
But, 25 bucks and deal with the weird power connection issue is likely preferred. The M2 Nanos are nice little cards. My replacement moshiboard is 10 days before it even gets on the Chinese boat. Though I could try homing before each raster attempt in the moshi branch where I tried to fix most of the errors. https://github.com/meerk40t/meerk40t/pull/878

And there's a couple other things I could try. It wouldn't be that hard to add in a bunch of potential raster fixes as settings that are disabled and ask for some attempts to be made.

1) Raster offset always set to 0.
2) Home before all rasters.
3) Raster start at first position for the given offset (never use move).
4) Avoid all negative numbers within rasters by setting correct minimum offset.

I seem to recall when I was trying to get rasters working they were a bit annoying and sort of broke a lot. And had a somewhat nasty habit of suddenly running into the left wall. I got a version of raster that seemed like it worked. Though I now realize it must have been off in positioning and without outlining the raster I messed that up. There's only so many ways to build a moshiblob, but even in the best conditions I still consider moshiboards to be a bit inferior to M2 Nano, except that they store their current position which is sometimes rather nice. And I don't know how big the buffer is but I've never needed to know since it seems like it hasn't tapped out for me.

TheGarageEngineer, 2022-03-18:
I think I will look into a new board

@dorsi90 @tatarize  Check out this video. The presenter sounds like he was having the same issue with moshi. Then upgraded to a cohesion 3d board.  

https://youtu.be/iB6wjvk6vRc

dorsi90, 2022-03-19:
@TheGarageEngineer it is not the same he used moshidraw without know how it works. Moshidraw crash when manipulating the object, can't import svg files, ecc but it works perfect when engrave and cut. It can use the position relative to the plane or to the object so if you use relative to the object without know how it woks then you have wrong positions.
Cohesion 3D is a good board to use with lightburn but you need to buy a license.

@tatarize the buffer is more then 4kb. When engraving large image in moshidraw it show the total data to send and after it has sends some data it show busy message then when not busy sends 4kb and then busy again.

tatarize, 2022-03-19:
Sweet. I had always wondered that. It isn't too hard to detect the device as busy but it wasn't clear where the cutoff was between one chunk of data and the next. But, there should be some data limit since infinite buffer isn't plausible.

tatarize, 2022-04-10:
Well, I got my new replacement moshi card. I can at least test this on hardware and I'll hopefully find a time to do that.

dorsi90, 2022-04-10:
I am still waiting for the m2 nano card. These packages from China are too slow :(
At the moment i do the job run it more times (the first time with laser turned off). Starting from the second time the raster is in the right place but sometimes after the head moves to the object location and when start the raster mode the head moves to the right direction and bumps into the rail so i need to turn off the machine and start it again.
Which moshiboard version do you have?

closed 2024-10-04 by jpirnay