meerk40t/meerk40t #886: 0.8.0 RC-1

tatarize, 2022-03-16:
### Summary Description

The latest Release Candidate is 1. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

### MeerK40t Version

0.8.0 RC-1

### MeerK40t Type

Executable

### Your Operating System

Other

tatarize, 2022-03-16:
https://github.com/meerk40t/meerk40t/releases/tag/0.8.0002-RC1

joerlane, 2022-03-16:
MeerK40t crash log. Version: 0.8.0002 RC1 git on Darwin:x86_64 - 4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "/Volumes/meerk40t/meerk40t/lihuiyu/gui/gui.py", line 65, in <lambda>
    "action": lambda v: service("window toggle Configuration\n"),
  File "/Volumes/meerk40t/meerk40t/kernel/context.py", line 33, in __call__
    return self._kernel.console(data)
  File "/Volumes/meerk40t/meerk40t/kernel/kernel.py", line 2004, in console
    self._console_parse(command, channel=self._console_channel)
  File "/Volumes/meerk40t/meerk40t/kernel/kernel.py", line 2083, in _console_parse
    data, remainder, input_type = command_funct(
  File "/Volumes/meerk40t/meerk40t/kernel/functions.py", line 215, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/Volumes/meerk40t/meerk40t/gui/wxmeerk40t.py", line 467, in window_open
    kernel.run_later(window_open, None)
  File "/Volumes/meerk40t/meerk40t/gui/wxmeerk40t.py", line 222, in run_later
    command(*args)
  File "/Volumes/meerk40t/meerk40t/gui/wxmeerk40t.py", line 449, in window_open
    path.open_as(window_uri, window_name, parent, *args)
  File "/Volumes/meerk40t/meerk40t/kernel/context.py", line 424, in open_as
    instance = open_object(self, instance_path, *args, **kwargs)
  File "/Volumes/meerk40t/meerk40t/lihuiyu/gui/lhystudiosdrivergui.py", line 1109, in __init__
    self.SetupPanel = ConfigurationSetupPanel(
  File "/Volumes/meerk40t/meerk40t/lihuiyu/gui/lhystudiosdrivergui.py", line 944, in __init__
    self.check_twitchless.SetValue(self.context.twitchless)
AttributeError: 'LihuiyuDevice' object has no attribute 'twitchless'

Sophist-UK, 2022-03-16:
#867 switched user option from "twitchless" to "twitchfull" i.e. changed from the user choosing to use twitchless to choosing to use the old twitchfull method.

joerlane, 2022-03-16:
Continues to try USB on quit; when there is no USB driver selected.  This list of errors goes on for quite a while:

Waiting for thread LhyPipe(lhystudios): <Thread(LhyPipe(lhystudios), stopped 123145511960576)>
    Thread LhyPipe(lhystudios) has finished. <Thread(LhyPipe(lhystudios), stopped 123145511960576)>
    Thread Spooler(lhystudios1) exited safely
    Finishing Thread LhyPipe(lhystudios1) for <Thread(LhyPipe(lhystudios1), started 123145545539584)>
    Asking thread to stop.
    Suspended Signal: pipe;buffer for (0,)
    Suspended Signal: pipe;usb_status for ('    No Windll interfacing. Skipping.',)
    Suspended Signal: pipe;usb_status for ('    Using LibUSB to connect.',)
    Suspended Signal: pipe;usb_status for ('    No Windll interfacing. Skipping.',)
    Suspended Signal: pipe;usb_status for ('    Attempting connection to USB.',)
    Suspended Signal: pipe;usb_status for ('    Using LibUSB to connect.',)
    Suspended Signal: pipe;state for ('STATE_USB_CONNECTING',)
    Suspended Signal: pipe;usb_status for ('    Attempting connection to USB.',)
    Suspended Signal: pipe;usb_status for ('    Attempting connection to USB.',)
    Suspended Signal: pipe;state for ('STATE_USB_CONNECTING',)
    Suspended Signal: pipe;usb_status for ('    Using LibUSB to connect.',)
    Suspended Signal: pipe;usb_status for ('    Attempting connection to USB.',)
    Suspended Signal: pipe;usb_status for ('    Finding devices.',)
    Suspended Signal: pipe;usb_status for ('    Using LibUSB to connect.',)
    Suspended Signal: pipe;usb_status for ('    Finding devices.',)
    Suspended Signal: pipe;usb_status for ('    Devices Not Found.',)
    Suspended Signal: pipe;usb_status for ('    Connection to USB failed.\n    ',)
    Suspended Signal: pipe;usb_status for ('    Connection to USB failed.\n    ',)
    Suspended Signal: pipe;state for ('STATE_CONNECTION_FAILED',)
    Suspended Signal: pipe;usb_status for ('    Using LibUSB to connect.',)
    Suspended Signal: pipe;usb_status for ('    Attempting connection to USB.',)
    Suspended Signal: pipe;state for ('STATE_USB_CONNECTING',)
    Suspended Signal: pipe;usb_status for ('    Attempting connection to USB.',)
    Suspended Signal: pipe;usb_status for ('    Using LibUSB to connect.',)
    Suspended Signal: pipe;usb_status for ('    Finding devices.',)
    Suspended Signal: pipe;usb_status for ('    Devices Not Found.',)
    Suspended Signal: pipe;usb_status for ('    Connection to USB failed.\n    ',)
    Suspended Signal: pipe;usb_status for ('    Connection to USB failed.\n    ',)
    Suspended Signal: pipe;state for ('STATE_CONNECTION_FAILED',)

Sophist-UK, 2022-03-16:
Looks like "twichful" had been renamed "twitches" but disks still refer to "twitchless".

Sophist-UK, 2022-03-16:
```
MeerK40t crash log. Version: 0.8.0002 RC1 git on Darwin:x86_64 - 4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
...
  File "/Volumes/meerk40t/meerk40t/lihuiyu/gui/lhystudiosdrivergui.py", line 944, in __init__
    self.check_twitchless.SetValue(self.context.twitchless)
AttributeError: 'LihuiyuDevice' object has no attribute 'twitchless'
```
Fixed in #888

joerlane, 2022-03-16:
Stroke appears nearly invisible when drawing shapes. It would also help to have any kind of feedback when in a drawing mode. Lighting up the selected ribbon button would be fine, or changing the mouse cursor. 
[image]

Sophist-UK, 2022-03-16:
Several tens if not hundreds of the the following when I started MK on a Linux virtual image:
```
20:36:05: Debug: ClientToScreen cannot work when toplevel window is not shown
```
And also loads of these:
```
20:39:58: Debug: window sipwxTreeCtrl(0x1ff1000) lost focus even though it didn't have it
```

tatarize, 2022-03-17:
Partially retconned bug after the last merge thing.

tatarize, 2022-03-17:
> Questions: are the dialog_xxxx commands supposed to appear by issuing "help" in the console - because they don't

No. They are specifically flagged "hidden=True". And they are prefixed with a non-echo `.` to start.

closed 2022-03-17 by tatarize