meerk40t/meerk40t #893: [Bug]: BBox not properly calculated

jpirnay, 2022-03-16:
### Summary Description

After applying a matrix command the bounding box for the resulting shape is not properly calculated.

### Additional Details

Steps to reproduce:
```
circle 4cm 4cm 1cm
scale 1 2
matrix 1 0 0.5 1 0 0
```

This is true for Meerk40t 0.7 legacy:
[image]
As well for Meerk40t 0.8:
[image]

### MeerK40t Version

meerk40t 0.8 / 0.7legacy

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

tatarize, 2022-03-17:
Arg. That seems accurate. The arcs are weird when you apply a sheer to them they get wonky. I think there's a somewhat deep bug in the that sort of thing that I've glossed over for a year now where it technically fails one of the WC3 tests.

Sophist-UK, 2022-03-17:
I suspect that if this was a raster element, the raster might be limited by the bbox and bits would not be burned.

tatarize, 2022-03-17:
Yeah, you'd actually have to figure out a way to sheer the designs to trigger the bug. Which isn't something most people would figure out how to do since there's not something instantly naively in there. Except for jp's code there.

Sophist-UK, 2022-03-17:
1. It depends whether the shear transform can be existing in the SVG file. If I create the thing in inkscape and have the correct (or incorrect) inkscape save settings, can I save it with the shear transformation and get it into MK as a raster element. If so that makes it significantly more likely.
2. I suspect that this is a bug in SVGelements (because that is where bbox resides) - in which case even though this might be rare in MK, it may be far less rare in a different project using SVGelements.

tatarize, 2022-03-19:
The coords-trans-08-t.svg test also fails since it uses sheered arcs. The problem might actually be somewhat fundamental within svgelements. I've never tracked it down.

[coords-trans-08-t.zip](https://github.com/meerk40t/meerk40t/files/8287110/coords-trans-08-t.zip)

Sophist-UK, 2022-03-19:
I suspect that it's fundamental to the calculations. However I suspect that the math had probably already been done by someone in open source code somewhere...

jpirnay, 2022-03-20:
> I suspect that it's fundamental to the calculations. However I suspect that the math had probably already been done by someone in open source code somewhere...

Agree, you don't need skew for it, try:
```
circle 4cm 4cm 2cm
rotate 30deg
scale 1 2
```
et voilà:
[image]

BTW as you can see in the screenshot, the item manipulation code is nearly ported to 0.8...

Sophist-UK, 2022-03-20:
Wow - that is crazy broken.

tatarize, 2022-03-21:
Hm.

`circle 4cm 4cm 2cm rotate 30deg scale 1 2` is totally wrong about the bounds there.
`circle 4cm 4cm 2cm  scale 1 2 rotate 30deg` gets the bounds perfectly right.

Yeah, okay, does not require skew.